libusb/libusb #182: Synchronous API cannot be effectively interrupted

philipl, 2016-04-11:
I spent some time this working on fixing a shutdown crash in the MTP backend for gvfs, which uses libmtp, which in turn uses libusb.

libmtp implements listening for mtp device events using the synchronous transfer API. I had originally thought there was no way to interrupt this call, and built some elaborate logic on my side to avoid races, but ultimately, I didn't attempt to join on the event thread - and I assumed the thread would remain blocked in the transfer call throughout the shutdown sequence, except in the rare case that an event arrived.

But I received reports of occasional crashes, and so I went and put together a serious of changes to allow for the use of the async transfer API in libmtp for event listening and then using that in gvfs. After all that was done, I did more research and discovered that closing a device is supposed to interrupt polling calls precisely to allow for clean shutdowns. (The documentation update to explain this isn't in the website API docs - I read it in the code comments)

That made me think that the synchronous API should be usable without me having to have done all my work. But I then saw that the synchronous API actually just turns around and starts trying to poll for events again if it's ever interrupted. This leads to one of two problems.

1) You get a crash because the device state is gone (device was closed afer all)

2) The code actually succeeds in polling for events again and my code blocks for a long time, trying to join the event thread

So, yay, my work to use the asynchronous API wasn't wasted, but it seems desirable that the synchronous API respects interruption.

Presumably the new interrupt_event_handler call also cannot interrupt synchronous API calls.

mcuee, 2016-12-02:
http://marc.info/?l=libusb-devel&m=147808669913635&w=2

I tend to think the above answered your question. If not, please comment.
"Yes, you are supposed to clean up any outstanding transfers
before closing. If you use synchronous calls, then I would
expect your apps to be single threaded, use timeouts
to detect errors, and close after timeout. If you've
a multi-threaded app you really should not be using
the synchronous API."

philipl, 2016-12-05:
Sure, as a matter of policy, this is fine. You should update the documentation to explain that interruption doesn't happen for synchronous calls.

Mind you, I don't think it's unreasonable to say that interruption happens for synchronous APIs. It doesn't require the caller to do anything crazy to react, and from an implementation point of view it's simpler than re-issuing the wait (what libusb does today).

But hey, I already re-wrote my code, so I'm not in the set of people affected by this behaviour. :-)

wm4, 2020-02-26:
I'm interested in this too, but for different reasons.  My issue isn't about documentation anymore, or the API user problem mcuee mentioned.

I'm mixing synchronous and asynchronous transfers. I'm connecting to multiple devices; connected devices use async transfers, but new devices need a larger number of control transfers for initialization first. These control transfers are better written as synchronous calls, would be complex and awkward in async.

Now I'd like to be able to gracefully stop the synchronous initialization, either when it took too long, or if the user requested a stop.

Is there really no way to do it? I could recompute the timeout for every control transfer and check for user aborts in between. For device closing, I'll use some sort of ref-counting with delayed closing on top of the libusb device handle, to make the higher level application logic simpler.

mcuee, 2020-07-29:
Reference:
https://github.com/libusb/libusb/issues/703

mcuee, 2021-08-19:
https://libusb.sourceforge.io/api-1.0/libusb_io.html
```
Additionally, there is no opportunity to cancel the transfer after the request has been submitted.
```

mcuee, 2021-08-19:
Further enhancement should be discussed in #703.

closed 2021-08-19 by mcuee