libusb/libusb #185: Error code -12 when opening Novint Falcon

agirault, 2016-05-10:
I am working on making libusb1 the default USB communication library for LibNifalcon on Windows:
https://github.com/libnifalcon/libnifalcon/pull/43. So far libftd2xx was used for Windows ans libusb1 for Unix and Mac only.

I built against [libusb-1.0.20](https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.20/).

I manage to get the device list with `libusb_get_device_list`, loop through it to find the Novint Falcon  (vendorId = 1027, productId = 52040) using the `libusb_get_device_descriptor`, but `libusb_open` returns **-12**.

Here is what the `libusb_device` gives me in the debugger:
```
falconDevice	0x00000000003af720	libusb_device *
	lock	0x0000000000000124	void *
	refcnt	1	int
	ctx	0x000000000037ec70 {debug=0 debug_fixed=0 event_pipe=0x000000000037ec78 {0, 0} ...}	libusb_context *
	bus_number	2 '\x2'	unsigned char
	port_number	1 '\x1'	unsigned char
	parent_dev	0x00000000003b5a80 {lock=0x0000000000000120 refcnt=3 ctx=0x000000000037ec70 {debug=0 debug_fixed=0 event_pipe=...} ...}	libusb_device *
	device_address	3 '\x3'	unsigned char
	num_configurations	1 '\x1'	unsigned char
	speed	LIBUSB_SPEED_FULL (2)	libusb_speed
	list	{prev=0x00000000003b60d0 {prev=0x00000000003b6780 {prev=0x00000000003b6f40 {prev=0x000000000037ec80 {...} ...} ...} ...} ...}	list_head
	session_data	841	unsigned long
	device_descriptor	libusb_device_descriptor
		bLength	18 '\x12'	unsigned char
		bDescriptorType	1 '\x1'	unsigned char
		bcdUSB	512	unsigned short
		bDeviceClass	0 '\0'	unsigned char
		bDeviceSubClass	0 '\0'	unsigned char
		bDeviceProtocol	0 '\0'	unsigned char
		bMaxPacketSize0	8 '\b'	unsigned char
		idVendor	1027	unsigned short
		idProduct	52040	unsigned short
		bcdDevice	1024	unsigned short
		iManufacturer	1 '\x1'	unsigned char
		iProduct	2 '\x2'	unsigned char
		iSerialNumber	3 '\x3'	unsigned char
		bNumConfigurations	1 '\x1'	unsigned char
	attached	1	int
	os_priv	0x00000000003af77c "\x1\x1\x1"	unsigned char[0]

```

Any idea?

Thank you.

mcuee, 2016-05-12:
In order to use libusb under Windows, you have to use a supported driver --
WinUSB for example and not FTDI vendor driver.

mcuee, 2016-06-12:
Closing this issue since this is not a libusb issue.

closed 2016-06-12 by mcuee