libusb/libusb #186: Crash in windows calling libusb_control_transfer() when another thread is handling usb events and submitting bulk transfers

HipsterSloth, 2016-05-13:
There is a crash I ran into in recently PS3EYEDRIVER, a libusb based PS3Eye Camera api (running against the 1.0.9 version of libusb). I was trying to set the exposure of the camera while the video feed was running. The crash happens sporadically in [windows_get_overlapped_result](https://github.com/libusb/libusb/blob/master/libusb/os/windows_nt_common.c#L574). 

In the current implementation of PS3EYEDRIVER has a separate thread is calling libusb_handle_events_timeout_completed() and submitting new bulk transfer requests every time a bulk transfer request is completed. This feeds a video frame queue. In the main thread, the user will occasionally make camera settings changes, which calls the synchronous libusb_control_transfer() api.

The [multithreading notes]((http://libusb.sourceforge.net/api-1.0/mtasync.html) in libusb seem to suggest that the polling thread has to be careful about using the right api when you have synchronous calls in the main thread:

> You may decide that it is OK for your multi-threaded application to ignore some of the rules and locks detailed above, because you don't think that two threads can ever be polling the descriptors at the same time. If that is the case, then that's good news for you because you don't have to worry. But be careful here; remember that the synchronous I/O functions do event handling internally. If you have one thread doing event handling in a loop (without implementing the rules and locking semantics documented above) and another trying to send a synchronous USB transfer, you will end up with two threads monitoring the same descriptors, and the above-described undesirable behaviour occuring. **The solution is for your polling thread to play by the rules**; the synchronous I/O functions do so, and this will result in them getting along in perfect harmony.

The video polling thread is using libusb_handle_events_timeout_completed() whose implementation appears to follow the multithreading safe event polling method outlined in that doc. So I _think_ we're trying to do the right thing.

My best guess at what is happening is that while the flying transfer queue is safely being accessed in [windows_handle_events](https://github.com/libusb/libusb/blob/master/libusb/os/windows_nt_common.c#L562-L574), it's _not_ safe to be accessing the transfer pointer fetched from the queue outside of the flying transfer lock. The main thread can interrupt, add a transfer (via libusb_control_transfer), and now that transfer pointer isn't looking at the same thing in the transfer queue it was when it was fetched from inside the lock.

I was able to reproduce the crash in the most recent version of libusb where the windows implementation has been refactored. It looks the the same potential underlying issue is still there:

https://github.com/libusb/libusb/blob/5e45e0741daee4fa295c6cc977edfb986c872152/libusb/os/windows_usb.c#L2298-L2311

I made a work around in PS3EYEDRIVER here that shuttles the control transfer quests onto the video transfer thread. This fixes the crash, but it's kind of gross. 

https://github.com/inspirit/PS3EYEDriver/pull/29

diabolo38, 2016-05-13:
you may try a more recent libusb than 1.0.9  issue in multi-thread context got fixed since that.

HipsterSloth, 2016-05-14:
Yeah I tested this issue against the most recent version in master (LIBUSB_NANO version 11108) and the crash still reproduces. That fix you mentioned was in LIBUSB_NANO version 10982. 

As I mentioned in my previous post:

> I was able to reproduce the crash in the most recent version of libusb where the windows implementation has been refactored. It looks the the same potential underlying issue is still there:
https://github.com/libusb/libusb/blob/5e45e0741daee4fa295c6cc977edfb986c872152/libusb/os/windows_usb.c#L2298-L2311

dickens, 2016-05-30:
> My best guess at what is happening is that while the flying transfer queue
> is safely being accessed in windows_handle_events [...]
This isn't a problem. The lock is taken to ensure that the list doesn't
change whilst iterating over it, but accessing an individual member of the
list is perfectly fine outside the lock. Once we have a pointer to the
transfer, the list can be modified in any way with no issues. The pointer
is to the transfer itself, not to a position inside the list.

Where exactly is the crash occurring? Do you have any logs or call stack?

HipsterSloth, 2016-05-30:
Thanks a bunch for taking a look at this!

The actual crash is a null pointer access:
```
Unhandled exception at 0x01395CF9 in PSMoveService.exe: 0xC0000005: Access violation reading location 0x00000000.
```

Here is the relevant portion of the callstack:

```
>	PSMoveService.exe!windows_get_overlapped_result(usbi_transfer * transfer, winfd * pollable_fd, unsigned long * io_result, unsigned long * io_size) Line 2025	C
 	PSMoveService.exe!windows_handle_events(libusb_context * ctx, pollfd * fds, unsigned int nfds, int num_ready) Line 574	C
 	PSMoveService.exe!handle_events(libusb_context * ctx, timeval * tv) Line 2242	C
 	PSMoveService.exe!libusb_handle_events_timeout_completed(libusb_context * ctx, timeval * tv, int * completed) Line 2328	C
 	PSMoveService.exe!ps3eye::USBMgr::transferThreadFunc() Line 489	C++
```

I looks like it's crashing trying to access a pointer inside pollable_fd->overlapped

```    
void windows_get_overlapped_result(struct usbi_transfer *transfer, struct winfd *pollable_fd, DWORD *io_result, DWORD *io_size)
{
	if (HasOverlappedIoCompletedSync(pollable_fd->overlapped)) {  // <-- Crashes in here
		*io_result = NO_ERROR;
		*io_size = (DWORD)pollable_fd->overlapped->InternalHigh;
	}
```

In particular the pointer in the ecx register is null, which comes from a field 8 bytes from the start of the overlapped structure:

```    
void windows_get_overlapped_result(struct usbi_transfer *transfer, struct winfd *pollable_fd, DWORD *io_result, DWORD *io_size)
{
01395CF0  push        ebp  
01395CF1  mov         ebp,esp  
	if (HasOverlappedIoCompletedSync(pollable_fd->overlapped)) {
01395CF3  mov         eax,dword ptr [pollable_fd]  
01395CF6  mov         ecx,dword ptr [eax+8]  
01395CF9  cmp         dword ptr [ecx],104h  
01395CFF  jne         windows_get_overlapped_result+2Ah (01395D1Ah)  
```

Looking at the definition of the _OVERLAPPED structure this appears to be the 'Pointer' field in the union:

```
typedef struct _OVERLAPPED {
    ULONG_PTR Internal;
    ULONG_PTR InternalHigh;
    union {
        struct {
            DWORD Offset;
            DWORD OffsetHigh;
        } DUMMYSTRUCTNAME;
        PVOID Pointer;
    } DUMMYUNIONNAME;

    HANDLE  hEvent;
} OVERLAPPED, *LPOVERLAPPED;
```

If we look at overlapped in watch window we can confirm that the pointer is in fact null:

```
pollable_fd->overlapped	0x098af640 {Internal=260 InternalHigh=1 Offset=0 ...}	_OVERLAPPED *
		Internal	260	unsigned long
		InternalHigh	1	unsigned long
		Offset	0	unsigned long
		OffsetHigh	0	unsigned long
		Pointer	0x00000000	void *
		hEvent	0x000004f8	void *
```

I've attached a zip containing:
* log file
* minidump (sadly I can't give the minidump with heap and have the zip still be small enough to fit in the issue)
* exe that is compiled against the static lib of libusb
* pdb file for the exe
* source files that correspond to all functions on the callstack

Let me know if there is anything else that would be helpful in debugging this issue. And thanks again.

[PSMoveServiceCrash.zip](https://github.com/libusb/libusb/files/288822/PSMoveServiceCrash.zip)

dickens, 2016-05-30:
So the crash is occurring because the overlapped member of the pollable_fd structure is NULL. It's not trying to access the Pointer field of overlapped, it is trying to read the value of the Internal field [1]. The 8-byte offset comes from the overlapped field inside struct winfd [2].

The only place where the overlapped field gets set to NULL is in usbi_free_fd() [3], which would be called in windows_clear_transfer_priv() after the transfer has been completed.

My gut says this is a race condition. Are transfers being submitted in rapid succession? How many threads are calling into libusb simultaneously? Are you able to turn on libusb's debug logging, or does the problem stop occurring?

Thanks for providing the info and details!

[1] https://github.com/libusb/libusb/blob/master/libusb/os/poll_windows.h#L39
[2] https://github.com/libusb/libusb/blob/master/libusb/os/poll_windows.h#L86
[3] https://github.com/libusb/libusb/blob/master/libusb/os/poll_windows.c#L419

HipsterSloth, 2016-05-30:
Ahh ok, Thanks for breaking down the details of the crash.

Yeah this is definitely a case of rapid submitted requests. The client application is mashing on the adjust camera exposure button. This results in a back-to-back set of control transfer requests on the main thread. Meanwhile the worker thread is pumping out async bulk transfer requests to read the video frames from the camera. So just two threads on total.

As for the logging I called libusb_set_debug like so:

```
    libusb_init(&usb_context);
    libusb_set_debug(usb_context, LIBUSB_LOG_LEVEL_DEBUG);
```

I also compiled with the `LIBUSB_DEBUG` proprocessor macro defined. I'm using boost logging to capture the logs and it's possible it's doing something funky with the io streams, but it looks like some of the other libusb logs are making it through so I presume that's not the issue?

dickens, 2016-05-30:
Hmm, that should have been enough to set the debug level, but the log
output would have contained *way* more output.

The appropriate preprocessor macro is ENABLE_DEBUG_LOGGING. Perhaps you can
give that a try?

HipsterSloth, 2016-05-30:
That did the trick. Here is the last 8 seconds worth of log. For some reason the boost log library I'm using isn't saving anything to the log file at the moment, only the console. So this is a snapshot of the 8 seconds leading up to the crash that I could copy out of the console. If you need more of the log I can monkey with the log library settings and see if I can get a file out.

You can see at the end where I'm mashing the exposure set request:

```
[10.195436] [000037d0] libusb: debug [usbi_add_pollfd] add fd 5 events 1
[10.195937] [0000448c] libusb: debug [libusb_unref_device] destroy device 1.9
[10.195937] [000037d0] libusb: debug [windows_handle_events] checking fd 6 with revents = 0001
[10.195937] [0000448c] libusb: debug [libusb_unref_device] destroy device 1.0
[10.195937] [000037d0] libusb: debug [usbi_remove_pollfd] remove fd 6
[10.195937] [0000448c] libusb: debug [libusb_exit]
[10.195937] [000037d0] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 16384
[10.195937] [0000448c] libusb: debug [libusb_exit] destroying default context
[10.196437] [000037d0] libusb: debug [usbi_handle_transfer_completion] transfer 00FC3160 has callback 0152A3A0
[2016-05-29 22:57:53.601118]: ClientConnection::start - Starting client connection id 0
[2016-05-29 22:57:53.605121]: ClientConnection::send_connection_info - Sending connection id to client 0
[2016-05-29 22:57:53.610123]: ClientConnection::start_tcp_write_queued_response - Sending TCP response
[2016-05-29 22:57:55.380226]: ClientConnection::handle_tcp_read_request_body - Read request body on connection0
[2016-05-29 22:57:55.385229]: ClientConnection::start_tcp_write_queued_response - Sending TCP response
[2016-05-29 22:57:56.416617]: ClientConnection::handle_tcp_read_request_body - Read request body on connection0
[2016-05-29 22:57:56.422121]: ClientConnection::start_tcp_write_queued_response - Sending TCP response
[2016-05-29 22:57:56.426623]: ClientConnection::handle_tcp_read_request_body - Read request body on connection0
```

[log_snippet.txt](https://github.com/libusb/libusb/files/288892/log_snippet.txt)

dickens, 2016-05-30:
Well, I think you've stumbled on the problem. You've got one thread doing event handling while another thread calls libusb_exit(). The context and its resources are being cleaned up by one thread while being utilized by another. That's a recipe for disaster...

HipsterSloth, 2016-05-30:
Aha! Good catch. Yeah at first I wasn't sure where that would be. After a bit of digging I discovered I have another chunk of code in the main thread update loop that is:

* Calling libusb_init(&usb_context);
* Iterating through a list of devices
* Calling libusb_exit()

It's an iterator looking for new attached cameras. I has assumed that libusb keeps a counter of the number of times libusb_init() and libusb_exit are called and it would no-op when libusb was already initialized, that or you could have multiple parallel contexts. 

So it sounds then like I need to make sure that I only call libusb_init and libusb_exit once in the application and share the single usb_context amongst everyone making libusb calls? If this is the case, how do people handle other libraries opening libusb and not exposing the context they are using? Is that considered bad practice?

dickens, 2016-05-30:
Is usb_context global and reused? From the looks of the log messages, I'm guessing so...

HipsterSloth, 2016-05-30:
No in both cases they should have there own context:

The camera enumerator:
```
TrackerDeviceEnumerator::TrackerDeviceEnumerator(CommonDeviceState::eDeviceType deviceType)
    : DeviceEnumerator(deviceType)
    , devs(nullptr)
    , cur_dev(nullptr)
    , dev_index(0)
    , dev_count(0)
    , camera_index(-1)
    , dev_valid(false)
{
    libusb_init(&usb_context);
    libusb_set_debug(usb_context, LIBUSB_LOG_LEVEL_DEBUG);
    dev_count = static_cast<int>(libusb_get_device_list(usb_context, &devs));
    cur_dev = (devs != nullptr) ? devs[0] : nullptr;

    ...
}

TrackerDeviceEnumerator::~TrackerDeviceEnumerator()
{
    if (devs != nullptr)
    {
        libusb_free_device_list(devs, 1);
    }

    libusb_exit(usb_context);
}
```

And the USB manager in PS3EYEDriver (where the crash is occuring in):

```
USBMgr::USBMgr() :
    thread_started(false),
	exit_signaled({ false }),
	active_camera_count({ 0 }),
    control_transfer(nullptr),
    control_transfer_status({ CONTROL_TRANSFER_EMPTY })
{
    libusb_init(&usb_context);
    libusb_set_debug(usb_context, LIBUSB_LOG_LEVEL_DEBUG);
}

USBMgr::~USBMgr()
{
    debug("USBMgr destructor\n");
    libusb_exit(usb_context);
}
```

If I understand the [context docs](http://libusb.sourceforge.net/api-1.0/contexts.html) because I'm using separate contexts the call to exit in the tracker iterator shouldn't effect the one in USBMgr correct?

dickens, 2016-05-30:
Correct, independent contexts are not affected by each other (except for the first context which becomes the default context...there are some caveats).

From the log, I would have expected the last line before the crash to be "checking fd X with revents = XXXX" since that's where the exception occurred.

HipsterSloth, 2016-05-30:
I just did a tweak to make sure that we weren't calling libusb exit from the TrackerDeviceEnumerator once the camera thread started (only run the iterator once). Unfortunately, I was able to repro the crash the same way. 

As for the logs, it turns out the debug logging enabled wasn't getting emitted from usb context in the video frame bulk transfer thread, which is why you weren't seeing the expected logs. I turned that on, but now it appears the main thread is getting deadlocked on a mutex outside of libusb (related to the video frame queue). 

Probably the best thing for me to do at this point is:
* Look over the libusb docs and make sure our usage of libusb is correct
* See if I can make a smaller test case program that reproduces the issue. There are too many other systems that interfere with my crash that it makes it difficult to get you the info you need to diagnose this.

Thanks a ton for your time helping me try and diagnose this! As soon as I have some reliable logs to  diagnose (or I figure out this is something funky we're doing) I'll let you know.

mcuee, 2016-11-05:
Closing old reports if no further info is provided and this does not seems to be a libusb issue.

closed 2016-11-05 by mcuee