libusb/libusb #188: Does it support Android marshmallow?

jacky830, 2016-05-25:
I am developing an usb stuff using libusb on Android platform.
However, I am fail to use libusb directly.
Even if I modify it to accept file descriptor and usb device path, I still no luck to use it without any problems.
I would like to know whether the latest libusb supports Android marshmallow?

tryBetter, 2016-05-26:
I use libusb on Android platform too. And it doesn't work for my application. I got the problem while use "libusb_init(NULL)", it return "-99" and then my application was crash. Do you have such a problem?

jacky830, 2016-05-26:
It can be solved by passing file descriptor from java to libusb_init.

tryBetter, 2016-05-27:
I have get file descriptor by use  
`UsbDeviceConnection connection = usbManager.openDevice(device);`
`int descriptor = connection.getFileDescriptor();`

And how can I use it ?
Thanks for your help.

nilstk, 2016-05-31:
Can you provide further hints? What do I need to change in libusb_init in core.c? I don't see where an fd is used in that function.

tryBetter, 2016-06-01:
@nilstk  see  [https://github.com/Gritzman/libusb](https://github.com/Gritzman/libusb) or [https://github.com/martinmarinov/rtl_tcp_andro-](https://github.com/martinmarinov/rtl_tcp_andro-) , both work for me . My app worked on Android L(5.0).

nilstk, 2016-06-02:
I tried the one by Gritzman earlier already, but it did not work on Android 6.01 and was blocked by SEAndroid apparantly. Can anyone confirm that, too? Any solutions?

mcuee, 2016-06-02:
Right now all these file descriptor changes are just hacks and are not
supported by the libusb project. So I think unless someone really works on
this topic and submit an good patch and discussed the patch in the libusb
mailing list and hopefully get the patches officially included in the
libusb git master, I will think it is better to avoid using libusb in
Android (unless you can root the device).

jacky830, 2016-06-10:
Ultimately, I give up adopting libusb on Android 6.0.1 after a number of tryings.
I am now using Android native USB host to make usb connections:
https://developer.android.com/guide/topics/connectivity/usb/host.html

I think libusb is no longer workable on/above Android 6.0.1.

mcuee, 2016-06-12:
The situation for libusb under Android is a bit unfortunate (nobody came up
to libusb mailing list and offer to help to sort out the issue). Before
that is sorted out, I guess to go back to the Android native USB Host API
is the way to go.

sonicdebris, 2016-07-25:
Ironically, the 1.0.9 version of the library works ok with marshmallow. I tried debugging for a while and I noticed various things have changed in the init code (related to devices detection and enumeration) and that's where the latest version seems to be failing with android 6.

kennethtang4, 2016-11-22:
I can confirm that libusb is working with Marshmallow but not Android 7.1 Nougat

mcuee, 2016-11-25:
Which libusb 1.0.20? The unmodified libusb release 1.0.20 or the hacked version?

mcuee, 2016-11-25:
To sonicdebris, when you say 1.0.9 version, which 1.0.9 version? I believe you mean to say some hacked version from the web and not the original release version of 1.0.9.

mcuee, 2016-12-29:
For those Android users, you may want to try Pull Request #242 to see if that helps.

sonicdebris, 2017-02-17:
@mcuee, sorry I have been imprecise, I meant the 1.0.9 version with the opportune modifications needed to make it accept a file descriptor.

mcuee, 2017-02-18:
Thanks for the updates. Please try Pull Request #242 to see if that helps and then provide comments there.

sonicdebris, 2017-02-18:
Saw that, thanks. I am actually trying another approach: using libusbhost sources from the aosp repos, basically a stripped down version of libusb used by the android framework. I put together a small test app / proof of concept here: https://github.com/sonicdebris/android-native-usb-test

hamza1592, 2017-04-18:
I don't know where to ask this question but since this is an android thread for libusb might as well ask here. Is it possible to turn usb power on and off using libusb for a usb connected through OTG?

matswebjorn, 2017-05-21:
Please have a look at [#301](https://github.com/libusb/libusb/issues/301). I've digged under the hood and found one reason why libusb_init() fails

akshaytaru2007, 2017-06-02:
Hi All,
First it is not issue, I am asking for help. I am trying to connect a Hardware(IoT based) via USB to Android. USB Manager not detecting the devices and not listing the device. The Android is non-rooted, 6.1 It does support OTG. 
As all of you successfully used libusb in Android, I am still struggling as there is no enough documentation or support. 
I have built the .SO file using NDK and added those under jniLibs/AIBs/libusb1.0.SO

Now in my native-lib.cpp I am trying to use libusb's functions to init or get list of device. But I am failing. 
I belive there must be some configuration needs to be done. 
Those who have succeeded kindly help. 

I have also posted Question on stackoverflow related to this: https://stackoverflow.com/q/44281689/1468354 

If you need any more info. I will provide

jwoolston, 2018-06-22:
I can confirm that the changes from #242 will work all the way through Android 8.1. I am currently working on a wrapper library for it at https://github.com/jwoolston/Android-Libusb-Wrapper. Snapshots are deploying but I have not pushed a release yet.

mcuee, 2020-01-06:
Close this for now.

closed 2020-01-06 by mcuee