libusb/libusb #191: Android 6.01 not supported by libUSB -  libusb_init -99

nilstk, 2016-06-01:
Hello everyone,
on Android 6.01 I'm getting error -99, which seems to be caused by SEAndroid, since it's enforced now.
Setting setenforce to 0 with root rights fixes the issue for me, but not for customers, of course.

Is there any solution for non-rooted devices? Does it need light modifications only? Sample source code? Thanks.

mcuee, 2016-06-12:
As mentioned in the other issue #188 , the situation for libusb under Android is a bit unfortunate (nobody came up to libusb mailing list and offer to help to sort out the issue). Before that is sorted out, I guess to go back to the Android native USB Host API is the way to go.

mcuee, 2016-06-12:
If you want, please continue on issue #188. I will close the issue here.

closed 2016-06-12 by mcuee