libusb/libusb #192: LibUsbDotNet dropping received messages under load

TAGC, 2016-06-09:
Hi, I'm experiencing an issue with LibUsbDotNet in which it drops messages if they're arriving fast, and I'd like to know what I should do to prevent messages from being dropped.

I'd submit this directly to the LibUsbDotNet team, but unfortunately their project site doesn't seem very active, so I probably wouldn't be able to get support directly from them. The issue I have only appears on a Ubuntu VM I'm running in which case LibUsbDotNet would be delegating to libusb, so hopefully I can find support for the issue here.

I've posted a detailed account of the issue I'm experiencing [here](http://stackoverflow.com/questions/37481972/libusbdotnet-dropping-received-messages-under-load).

Essentially, I'm finding that the "Read endpoint" thread isn't polling for received messages fast enough on a Linux VM environment and so the "data received" event isn't firing for all messages that are sent from the device to the PC. I've tried to make the event handling thread as fast as possible (by making it store received messages into a queue while another thread consumes from it), which has helped a lot on Windows but still gives me problems on the Ubuntu VM.

mcuee, 2016-06-12:
Are you running the Ubuntu VM from Windows? What is the VM Software used?
VMs are usually problematic with regard to USB enumeration.

BTW, libusbdotnet is supported under libusb-win32 mailing list (the forum
is deprecated). You can subscribe to libusb-win32 mailing list and ask
there.

TAGC, 2016-06-12:
> Are you running the Ubuntu VM from Windows

I am, yes. Windows 7 64-bit.

> What is the VM Software used?

I'm using VMWare Workstation Pro (I believe version 11 but I'm not at work to check).

> VMs are usually problematic with regard to USB enumeration.

It detects my device via VID/PID correctly and does communicate with it, it just doesn't seem to be performant. Is this something that can be caused by improper USB enumeration?

> You can subscribe to libusb-win32 mailing list and ask there.

Cheers.

mcuee, 2016-09-20:
Close this issue as it is not really related to libusb project.

closed 2016-09-20 by mcuee