libusb/libusb #196: Undesired Data Buffering On libusb_open

dsanders11, 2016-07-20:
**OS**: Windows 7
**libusb Version**: 1.0.20
**USB Device**: [Dymo M10 Digital Scale](https://www.amazon.com/DYMO-Digital-Shipping-10-pound-1772057/dp/B0053HCWRE) (VID: 0x922, PID: 0x8003)
* * *

I'm seeing some undesired behavior with interrupt data being buffered from a USB digital scale which I've tracked down to libusb (I believe). From what I can tell with doing USB data captures using USBPcap, the scale is constantly sending an interrupt URB with the latest scale information (weight, units, status), about every 200ms from the looks of the capture<sup>[1](https://github.com/libusb/libusb/files/372871/libusb1.pcap.zip)</sup>.

With libusb I'm seeing that this data seems to be buffered as soon as the USB device has been opened using `libusb_open`. This is very obvious if there is any delay between opening the device and attempting to read data from the scale.

I've created a reproduce case using [libusb1](https://github.com/vpelletier/python-libusb1) for Python (thin wrapper around libusb) as it was the easiest way to narrow down the exact call which started the buffering. This narrowing can be accomplished by moving around a large sleep (20 seconds), then starting the script with an object on the scale and removing it after two seconds. If the sleep comes before the call which triggers the buffering, then when the read occurs after 20 seconds it shows a zero value for the scale weight, which is accurate. If you place the sleep after the call which starts buffering, then when the read occurs it shows the weight for the object that is no longer on the scale, even though that data is ~18 seconds old at that point.

I've unraveled what would normally just be a `device.open()` call in `libusb1` into the script inline to help pinpoint the exact call where the buffering begins.

If you set the sleep to a very long value (~103 seconds) then the buffer appears to be exhausted and the oldest data (when the item was on the scale) is flushed. I did some very rough math, but with interrupts every 200ms with 8 bytes of data that's about 4k bytes as an estimated size of the offending buffer.

I've tested another Python USB library for comparison, and found that [pywinusb](https://github.com/rene-aguirre/pywinusb) does not suffer from this problem, even when putting a sleep after opening the device. For comparison that library is using the HID interface.

Is this a bug, or expected behavior? If expected behavior, is it possible to disable? Reading old values from the scale is strange and makes the processing speed of the computer very important. It's possible to skirt the problem by constantly reading data from the device and trying to keep up, but if the machine can't keep up with the data every 200ms then it will fall woefully behind and any attempt to read the scale data will give you a very stale value.

````python
import usb1
import libusb1

from ctypes import byref

with usb1.USBContext() as context:
    result = context.getByVendorIDAndProductID(0x0922, 0x8003)
    device_handle = libusb1.libusb_device_handle_p()
    ref = byref(device_handle)
    libusb1.libusb_open(result.device_p, ref)  # CAUSES BUFFERING
    sleep(20)
    handle = usb1.USBDeviceHandle(result._USBDevice__context, device_handle, result)
    result._USBDevice__close_set.add(handle)  # Internal cleanup code
    if handle:
        with handle.claimInterface(0x00):
            print handle.interruptRead(0x82, 8)
````

1. [USB Packet Capture](https://github.com/libusb/libusb/files/372871/libusb1.pcap.zip)

atvvarna, 2016-09-02:
With Dymo m5 on Windows 10... Same issue :(

diabolo38, 2016-09-03:
if the devices is an hid class with interrupt  in report end point then i think issue is more about the window hid driver than libusb.
What  i noticed recently using custom hid device under window7 with default driver is that i had traffic and data request done by "os" on behalf of my application code (seen on usb analyzer ). exactly i had "in"  happening after an "out"  doen by app before app did request for in  , when in was requested it get data imadilaty nto issue a new in request on bus. 
I suspect that this the normal os hid driver  behavior to keep getting data from int ep  (buffer overflow on device) and buffer reports to app.

You may trynot use the  default hid device driver for that use zadig to install  winusb or libusk back-end (replace default hid device can be restore after by unsnatling zadig driver) , with that you should have full control over the device and do not get any buffering. 

Note that if your device  is well hid it could be best to use the native "hid" api not libusb.

dickens, 2017-01-18:
This is not a libusb issue. There are no transfers whatsoever initiated by libusb when libusb_open() is called. For Windows, all this does is open one more handles, depending on the specific driver being used. As @diabolo38 suggested, drivers may do a number of things at the time the handle is created. Whatever traffic you see is likely coming from the device or interface driver and not libusb.

I'll close this issue for now, but feel free to reopen if you find evidence to the contrary.

closed 2017-01-18 by dickens