libusb/libusb #199: libusb on windows error

skxatbj, 2016-07-27:
$ ./listdevs.exe
8086:9ca6 (bus 1, device 1)
046d:c05a (bus 2, device 10) path: 2
04f2:b449 (bus 2, device 4) path: 8
058f:6387 (bus 2, device 16) path: 1
138a:0017 (bus 2, device 3) path: 6
8087:8001 (bus 1, device 2) path: 1
8086:9cb1 (bus 2, device 1)

Thinkpad@THINKPAD-PC /c/skyeyeskx/work/libusb/examples
$ ./xusb.exe -i 058f:6387
Using libusb v1.0.21.11127

Opening device 058F:6387...
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
  Failed.

skxatbj, 2016-07-27:
[ 0.041002] [0000172c] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.041002] [0000172c] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.041002] [0000172c] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.041002] [0000172c] libusb: debug [htab_hash] hash collision ('\\.\USB#VID_138A&PID_0017#B84B868680BF' vs '\\.\USB#VID_058F&PID_6387#6C7E5B6E')
[ 0.042002] [0000172c] libusb: debug [windows_get_device_list] found existing device for session [3FC] (2.3)
[ 0.042002] [0000172c] libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
[ 0.043002] [0000172c] libusb: debug [windows_get_device_list] setting HID interface for [2FE]:
[ 0.043002] [0000172c] libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_046D&PID_C05A#6&31765F77&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.043002] [0000172c] libusb: debug [libusb_get_device_descriptor]
[ 0.043002] [0000172c] libusb: debug [libusb_get_device_descriptor]
[ 0.044003] [0000172c] libusb: debug [libusb_get_device_descriptor]
[ 0.044003] [0000172c] libusb: debug [libusb_get_device_descriptor]
[ 0.045003] [0000172c] libusb: debug [libusb_open] open 2.16
[ 0.045003] [0000172c] libusb: debug [unsupported_open] unsupported API call for 'open' (unrecognized device driver)
[ 0.045003] [0000172c] libusb: debug [libusb_open] open 2.16 returns -12
[ 0.045003] [0000172c] libusb: debug [libusb_unref_device] destroy device 2.10
[ 0.045003] [0000172c] libusb: debug [libusb_unref_device] destroy device 2.4
[ 0.045003] [0000172c] libusb: debug [libusb_unref_device] destroy device 2.16
[ 0.045003] [0000172c] libusb: debug [libusb_unref_device] destroy device 2.3
[ 0.046003] [0000172c] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.046003] [0000172c] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.046003] [0000172c] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.046003] [0000172c] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.046003] [0000172c] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 0.046003] [0000172c] libusb: debug [libusb_exit]
[ 0.046003] [0000172c] libusb: debug [libusb_exit] destroying default context
[ 0.047003] [0000172c] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.051003] [00001c74] libusb: debug [windows_clock_gettime_threaded] timer thread quitting

skxatbj, 2016-07-27:
myusb driver is: C:\Windows\system32\drivers\USBSTOR.SYS

pbatard, 2016-07-27:
> myusb driver is: `C:\Windows\system32\drivers\USBSTOR.SYS`

Which is __NOT__ supported by libusb.
You MUST install a supported libusb driver, such as WinUSB, for your device to be accessible with libusb. This is also insicated by the following line:
```
[ 0.045003] [0000172c] libusb: debug [unsupported_open] unsupported API call for 'open' (unrecognized device driver)
```
`USBSTOR.SYS` is not one of the drivers libusb can support.

Please see [this](https://github.com/libusb/libusb/wiki/Windows#Driver_Installation), bearing in mind that that once you replace your driver with WinUSB, you will no longer be able to access your flash drive from Windows Explorer.

If you aren't able to access your device with libusb after installing WinUSB as its driver, feel free to reopen this issue.

closed 2016-07-27 by pbatard

skxatbj, 2016-07-27:
Thanks, I scaned the source code and found in windows_winusb.c I saw windows backend support drivers，but I want to know why not to support USBSTOR.SYS? I think this driver is used widely。

pbatard, 2016-07-27:
> but I want to know why not to support USBSTOR.SYS? I think this driver is used widely

libusb is a __generic__ USB library. Therefore we can only support drivers that offer an API that offers generic USB access. `USBSTOR.SYS` is no such driver. It is a Microsoft driver that is aimed only at providing disk access, so it doesn't not provide any means of doing things such as retrieving device descriptors or issuing a generic control request.

In other words, libusb is not providing support for `USBSTOR.SYS` not because we overlooked it, or don't want to, or because nobody wanted to work on it, but because pretty much all of libusb's API call would have to return "unsupported" due to the Microsoft API for `USBSTOR.SYS` being way too limited for anything but issuing very specific disk requests... in which case you're better of using the Microsoft disk API calls directly, and not libusb.

skxatbj, 2016-07-27:
Thank you for your patience answer. I'm doing a simulator same as virtualbox and I want to get usb which is pluged on host PC from virtual machine, I found libusb is used on qemu which run on linux, but libusb on windows seems not support very good.

shuffle2, 2016-09-06:
Hi, I have been having nearly the same problem as @skxatbj (it would be nice to hear if they managed to solve it).  
In my case, I am trying to use libusb with my keyboard of VID/PID 0x05ac/0x0220 (apple keyboard). Judging from Windows API documentation, it did seem like it would be usable with libusb via the HID backend. However, I did not have luck with that so I went to the last resort of replacing the default HidUsb driver with alternatives (I've tried WinUSB and libusb-win32 so far, using Zadig).  
This still does not work. Here is my current full log from attempting to `libusb_open_device_with_vid_pid`, please let me know if you'd like to see more: http://pastebin.com/raw/M6kNKA7U (yes i've experimented with both endpoints :p)  
I'm on Windows 10 (1607 14393.105)

shuffle2, 2016-09-06:
btw here is a libusb log under normal drivers (HidUsb): http://pastebin.com/raw/Un325LYe  
and log of same config from Zadig: http://pastebin.com/raw/MVVsTGpW