libusb/libusb #200: libusb_bulk_transfer read misses byte

jacky830, 2016-07-28:
Hi All,
I am using libusb_bulk_transfer  to read from serial. Here is the detail:
devh - initial somewhere
mReadEndpoint - initial somewhere as read end point
int readAmt=8;
unsigned char mReadBuffer[readAmt];
int timeoutMillis=50;
libusb_bulk_transfer(devh, mReadEndpoint, mReadBuffer, readAmt, &numBytesRead, timeoutMillis);

It works for reading. However, in some cases, it fails to read one byte data (i.e. ACK byte). I am sure I grabbed the data bytes even it return timeout code, ie. -7. I have no idea why it happens.

It seems to work if I increase the timeout value from 50 to 60, but seems not a good way to solve it.

Does anyone meet this problem and know how to solove this problem?

Thanks.

Jacky

diabolo38, 2016-07-28:
if device is cdc/acm compliant why no to use the os driver ? 

When reading  from bulk end point your request  buffer should be at least the end point max transfer size (64 byte FS or 512 HS 1024SS speed device)  not 8 . Partial read will lead to probem , and if device send less data don't worry the transfer will end on last byte sent before time out.

50ms even 60ms time out doesn't  look as a big value to me so , if increasing time out work then just use bigger value.

 i'm scared that is not a  libusb  issue.

jacky830, 2016-07-29:
The reason why not to use native usb host is that, I have some legacy devices which do not support native usb host. So, I need to root the devices and use libusb.

jacky830, 2016-07-29:
Any advise to me if it is not a libusb issue?

diabolo38, 2016-07-29:
i have use libusb to handle cdc/acm in custom way and quite few other bulk device under window/mac/linux  it work well if you do things right.
so i'm quite confident it is not libusb issue 
try following this two basic : use correct bulk request size (512 byte), use a longer time out (200ms)
you can instrument your code to time the bulk transfer and see if this impact result  but i bet it will not.

jacky830, 2016-07-29:
I also implemented it with cdc/acm, and I still got the same result with the cdc implementataion.
This is what I have implemented for cdc using libusb:

int cdc_acm_read(unsigned char *dest, int length, int timeoutMillis)
{
	int numBytesRead=0;
	int readAmt, ret;

	readAmt = MIN(length, BUFSIZE);
    ret = libusb_bulk_transfer(devh_cdc, mReadEndpoint, mReadBuffer, readAmt, &numBytesRead, timeoutMillis);
    if (ret < 0) {
        if (numBytesRead<=0) {
            return 0;
        }
    }
    memcpy(dest, mReadBuffer, numBytesRead);
    return numBytesRead;
}

jacky830, 2016-07-29:
libusb_bulk_transfer is a blocking function. If I use longer timeout, it blocks my main thread operation since I haven't used separated thread for libusb_bulk_transfer.

diabolo38, 2016-07-29:
if you can offord to wait more then use asynchronous transfer api or create a new thread
this code still read less than a bulk packet size what will soon pr late create ssue or data loss
readAmt = MIN(length, BUFSIZE);
=> this is != at least 512 or 64 HS/FS

jacky830, 2016-07-29:
I would like to try having two pthreads for libusb_bulk_transfer (read and write), and set the timeout to 0.
Is it possible to do read/write in two separate threads?

diabolo38, 2016-07-29:
yes 2 thread shall work  but 0 time out is not reasonable and may not work you just burn cpu time for nothing.  it will not make you get the data any sooner than the  device is done sending the data to your "request". if you cancel your request (0 time out) even before device had a chance to give you the datat how can it work ? (usb is host driven device don"t send data on their iown  you have to ask it ). 
If you  use 1sec  or 10 sec  tim eout ,  call will end when device will be dine sendinf all data is  have with them  a tiem you ask that want most serial adapter do (so it is much less than what you asked in the end ) 

please try to long enoutgh time out with 512 buffer , threaded or assynch API  up to you , in all the case you will get data when they will be ready (also be ready to put several read request until you do not get all data you need devcie may brake it in mutiple xfer )
if you do not know about the usb lower level basic and do "crazy thing" it's not going to work any well ;)

jacky830, 2016-07-30:
In my case the timeout value set to zero means INFINITY timeout. I use one thread to read for a few bytes for each looping; other thread uses for writing purpose. I got the app crashes when the read thread calls libusb_bulk_transfer.
If I can solve this issue I believe my implementation would works.
However, I don‘t have any idea on the crashes at this moment.

jacky830, 2016-08-02:
The method I am using now is pthread to call libusb_bulk_transfer.
I initial 
    pthread_create();

Then, call libusb_bulk_transfer within a while loop, i.e.
ret = libusb_bulk_transfer(dev_handle, ireadEndpoint, readBuf, readAmt, &numBytesRead, 1000);

I would like to know is it a normal to do libusb_bulk_transfer?

mcuee, 2016-11-13:
No further info from the user and this is not really a libusb issue but rather a support question. It is better to discuss this further in the libusb mailing list. I will close this issue here.

closed 2016-11-13 by mcuee