libusb/libusb #202: USB disconnect during bulk transmission

elfmedy, 2016-08-01:
Hi,i use libusb to send and receive data from an usb3.0 slave device. Linux host probably disconnect or timeout(20S) after a period time.
libusb-1.0.21-rc2 is better than libusb-1.0.9, but still can occur. host linux version is 3.4.2.

dmesg is below

```

[   96.856542] irq 54: nobody cared (try booting with the "irqpoll" option)

[   96.863257] CPU: 0 PID: 843 Comm: usb_main Tainted: P           O   3.18.20 #2

[   96.870503] [<c00147d4>] (unwind_backtrace) from [<c0011990>] (show_stack+0x10/0x14)

[   96.878250] [<c0011990>] (show_stack) from [<c0480cf8>] (dump_stack+0x88/0x98)

[   96.885474] [<c0480cf8>] (dump_stack) from [<c0085f7c>] (__report_bad_irq+0x28/0xc4)

[   96.893216] [<c0085f7c>] (__report_bad_irq) from [<c008651c>] (note_interrupt+0x270/0x2c0)

[   96.901479] [<c008651c>] (note_interrupt) from [<c00840d4>] (handle_irq_event_percpu+0xcc/0x134)

[   96.910262] [<c00840d4>] (handle_irq_event_percpu) from [<c0084180>] (handle_irq_event+0x44/0x64)

[   96.919132] [<c0084180>] (handle_irq_event) from [<c0086b24>] (handle_fasteoi_irq+0xac/0x18c)

[   96.927655] [<c0086b24>] (handle_fasteoi_irq) from [<c00837b0>] (generic_handle_irq+0x2c/0x3c)

[   96.936263] [<c00837b0>] (generic_handle_irq) from [<c0083a48>] (__handle_domain_irq+0x5c/0xa4)

[   96.944958] [<c0083a48>] (__handle_domain_irq) from [<c00085d0>] (gic_handle_irq+0x24/0x5c)

[   96.953307] [<c00085d0>] (gic_handle_irq) from [<c0012440>] (__irq_svc+0x40/0x54)

[   96.960781] Exception stack(0xca671d38 to 0xca671d80)

[   96.965826] 1d20:                                                       00000001 c0816980

[   96.973999] 1d40: 00000000 00000000 00000282 00000000 00000000 00000001 ca670000 00000000

[   96.982172] 1d60: c084aeec cf476dc0 00400040 ca671d80 c0816980 c0040190 60000113 ffffffff

[   96.990352] [<c0012440>] (__irq_svc) from [<c0040190>] (__do_softirq+0xd4/0x2a4)

[   96.997747] [<c0040190>] (__do_softirq) from [<c004062c>] (irq_exit+0xc8/0x130)

[   97.005055] [<c004062c>] (irq_exit) from [<c0083a4c>] (__handle_domain_irq+0x60/0xa4)

[   97.012881] [<c0083a4c>] (__handle_domain_irq) from [<c00085d0>] (gic_handle_irq+0x24/0x5c)

[   97.021230] [<c00085d0>] (gic_handle_irq) from [<c0012440>] (__irq_svc+0x40/0x54)

[   97.028703] Exception stack(0xca671e10 to 0xca671e58)

[   97.033749] 1e00:                                     ceca1360 009fe410 00002c20 00000000

[   97.041921] 1e20: 00000000 00000000 00000000 00000000 00000000 00000000 c084aeec cf476dc0

[   97.050092] 1e40: 00000000 ca671e5c 00000000 c021af18 20000013 ffffffff

[   97.056710] [<c0012440>] (__irq_svc) from [<c021af18>] (__copy_from_user+0x4c/0x3b4)

[   97.064443] handlers:

[   97.066714] [<c02ddf00>] usb_hcd_irq

[   97.070286] Disabling IRQ #54

[  103.623891] xhci-hcd 10180000.xhci: xHCI host not responding to stop endpoint command.

[  103.631809] xhci-hcd 10180000.xhci: Assuming host is dying, halting host.

[  103.658352] xhci-hcd 10180000.xhci: Host not halted after 16000 microseconds.

[  103.665494] xhci-hcd 10180000.xhci: Non-responsive xHCI host is not halting.

[  103.672531] xhci-hcd 10180000.xhci: Completing active URBs anyway.

[  103.679090] xhci-hcd 10180000.xhci: HC died; cleaning up

[  103.688775] usb 2-1: USB disconnect, device number 2
```

or

```
[ 9816.027719] irq 54: nobody cared (try booting with the "irqpoll" option)

[ 9816.034434] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           O   3.18.20 #2

[ 9816.041594] [<c00147d4>] (unwind_backtrace) from [<c0011990>] (show_stack+0x10/0x14)

[ 9816.049339] [<c0011990>] (show_stack) from [<c0480cf8>] (dump_stack+0x88/0x98)

[ 9816.056564] [<c0480cf8>] (dump_stack) from [<c0085f7c>] (__report_bad_irq+0x28/0xc4)

[ 9816.064307] [<c0085f7c>] (__report_bad_irq) from [<c008651c>] (note_interrupt+0x270/0x2c0)

[ 9816.072570] [<c008651c>] (note_interrupt) from [<c00840d4>] (handle_irq_event_percpu+0xcc/0x134)

[ 9816.081353] [<c00840d4>] (handle_irq_event_percpu) from [<c0084180>] (handle_irq_event+0x44/0x64)

[ 9816.090222] [<c0084180>] (handle_irq_event) from [<c0086b24>] (handle_fasteoi_irq+0xac/0x18c)

[ 9816.098744] [<c0086b24>] (handle_fasteoi_irq) from [<c00837b0>] (generic_handle_irq+0x2c/0x3c)

[ 9816.107352] [<c00837b0>] (generic_handle_irq) from [<c0083a48>] (__handle_domain_irq+0x5c/0xa4)

[ 9816.116047] [<c0083a48>] (__handle_domain_irq) from [<c00085d0>] (gic_handle_irq+0x24/0x5c)

[ 9816.124396] [<c00085d0>] (gic_handle_irq) from [<c0012440>] (__irq_svc+0x40/0x54)

[ 9816.131870] Exception stack(0xc07d5f68 to 0xc07d5fb0)

[ 9816.136918] 5f60:                   cfddd210 00000000 0097dd5a c0019600 c07d4038 c07dd4a0

[ 9816.145091] 5f80: c07dd504 c0488744 c0812506 c0812506 c07d4000 00000000 00000000 c07d5fb0

[ 9816.153261] 5fa0: c000f104 c000f108 60000013 ffffffff

[ 9816.158314] [<c0012440>] (__irq_svc) from [<c000f108>] (arch_cpu_idle+0x38/0x3c)

[ 9816.165709] [<c000f108>] (arch_cpu_idle) from [<c0070994>] (cpu_startup_entry+0x108/0x15c)

[ 9816.173975] [<c0070994>] (cpu_startup_entry) from [<c05e3c70>] (start_kernel+0x3b0/0x3bc)

[ 9816.182141] handlers:

[ 9816.184412] [<c02ddf00>] usb_hcd_irq

[ 9816.187984] Disabling IRQ #54

[ 9840.884047] xhci-hcd 10180000.xhci: xHCI host not responding to stop endpoint command.

[ 9840.891961] xhci-hcd 10180000.xhci: Assuming host is dying, halting host.

[ 9840.918474] xhci-hcd 10180000.xhci: Host not halted after 16000 microseconds.

[ 9840.925597] xhci-hcd 10180000.xhci: Non-responsive xHCI host is not halting.

[ 9840.932634] xhci-hcd 10180000.xhci: Completing active URBs anyway.

[ 9840.938833] xhci-hcd 10180000.xhci: HC died; cleaning up
```

Some times “irq err” is miss but xchi still die.
I'm not sure this is the problem of kernel or libusb

diabolo38, 2016-08-01:
i have seen some xhci divers issues on linux 3.3 up to 3.6  but  not like this one

more recently i did face issue on recent intel host controller ( win7) and SS device due to link power  management (LPM) some device phy weakness were causing it to disconnect  on link training failure (once in while). 
Just in case you may try disable "lpm" or usb power management feature if anyin  your xhci host controler see if it can help.
as lookign at lpm control if find this post http://www.linuxquestions.org/questions/slackware-14/xhci-hcd-usb3-issues-with-new-motherboard-4175552632/
some of the message appear the same as your 

up to me the issue is not on libusb side but more in h/w or xhci driver

mcuee, 2016-09-20:
Closing this issue as this seems to be related to kernel xhci driver and not libusb.

closed 2016-09-20 by mcuee