libusb/libusb #206: program assertion failed - existing device should share parent

ecejeff, 2016-09-08:
Every time I get a list of USB devices through libusb on windows I see a bunch of log messages like these:

```
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
```

Everything seems to be working correctly. The wall of error messages is annoying though.

This problem was introduced after the last stable release, but exists in 1.0.21-rc3.

mcuee, 2016-09-10:
Maybe you want to post the full debug message (you need to set
LIBUSB_DEBUG=4) to see what is the offending device.

ecejeff, 2016-09-10:
I already posted this on the mailing list about 6 months ago with no response. Here it is again:

listdevs.exe with LIBUSB_DEBUG=4

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000000] [000026bc] libusb: debug [libusb_init] created default context
[ 0.000000] [000026bc] libusb: debug [libusb_init] libusb v1.0.20.11106
[ 0.000000] [000026bc] libusb: debug [windows_init] Windows 7 SP1 32 bit
[ 0.000000] [000026bc] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.007000] [000026bc] libusb: info [winusbx_init] libusbK DLL is not available, will use native WinUSB
[ 0.008000] [000026bc] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.008000] [000026bc] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.008000] [000026bc] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.008000] [000026bc] libusb: debug [windows_init_clock] hires timer available (Frequency: 3914150 Hz)
[ 0.008000] [000026bc] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.009000] [000026bc] libusb: debug [htab_create] using 1021 entries hash table
[ 0.009000] [000026bc] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.009000] [000026bc] libusb: debug [libusb_get_device_list] 
[ 0.012000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [1E7]
[ 0.012000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [272]
[ 0.017000] [000026bc] libusb: debug [get_api_type] driver(s): ausb3hub
[ 0.017000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [33D]
[ 0.017000] [000026bc] libusb: debug [get_api_type] driver(s): iusb3hub
[ 0.017000] [000026bc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.017000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [24C]
[ 0.018000] [000026bc] libusb: debug [get_api_type] driver(s): iusb3hub
[ 0.018000] [000026bc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.018000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [3ED]
[ 0.019000] [000026bc] libusb: debug [get_api_type] driver(s): iusb3hub
[ 0.019000] [000026bc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.019000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [240]
[ 0.019000] [000026bc] libusb: debug [get_api_type] driver(s): iusb3hub
[ 0.019000] [000026bc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.019000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [2F6]
[ 0.020000] [000026bc] libusb: debug [get_api_type] driver(s): iusb3hub
[ 0.020000] [000026bc] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.020000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [88]
[ 0.023000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [B9]
[ 0.024000] [000026bc] libusb: debug [init_device] got bus number from ancestor #4
[ 0.024000] [000026bc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.024000] [000026bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 46 bytes)
[ 0.024000] [000026bc] libusb: debug [init_device] (bus: 2, addr: 15, depth: 1, port: 4): '\\.\USB#VID_03EB&PID_2105#7&129AFCD5&0&4'
[ 0.024000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [37A]
[ 0.024000] [000026bc] libusb: debug [init_device] got bus number from ancestor #3
[ 0.025000] [000026bc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.025000] [000026bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 2466 bytes)
[ 0.025000] [000026bc] libusb: debug [init_device] (bus: 2, addr: 8, depth: 1, port: 2): '\\.\USB#VID_046D&PID_0825#197F4800'
[ 0.026000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [2D5]
[ 0.026000] [000026bc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.026000] [000026bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.027000] [000026bc] libusb: debug [init_device] (bus: 2, addr: 12, depth: 1, port: 3): '\\.\USB#VID_046D&PID_C069#7&129AFCD5&0&3'
[ 0.027000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [E9]
[ 0.027000] [000026bc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.027000] [000026bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.027000] [000026bc] libusb: debug [init_device] (bus: 2, addr: 11, depth: 1, port: 2): '\\.\USB#VID_046D&PID_C626#7&129AFCD5&0&2'
[ 0.028000] [000026bc] libusb: debug [windows_get_device_list] allocating new device for session [89]
[ 0.028000] [000026bc] libusb: debug [init_device] got bus number from ancestor #2
[ 0.028000] [000026bc] libusb: debug [init_device] found 1 configurations (active conf: 1)
[ 0.028000] [000026bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[... cut ...]
[ 0.063000] [000026bc] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.063000] [000026bc] libusb: debug [libusb_unref_device] destroy device 2.0
[ 0.063000] [000026bc] libusb: debug [libusb_exit] 
[ 0.063000] [000026bc] libusb: debug [libusb_exit] destroying default context
[ 0.064000] [000026bc] libusb: debug [usbi_remove_pollfd] remove fd 0
[ 0.064000] [00002b6c] libusb: debug [windows_clock_gettime_threaded] timer thread quitting
```

This was taken on my desktop computer. My laptop (also Win 7 x64) has a similar output.

mcuee, 2016-09-10:
I believe this has been in the Windows backend for a long time and it maybe
this should be downgraded to a warning or similar.

mcuee, 2016-09-10:
Are you saying 1.0.20 is okay but 1.0.21-rc3 has the above warnings?

ecejeff, 2016-09-10:
Yes. I wanted to make sure that someone at least acknowledged the issue before releasing 1.0.21.

hjelmn, 2016-09-13:
Is this an existing bug or a regression?

-Nathan

dickens, 2016-09-13:
This is a new "artifact" introduced by 1d54ee1b. It is correctly pointing
out an inconsistency (that has existing since the beginning of the Windows
backend), but the message is a bit misleading. The root cause is due to
differences across systems in the order in which devices are enumerated on
Windows.

The basic problem is this. Suppose there is a device connected to the
system as follows:
HCD <-> Root Hub <-> Hub <-> Device

Windows will first "enumerate" the HCD, creating psuedo libusb_device
structures. When enumerating actual devices and hubs, the order is
non-deterministic (and not necessarily topological). The warning message is
triggered when the Device is enumerated before the Hub. When any device or
hub is enumerated, Windows will follow the ancestor chain to find the
"parent" device. In this example, suppose the Root Hub is enumerated first,
followed by the Device. Windows will calculate the session ID for the Hub
and attempt to look up the libusb_device, but as it has not yet been
enumerated this will fail, so it proceeds up the tree to the Root Hub and
successfully finds a libusb_device for it. The Root Hub then becomes the
Device's "parent" device. Later on, the Hub is finally enumerated. During
later phases of enumeration, Windows will enumerate specific interfaces for
Device. When it does so, it will again perform a parent device lookup, but
this time it will actually find a libusb_device for the Hub. As the warning
suggests, the parent device found (Hub) doesn't match the assigned parent
device (Root Hub).

I have a patch that should fix this. I vote for including this in the
release, as this will probably annoy quite a few Windows users. I will
thoroughly test the patch and attach it tomorrow. @ecejeff, would you be
able to test this?

ecejeff, 2016-09-13:
Yes, I'm traveling later this week but I'll make a point to test it, because, like you said, it'll annoy Windows users if it isn't included in the next release.

dickens, 2016-09-14:
Attached is a patch which should address this issue. Thanks in advance for testing it!

[windows-error-msg.patch.zip](https://github.com/libusb/libusb/files/471327/windows-error-msg.patch.zip)

ecejeff, 2016-09-14:
Listdevs no longer shows the error, but in my program I get the following:

```
libusb: error [windows_transfer_callback] detected I/O error 1167: [1167] The device is not connected.
libusb: error [cache_config_descriptors] could not access configuration descriptor (actual) for '\\.\USB#VID_0BDA&PID_570C#200901010001': [995] The I/O operation has been aborted because of either a thread exit or an application request.
libusb: info [cache_config_descriptors] could not access configuration descriptor (dummy) for '\\.\USB#VID_0BDA&PID_570C#200901010001': [995] The I/O operation has been aborted because of either a thread exit or an application request.
```

In my application (unit tests) I do some disconnecting and reconnecting, and that's when I see this.

I don't know what 0bda:570c is, but it's not one of the devices I have control over.

Everything still seems to be working though.

commit 9bf44223 (2016-10-01): Windows: Fix possible false error message during device enumeration

During early phases of enumeration, it is possible for a device to be
enumerated before its parent. When this occurs, the device assigned as
the parent will actually be a grandparent. During later phases of
enumeration, the parent device will in fact exist but will not match
what has already been assigned to the device.

This commit adds code to check for and update the parent device when
this situation occurs.

Closes #206
closed 2016-10-01 by hjelmn