libusb/libusb #207: Error C2001: Newline in constant

Grix, 2016-09-10:
I get this error when trying to build the unmodified libusb_2013.sln VS2013 solution included in your releases. 

Error C2001: Newline in constant

It refers to line 111 of strerror.c, which is part of the russian error message translations. I can't get the library to build without deleting russian from the locale arrays. Not sure if there is a syntax error you missed if or if there is some unicode problems caused by the russian characters on my system.

LudovicRousseau, 2016-09-11:
It looks like the libusb_2013.sln solution is working fine on the Continuous Integration system: https://ci.appveyor.com/project/LudovicRousseau/libusb

Maybe it is a problem on your side?

PS: I am not a Windows expert or even user.

Grix, 2016-09-11:
>Maybe it is a problem on your side?

Quite possibly. I wouldn't know how to fix it though

Grix, 2016-09-20:
It is caused by my system locale it seems. If I change locale to english (US) it works. But my usual locale, Japanese, results in the error. You should  be able to reproduce it if you change locale to Japanese and try building the project.

LudovicRousseau, 2016-09-21:
The file https://github.com/libusb/libusb/blob/master/libusb/strerror.c#L111 looks like a valid UTF-8 encoded file.
It is strange the compiler complains about the Cyrillic text but not about the French accents in the section just before.

It looks like VS2013 does not handle UTF-8 correctly. Maybe you should report the problem to Microsoft.

undisputed-seraphim, 2016-12-30:
It fails to compile on VS2015 as well.

However, I have found that it can be very easily resolved simply by putting one space at the back of the two offending strings: [Lines 111](https://github.com/libusb/libusb/blob/master/libusb/strerror.c#L111) and 117.

Kindly consider my suggestion or inform me if you prefer a pull request. This should be a rather painless solution.

mcuee, 2016-12-31:
What is your system locale? If you use English locale, it should be okay. 
For your solution, have you tested using a few different locale? Say the Japanese locale the original poster mentioned. Thanks.

undisputed-seraphim, 2017-01-03:
I'm on the Japanese locale as well.

NoviceLive, 2017-03-23:
Hope this comment helps.

Environment: Windows 8.1 + Visual Studio 2015, with Chinese being the system locale and UTF-8 being the source code encoding.

My Visual C++ projects failed to compile with the same error, i.e. `Newline in constant`, upon some (**not all**) Chinese string literals, after migrating  source encoding from CP936 (the Chinese encoding that Windows defaults to) to UTF-8 **without BOM**.

And they compiled without problem after explicitly telling the compiler that the source encoding is UTF-8, say, adding `/source-charset:utf-8` to `Additional Options` in `C/C++` `Command Line`.

See https://msdn.microsoft.com/en-us/library/mt708819.aspx for more details.

Quote with my emphasis.

> By default, Visual Studio detects a **byte-order mark** to determine if the source file is in an encoded Unicode format, for example, UTF-16 or UTF-8. If **no byte-order mark** is found, it assumes the source file is encoded using the **current user code page**, unless you specify a character set name or code page by using the /source-charset option.

mick-p1982, 2019-03-01:
Ran into this with vcpkg that is a new package framework on Windows. I don't know what the right solution is. But maybe however libusb is using this file, it doesn't have to be a source code file?

suaviant, 2019-10-08:
> And they compiled without problem after explicitly telling the compiler that the source encoding is UTF-8, say, adding `/source-charset:utf-8` to `Additional Options` in `C/C++` `Command Line`.

Thanks. This option helped me compile the cocos2dx tests

dickens, 2020-08-13:
Closing this.  The issue was resolved in the libusb build system by commits 77037c4dd64cd4e28ed5ab1bc0b893c81fb3e20a and de4a59af87bf0b6df04d9f1c9439c957681f5e53

closed 2020-08-13 by dickens