libusb/libusb #208: Device Manager "Group"

arrmo, 2016-09-11:
Hi,

I installed libusb on two different Windows machines. On one, in Device Manager I see a group titled "libusb-win32", and my device (UPS) is inside there - perfect! On the second machine though, the device I applied libusb to shows up in a group of "Jawbone Devices", which seems very odd to me.

Thoughts?

Thanks!

dickens, 2016-09-14:
This is a libusb-win32 issue, not a libusb issue. The project isn't in active development, so I don't know how much support you'll receive.

closed 2016-09-14 by dickens

arrmo, 2016-09-14:
NP, understood. But to figure this out (just curious) - there are updates still ongoing to the code ... but not active development? Just confused ... :(.

Thanks!

mcuee, 2016-09-14:
For libusb-win32 support, please use libusb-win32 mailing list. It is a
separate project from libusb.

-- 
Xiaofan

arrmo, 2016-09-14:
Ahhh - there is my confusion. Thanks for clarifying!