libusb/libusb #209: libusb_control_transfer return -9

goodcyg, 2016-09-12:
With the synchronous operation flow is as follows：
1.libusb_init
2.libusb_open_device_with_vid_pid
3.send SCSI command 0x00
   sendUnitTestReady()
   {
     ezusb_write(.....);
   
   }
   
4.static int ezusb_write(libusb_device_handle* device, const char* label,
                       uint8_t opcode, uint32_t addr, const unsigned char* data, size_t len)
{
    int status;
    if (verbose > 1)
        LOG("%s, addr 0x%08x len %4u (0x%04x)\n", label, addr, (unsigned)len, (unsigned)len);
    status = libusb_control_transfer(device,
                                     LIBUSB_ENDPOINT_OUT | LIBUSB_REQUEST_TYPE_VENDOR | LIBUSB_RECIPIENT_DEVICE,
                                     opcode, addr & 0xFFFF, addr >> 16,
                                     (unsigned char*)data, (uint16_t)len, 1000);
    if ((status > 0) && (status == len)) {
        LOG("success:%s ==> %d\n", label, status);
    } else {
        LOG("label:%s: %s:%d\n", label, libusb_error_name(status), status);
    }
    return (status < 0) ? -1 : 0;
}

5.libusb_control_transfer return -9 :LIBUSB_ERROR_PIPE
why？help me!

mcuee, 2016-09-14:
This usually means that the device does not understand the command. Are you
using libusb on a USB mass storage device?

goodcyg, 2016-09-15:
Yes, I was using the U disk, and I was checking to try and thank you.

goodcyg, 2016-09-15:
My code structure is like this:
1.libusb_init(NULL)；
2.libusb_open_device_with_vid_pid
3.  ....
4.libusb_close(handle);
5. libusb_exit(NULL);

I have to return in time, use libusb_open_device_with_vid_pid to open the U value, sometimes null, sometimes is not null, if NULL, again in the run became not null?

goodcyg, 2016-09-15:
My system is mac OS X Yosemite 10.10.5

goodcyg, 2016-09-16:
hello!

My system is mac OS X Yosemite 10.10.5
I send the SCSI command to U disk(flashMass storage) on USB Mac, read and write U disk data. I refer to the xusb.c in the examples to do the corresponding modification, but encountered some problems.
I have three questions:

1)
 My code structure is like this:
1.libusb_init(NULL)；
2.libusb_open_device_with_vid_pid
3. ....
4.libusb_close(handle);
5. libusb_exit(NULL);
I have to return in time, use libusb_open_device_with_vid_pid to open the U value, sometimes null, sometimes is not null, if NULL, again in the run became not null?

2)
r = libusb_control_transfer(handle,
LIBUSB_ENDPOINT_IN | LIBUSB_REQUEST_TYPE_CLASS |
LIBUSB_RECIPIENT_INTERFACE,
BOMS_GET_MAX_LUN, 0, 0, lun, 1, 1000);
return -7 Operation timed out. why?

3)libusb_claim_interface not work
 libusb_claim_interface returns -3

thank you.

原始邮件

mcuee, 2016-09-20:
Your question has already been answered in the mailing list. Please do not raise more tickets. Thanks.

As mentioned in the mailing list, you can not use libusb for a USB Mass Storage Device under Mac OS X since you can not get the exclusive access. So "Claiming interface 0" will fail.

mcuee, 2016-09-20:
Closing this ticket.

closed 2016-09-20 by mcuee

goodcyg, 2016-09-20:
I read the libusb code, in fact, is the iokit code package, os X iokit has SDK OS, but more cumbersome. But it is also possible to access the USB device? Can not get the exclusive access. I how do I access Mass Storage Device under USB Mac OS X? This question puzzled me for a long time? help me

原始邮件

Closed #209.