libusb/libusb #214: xusb.c not work on mac

goodcyg, 2016-09-17:
hello!
My system is mac OS X Yosemite 10.10.5，I modified the VID and PID in the Xcode MAC compiler, run, reported the error, the log is as follows: 

Opening device 2BA3:1220...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 2BA3:1220
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 02
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 82
           max packet size: 0200
          polling interval: 00

Claiming interface 0...
libusb>>USBInterfaceOpen: another process has device opened for exclusive access       libusb: error line:1306 fun:darwin_claim_interface file:darwin_usb.c
   Failed.

libusb>>transfer error: timed out       libusb: warning line:1889 fun:darwin_transfer_status file:darwin_usb.c
Reading Max LUN:
libusb>>transfer error: timed out       libusb: warning line:1889 fun:darwin_transfer_status file:darwin_usb.c
   Failed: Operation timed out   Max LUN = 0
Sending Inquiry:
libusb>>no pipeRef found with endpoint address 0x02.       libusb: warning line:192 fun:ep_to_pipeRef file:darwin_usb.c
libusb>>endpoint not found on any open interface       libusb: error line:1535 fun:submit_bulk_transfer file:darwin_usb.c
   send_mass_storage_command: Entity not found
libusb>>no pipeRef found with endpoint address 0x82.       libusb: warning line:192 fun:ep_to_pipeRef file:darwin_usb.c
libusb>>endpoint not found on any open interface       libusb: error line:1535 fun:submit_bulk_transfer file:darwin_usb.c
   Entity not found
   Input/Output Error
libusb>>application left some devices open       libusb: warning line:2234 fun:libusb_exit file:core.c
Program ended with exit code: 0

This is why?hello!
My system is mac OS X Yosemite 10.10.5，I modified the VID and PID in the Xcode MAC compiler, run, reported the error, the log is as follows: 

Opening device 2BA3:1220...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 2BA3:1220
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 02
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 82
           max packet size: 0200
          polling interval: 00

Claiming interface 0...
libusb>>USBInterfaceOpen: another process has device opened for exclusive access       libusb: error line:1306 fun:darwin_claim_interface file:darwin_usb.c
   Failed.

libusb>>transfer error: timed out       libusb: warning line:1889 fun:darwin_transfer_status file:darwin_usb.c
Reading Max LUN:
libusb>>transfer error: timed out       libusb: warning line:1889 fun:darwin_transfer_status file:darwin_usb.c
   Failed: Operation timed out   Max LUN = 0
Sending Inquiry:
libusb>>no pipeRef found with endpoint address 0x02.       libusb: warning line:192 fun:ep_to_pipeRef file:darwin_usb.c
libusb>>endpoint not found on any open interface       libusb: error line:1535 fun:submit_bulk_transfer file:darwin_usb.c
   send_mass_storage_command: Entity not found
libusb>>no pipeRef found with endpoint address 0x82.       libusb: warning line:192 fun:ep_to_pipeRef file:darwin_usb.c
libusb>>endpoint not found on any open interface       libusb: error line:1535 fun:submit_bulk_transfer file:darwin_usb.c
   Entity not found
   Input/Output Error
libusb>>application left some devices open       libusb: warning line:2234 fun:libusb_exit file:core.c
Program ended with exit code: 0

This is why?hello!
My system is mac OS X Yosemite 10.10.5，I modified the VID and PID in the Xcode MAC compiler, run, reported the error, the log is as follows: 

Opening device 2BA3:1220...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 2BA3:1220
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 02
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 82
           max packet size: 0200
          polling interval: 00

Claiming interface 0...
libusb>>USBInterfaceOpen: another process has device opened for exclusive access       libusb: error line:1306 fun:darwin_claim_interface file:darwin_usb.c
   Failed.

libusb>>transfer error: timed out       libusb: warning line:1889 fun:darwin_transfer_status file:darwin_usb.c
Reading Max LUN:
libusb>>transfer error: timed out       libusb: warning line:1889 fun:darwin_transfer_status file:darwin_usb.c
   Failed: Operation timed out   Max LUN = 0
Sending Inquiry:
libusb>>no pipeRef found with endpoint address 0x02.       libusb: warning line:192 fun:ep_to_pipeRef file:darwin_usb.c
libusb>>endpoint not found on any open interface       libusb: error line:1535 fun:submit_bulk_transfer file:darwin_usb.c
   send_mass_storage_command: Entity not found
libusb>>no pipeRef found with endpoint address 0x82.       libusb: warning line:192 fun:ep_to_pipeRef file:darwin_usb.c
libusb>>endpoint not found on any open interface       libusb: error line:1535 fun:submit_bulk_transfer file:darwin_usb.c
   Entity not found
   Input/Output Error
libusb>>application left some devices open       libusb: warning line:2234 fun:libusb_exit file:core.c
Program ended with exit code: 0

This is why? How to modify?

goodcyg, 2016-09-19:
Raised so many questions, why no one answer it?

mcuee, 2016-09-20:
Your question has already been answered in the mailing list. Please do not
raise more tickets. Thanks.

As mentioned in the mailing list, you can not use libusb for a USB Mass
Storage Device under Mac OS X since you can not get the exclusive access.
So "Claiming interface 0" will fail.

closed 2016-09-20 by mcuee

goodcyg, 2016-09-20:
I read the libusb code, in fact, is the iokit code package, os X iokit has SDK OS, but more cumbersome. But it is also possible to access the USB device? Can not get the exclusive access. I how do I access Mass Storage Device under USB Mac OS X? This question puzzled me for a long time? help me

goodcyg, 2016-09-20:
@mcuee 
I read the libusb code, in fact, is the iokit code package, os X iokit has SDK OS, but more cumbersome. But it is also possible to access the USB device? Can not get the exclusive access. I how do I access Mass Storage Device under USB Mac OS X? This question puzzled me for a long time? help me

goodcyg, 2016-09-20:
re-open