libusb/libusb #215: xusb vid:pid failed on windows10

tinysniper, 2016-09-20:
It failed when using `xusb vid:pid`. And I used zadig to install the winusb driver.
The errors shows like this:

> e: > xusb 0x2837:0x0200
Using libusb v1.0.20.11004
Opening device 2837:0200...
libusb: error [init_device] device '\\.\USB#VID_2837&PID_0200&MI_02#7&1E1FD5AE&0&0002' is no longer connected!
libusb: info [cache_config_descriptors] could not access configuration descriptor (dummy) for '\\.\USB#VID_1A40&PID_0101#6&1662E132&0&3': [31] A device attached to the system is not functioning.
libusb: error [init_device] device '\\.\USB#VID_2837&PID_0200&MI_01#7&1E1FD5AE&0&0001' is no longer connected!
  Failed.

And the debug information is in file xusb-error.txt.
[xusb-error.txt](https://github.com/libusb/libusb/files/481808/xusb-error.txt)

mcuee, 2016-09-20:
Are you using USB 3.0? What is the USB Host Chipset used?

tinysniper, 2016-09-20:
Nope. The chipset is STM32.
Thanks for your reply.

mcuee, 2016-11-03:
I am asking about the host side (PC side) and not the device side (STM32 MCU based).

mcuee, 2016-11-05:
And make sure your device works properly first.

tinysniper, 2016-11-07:
It's USB 2.0.
My device works well when using libusb0(libusb-win32) driver on another PC.

diabolo38, 2016-11-07:
for informatio, i have done some test on win10 (VM) with stm32 devices (cdc and hid base device class) and it was ok  with previous 1.0.21 (llibusbk back end) 
so i would not expect regression here.

Most issue i had so far with libusb were related to usb3.0 host controller or topology .

mcuee, 2016-12-02:
To OP: when you say your device works using libusb-win32 driver on another PC, does your device work with libusb-win32 driver on this Windows 10 PC? I am suspecting that your device has some problem with this PC due to HW/FW problem.

mcuee, 2017-01-07:
This could be a problematic device as I encountered similar error messages under my old Windows laptop.

045e:0773 is the Microsoft Wireless Mouse/Keyboard receiver (USB Composite
Device).
058f:6366 is a USB Card Reader

Ref: http://marc.info/?l=libusb-devel&m=148336155619302&w=2
libusb: error [init_device] device
'\\.\USB#VID_045E&PID_0773&MI_01#6&1315E060&0&0001' is no longer
connected!
libusb: error [init_device] device
'\\.\USB#VID_045E&PID_0773&MI_02#6&1315E060&0&0002' is no longer
connected!
libusb: error [cache_config_descriptors] could not access
configuration descriptor (actual) for
'\\.\USB#VID_058F&PID_6366#058F63666433': [995] The I/O operation has
been aborted because of either a thread exit or an application
request.

closed 2018-01-05 by dickens