libusb/libusb #216: libusb_bulk_transfer bulk read return 0 but buffer is empty

tinysniper, 2016-09-22:
```
int libusb_bulk_transfer	(struct libusb_device_handle * dev_handle,
                             unsigned char 	endpoint,
                             unsigned char * 	data,
                             int 	length,
                             int * 	transferred,
                             unsigned int 	timeout 
)	
```

The libusb_bulk_transfer returns 0(success) while bulk reading. 
And the actual read length `transferred` is the same as `length`. But the buffer readed `data` is empty with `0`. I don't understand what's wrong.
* Using zadig to install WinUSB driver
* libusb0(libusb-win32) works well while reading/writing
* Test is on Windows10

mcuee, 2016-11-03:
Please provide more information. 
You may want to try the latest release version of libusb (1.0.21).
You may want to attach the debug log as well.

tinysniper, 2016-11-07:
My libusb is right the 1.0.21.

mcuee, 2016-11-13:
You need to provide more details, like what is the device, what your program is doing and probably the relevant codes as well.

mcuee, 2016-12-02:
Will close this issue if no more details are provided within another week.

closed 2016-12-31 by mcuee