libusb/libusb #219: c++11 async feature with synchronous I/O

pdaddy26, 2016-10-04:
Hi, 

Is it possible to use the Synchronous I/O inside a c++11 async? 

What I tried, was in my hotplug_callback for the arrived event. I would create an async and do all the synchronous I/O in the async. The reason I'm doing this is to simply the code. I'm aware that libusb offer asynchronous I/O, but the sync I/O seems simpler to use so I thought I could maybe use the synchronous I/O inside an async. 

```
int LIBUSB_CALL hotplug_callback(libusb_context *ctx, libusb_device *dev, libusb_hotplug_event event, void *user_data)
{
  std::future<void> result(std::async([] (libusb_device *d) {

    libusb_device_handle *h;

    printf("%s\n", "Opening");
    int rc = libusb_open(d, &h);
    if(rc != LIBUSB_SUCCESS) {
      printf("ERROR: %s\n", libusb_error_name(rc));
      return;
    }
    printf("%s\n", "Opened");

    printf("%s\n", "Closing");
    libusb_close(h);
    printf("%s\n", "Closed!");

  }, dev));

  result.get();

  return 0;
}
```

But some call would hang ..... not sure why .... 

Any help is appreciated.

rom1v, 2016-10-16:
By calling `result.get()`, you're blocking your event handling thread (from which your `hotplug_callback` is called). I guess this can cause a deadlock with `libusb_open()`…

closed 2016-11-05 by mcuee

f1xpl, 2017-12-05:
So is it not allowed to open device from the callbacks (hotplug callbacks, transfer callbacks)? I discovered a deadlock when I want to open device in transfer callback (libusb_open() hangs).