libusb/libusb #222: libusb_control_transfer fails with -2 when transfering 1024 bytes with HID reports

falstaff84, 2016-11-02:
The device I use has two output reports, 0x01 and 0x02. The latter expects 1024 bytes (and reports so in the HID Report Descriptor). This is how libusb output when I open the device:

    libusb: debug [hid_open] set maximum input buffer size to 512
    libusb: debug [hid_open] 2 HID input report value(s) found
    libusb: debug [hid_open]   Report ID: 0x03
    libusb: debug [hid_open]   Report ID: 0x04
    libusb: debug [hid_open] 2 HID output report value(s) found
    libusb: debug [hid_open]   Report ID: 0x01
    libusb: debug [hid_open]   Report ID: 0x02
    libusb: debug [hid_open] 0 HID feature report value(s) found

Later I try to transfer 16 bytes using report 0x01 (which works nicely) and then 1024 bytes over report 0x02 (which fails):

    libusb: debug [libusb_alloc_transfer] transfer 00FEEE28
    libusb: debug [libusb_submit_transfer] transfer 00FEEE28
    libusb: debug [hid_submit_control_transfer] will use interface 0
    libusb: debug [_hid_set_report] report ID: 0x01
    libusb: debug [usbi_add_pollfd] add fd 1 events 1
    libusb: debug [libusb_get_next_timeout] next timeout in 0.999620s
    libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
    libusb: debug [handle_events] poll fds modified, reallocating
    libusb: debug [handle_events] poll() 2 fds with timeout in 1000ms
    libusb: debug [handle_events] poll() returned 1
    libusb: debug [windows_handle_events] checking fd 1 with revents = 0001
    libusb: debug [usbi_remove_pollfd] remove fd 1
    libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 17
    libusb: debug [usbi_handle_transfer_completion] transfer 00FEEE28 has callback 6B807190
    libusb: debug [sync_transfer_cb] actual_length=17
    libusb: debug [libusb_free_transfer] transfer 00FEEE28
    libusb: debug [libusb_alloc_transfer] transfer 00FEEE28
    libusb: debug [libusb_submit_transfer] transfer 00FEEE28
    libusb: debug [hid_submit_control_transfer] will use interface 0
    libusb: debug [_hid_set_report] invalid size (1025)
    libusb: debug [libusb_free_transfer] transfer 00FEEE28
    err=-2, last_trans=0x0, cnt=0x4000, max=0x400

I then lowered the amount of data to be transferred to 1023, which fails in a different way:

    libusb: debug [hid_submit_control_transfer] will use interface 0
    libusb: debug [_hid_set_report] report ID: 0x02
    libusb: debug [_hid_set_report] Failed to Write HID Output Report: [87] The parameter is incorrect.
    libusb: debug [libusb_free_transfer] transfer 006FEE28
    err=-1, last_trans=0x0, cnt=0x4000, max=0x3ff

It seems that in the first case libusb is rejecting the transfer due to the size, but I am pretty sure that it is fine to transfer 1025 (including report ID), so to me it seems an issue in libusb.

The second case seems Windows related. [This issue](https://github.com/signal11/hidapi/issues/155) in hidapi notes that Windows by default expects the reported length of reports, and fails otherwise.

mcuee, 2016-11-03:
Just a question, why not use HIDAPI? It is more portable than libusb for HID device.

falstaff84, 2016-11-03:
I thought about it, but the same utility also supports devices which use a bulk class driver mode... I'd rather prefer to have only one library/API. And libusb was otherwise working fine for Linux already...

commit 18a1452e (2017-03-01): Windows: Allow HID reports up to 1024 + report id

HID reports can contain up to 1024 bytes. When using a report id
the complete buffer will be 1025 bytes. Do not return an error
if the user supplies a report id and a buffer of 1025 bytes size.

Closes #222
Closes #226
closed 2017-03-01 by dickens