libusb/libusb #224: Want: LibUSBDotNetCore - LibUSB wrapper for .NET Core (Windows 10 Cross Plattform apps)

mobidev111, 2016-11-03:
Windows 10 cross-platform apps are based on .Net Core, not the traditional .NET Framework. LibUSBDotNet doesn't work as it's focused on .NET.

Microsoft currently leaves SerialPort Support (via USB) up to the community: https://github.com/dotnet/corefx/issues/984

Idea: create a .NET Core wrapper like LibUSBDotNet to make LibUSB available to Windows 10 cross-platform apps.

"LibUSBDotNetCore - .NET Core C# USB library for LibUSB. Using the common device classes, applications work with all operating systems and drivers without modification."

Anyone interested? Already working on this? Any starting points?

gusmanb, 2016-11-03:
I would be glad to help in my spare time, I think it's a very necessary function for .net core.

mcuee, 2016-11-05:
This could be interesting. On the other hand, this will not be part of the libusb project. Probably you may want to ask in the libusb mailing list and I believe there will be people who are interested in cross-platform .Net Core libusb wrapper.

closed 2016-11-05 by mcuee