libusb/libusb #227: libusb_get_device_list() cann't list USB 3.1 Device?

Tody-Guo, 2016-11-07:
We have device Asmedia USB 3.1 device. When I plugin Type-C device, this function cann't list the devices.

Tody-Guo, 2016-11-08:
I found that, if install Asmedia driver, libusb_get_device_list() can't list the device, but use inbox-driver of asmedia 3.1 driver. libusb_get_device_list() functon can list deivces.

hjelmn, 2016-11-10:
Need more info. Is this windows?

Tody-Guo, 2016-11-10:
Yes, in windows. But, there is something strange. First time, libusb_get_device_list() can list the device, then plug-out type-c device and plug-in other side libusb_get_device_list() can still list the device. but when change it back to original side, libusb_get_device_list() can't list the type-c devices. type-c device is missing. I use another tool to check. it is fine. but libusb_get_device_list() can't list the type-c devices. 
then reboot the system, libusb_get_device_list()  works. if  you try again, libusb_get_device_list() can't list type-c device, too.

diabolo38, 2016-11-10:
what windows version ? 
you can  expect strange issue with win 7 cos usb 3.0 3.1 support is done by device driver and it may not implement all ioctl use by libusb enumeration.
For instance strange enumeration issue (single port work and port may swap) occur on TI xhci controller .

Tody-Guo, 2016-11-11:
Windows 10 14393.

mcuee, 2016-11-13:
If the Windows inbox driver works but ASMedia driver does not work, it is
quite possible the name of the hub driver is not included in libusb Windows
backend yet. What is the hub driver name listed under Device Manager for
the ASMedia driver? Right now the one included for ASMedia is ASMTHUB3.

In libusb/os/windows.c file, look at the line const char* hub_driver_names[]

mcuee, 2016-11-21:
I just googled and find out that the hub driver for ASMedia driver is still ASMTHUB3. Based on your description, it is a bit strange. I am wondering what you have done. Take note that libusb Windows does not support hotplug as of now. You may need to wait for sometime before you run libusb_get_device_list().

mcuee, 2017-02-13:
Will close this issue for now. It may be related to ASMedia driver. 
To OP: please reopen this issue if you have more info.

closed 2017-02-13 by mcuee

ArnonCh, 2017-10-10:
Hi,

I'm facing the same problem as described by Tody-Guo.
My windows version is 8.1 pro.

I tried to debug it and see where it fails - 
Inside the function **_init_device_** (windows_winusb.c)
When it calls the funcion **_DeviceIoControl_** with **_IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX_** it fails with the error of no device connected

photron, 2018-03-06:
I had problems with an ASMedia hub on Windows before and in my case the enumeration changes in libusb 1.0.22-rc3 have fixed the problem now.