libusb/libusb #232: usbdk: Crash during a control transfer submission

leoetlino, 2016-11-21:
It looks like the recently added usbdk backend is causing crashes when submitting a control transfer.

The crash happens in [windows_usbdk.c, at line 562](https://github.com/libusb/libusb/blob/ad6f234/libusb/os/windows_usbdk.c#L562):

```c
wfd = usbi_create_fd(sysHandle, RW_READ, NULL, NULL);
```

Here is the stack trace:

[image]

Could this be because `sysHandle` (returned by `GetRedirectorSystemHandle`) is invalid? Note that I've also had two other issues with the usbdk backend, which may or may not be related (even though they're just as important imo):

* `usbdk_open` (libusb_open) sometimes fail to start redirecting a device for no obvious reason, even when no other application is using it. This [error](https://github.com/libusb/libusb/blob/ad6f234/libusb/os/windows_usbdk.c#L423) gets shown in the debug logs. I think this may be related, as it also has to do with the redirector.

* `usbdk_get_device_list` randomly fails to return a device list. It apparently has an issue opening the usbdk driver: "Driver file operation error. Failed to open device symlink \\.\UsbDk (Error code = 0x5)"

None of these issues occur with the WinUSB backend, but we wanted to try and switch to the usbdk backend because (1) it's more convenient to use (assuming it works, of course) as drivers don't have to be installed for every device to use; (2) it has official support for isochronous transfers, which are needed for webcams and microphones.

Let me know if I can help with something.

(note: I don't have a Windows box. The following info and stack trace was provided by someone else)

(PS: libusb 1.0.21 was used for testing the usbdk backend. Here are the relevant [SubmitTransfer function in Dolphin](https://github.com/leoetlino/dolphin/blob/352fff4df71108e9bfd66e37d0dbf1b2efc60482/Source/Core/Core/IPC_HLE/USB/LibusbDevice.cpp#L115) and the [Dolphin PR build](http://dl.dolphin-emu.org/prs/pr-4465-dolphin-latest-dbg-x64.7z), in case these are useful.)

leoetlino, 2016-11-22:
I forgot to indicate that the crash occured with the camera bundled with the Wii game _Your Shape_ (057e:030a). But I suspect it also occurs with other devices.

leoetlino, 2016-12-14:
Is this the appropriate place for reporting this sort of issues? Or should I post on the mailing list?

mcuee, 2016-12-14:
This is the right place to report potential anomalies/bugs. But you can post to the mailing list as well.

I did ask in the mailing list already. You can reply my email.
http://marc.info/?l=libusb-devel&m=148068446122008&w=2

dmitryfleytman, 2016-12-14:
Hi @leoetlino, thanks for the report.

Regarding the crash - could you check where does it happen exactly in usbi_create_fd()?

Regarding other issues - could you please collect UsbDk driver logs as described at https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L52? These logs will help us understand what is happening on your system.

Also, please make sure you're using the latest UsbDk version (https://github.com/daynix/UsbDk/releases/tag/v1.00-15)

~Dmitry

leoetlino, 2016-12-14:
Unfortunately, I can't check that since I don't have a real Windows box. According to the person who provided the stack trace though, that's all there is in the call stack; there's nothing above `usbdk_do_control_transfer`.

I'll try to get a driver log and post it.

And yes, we were using the latest version.

dmitryfleytman, 2016-12-15:
@leoetlino, there must be something above usbdk_do_control_transfer, not sure why debugger does not show it, but could you ask the person who observes the issue to add printouts and try to understand where exactly it crashes?

JMC47, 2016-12-16:
I tried to setup the needed tools for debugging as per https://github.com/daynix/UsbDk/blob/master/Documentation/Tracing.txt#L52 but they aren't there on my machine.  I'm using Windows 10, and I installed the debug tools from Microsoft to see if maybe not having that was causing the issue, but I still was unable to find it in any location.

dmitryfleytman, 2016-12-18:
@JMC47, do you have WDK installed on your machine?
If not, get it from https://developer.microsoft.com/en-us/windows/hardware/windows-driver-kit

JMC47, 2016-12-25:
Sorry about the delays.  Been a busy week + testing the branch on the current LibUSB implementation so bouncing back and forth between drivers.  I will get back to you likely today or tomorrow when I get a little time.

JMC47, 2016-12-25:
This is a session recorded in which Dolphin was started (but the game was not running yet,) and the USB Camera the game uses was plugged in.

I then booted the game from the gamelist, and when it attempted to connect to the camera, Dolphin crashed.

[UsbDkTrace.zip](https://github.com/libusb/libusb/files/672199/UsbDkTrace.zip)

dmitryfleytman, 2016-12-25:
Thanks for logs. I am on vacation during this week and will take a look after that.

sameehj, 2017-01-15:
@leoetlino Hi,
I am currently trying to reproduce the issue using the dolphin binaries that you have provided, however when I try to open the iso of "your shape" I get the error: "failed to init libusb error". Do you have any idea what that could be?

leoetlino, 2017-01-15:
Do you have UsbDk installed?

leoetlino, 2017-01-16:
@sameehj I forgot to say, but you also need to have the bundled camera device. Or the game will not try to use it, so the crash never happens.

sameehj, 2017-01-16:
@leoetlino I have successfully solved the issue of the "failed to init libusb error", it was an issue with usbdk as you have stated. Anyways I am getting "The USB Camera is required. Turn the Wii console off and connect the USB Camera to the USB connector.", I think this happens because I don't have the bundled camera! Is there any different Wii game which I could use in order to reproduce this with a regular webcam?

leoetlino, 2017-01-16:
@sameehj unfortunately, there are only two camera games for the Wii, and they both use the same camera (same publisher, ofc). The issue seems to happen for microphones too, though. Do you have a Logitech microphone (046d:0a03)? If so, you can try using it in any microphone game. Here's a [list](https://wiki.dolphin-emu.org/index.php?title=Category:Microphone_(Input_supported)) of games which support mic input (make sure to use a Wii game though).

dmitryfleytman, 2017-01-26:
As our investigation has shown the crash is caused by submission of transfer with invalid device handle.

When device redirection fails UsbDk returns error and invalid device handle value (-1). Attempt to submit a transfer with this handle leads to crash in UsbDk user mode API DLL. While the crash is a consequence and not the root cause of the issue, it is misleading and fixed in UsbDk by commit https://github.com/daynix/UsbDk/commit/b0660af0d1628885252cac95b5198aa482d13883 by adding additional handle verification logic.

Fixed will be included into the next UsbDk version (1.0.16) that will be released in a next few days.

The real reason for the failure is attempt to redirect a USB device twice at the same time. This is not support by UsbDk and not by WinUsb (AFAIK).

diabolo38, 2017-01-27:
@dmitryfleytman  what do you consider " redirect a USB device twice "  , claim  an interface  or open (in libsusb sense) multiple time in one or multiple process  ?

dmitryfleytman, 2017-01-27:
Open it twice in libusb sense from one or multiple processes.

diabolo38, 2017-01-27:
@dmitryfleytman  maybe silly question should it not be done at claim only ? what will occur if opened device is composite with interface(s) handle by some driver to keep ? would it get filtered as well ?

dmitryfleytman, 2017-01-29:
@diabolo38, the way UsbDk works now, composite device will be acquired exclusively as a whole. Composite device splitting is not currently supported.

diabolo38, 2017-01-29:
@dmitryfleytman  that si a real concern and a limitation somehow. Many embedded device  are a mix of standard function (cdc, mass storage) used by mean of native os driver and custom functions to be  handle by app via libusb. Today in window this may requires specific inf file or installer but is fully possible and supportedto some extent (i do use it )  and i guess is fully transparent for OsX and linux. It shalll still be the case an best have salme featre otehr all OS . Somehow this is linked and will be solved by usbdk s/w activation.

dmitryfleytman, 2017-01-30:
@diabolo38, you are right, this feature may be useful indeed and it is missing at the moment.

It might be implemented in UsbDk, i.e. there are no architectural limitations for that. This feature is in our roadmap, however there is no specific timeline for it.

And of course, contributions are welcome!

PipeRojas, 2018-05-27:
Hi everybody, I'm having an issue similar to this one, I'm trying to play guitar hero and rock band on dolphin emulator with the USB Logitech vantage microphone with the USB passthrough feature and usbdk 1.00-19 but after registering the device in the white list of dolphin usb passthrough and starting the game, it crashes randomly, sometimes I can play a half of a song and dolphin crashes with Windows 10 Pro up to date (fall creators update).
Anybody know any solution? 

And thanks a lot btw

sameehj, 2018-05-27:
@PipeRojas Hi,

Thanks for reporting the issue, in order to make this efficient please do the following;
* Open a separate issue for your report as the issue you are facing might be completely different.
* Provide us with dump file generated by Windows during the crash. This usually resides in c:/Windows/MEMORY.dmp

Thanks :)

Hacsev, 2019-02-15:
@sameehj I would like to know the status of this bug. Did the developers ever attempt to fix it? @PipeRojas followed your instructions and created a separate issue, but all that really happened was that @mcuee just referenced this issue back with no signs of ever addressing the issue.

Hacsev, 2019-02-15:
@sameehj I would like to know the status of this bug. Did the developers ever attempt to fix it? @PipeRojas followed your instructions and created a separate issue, but all that really happened was that @mcuee just referenced this issue back, making it seem like the issue will never be addressed.

leoetlino, 2019-05-21:
Closing as this appears to have been fixed in latest libusb.

closed 2019-05-21 by leoetlino