libusb/libusb #233: libusb is not working with Android 7.1

kennethtang4, 2016-11-22:
To start with, it is not a problem of libusb itself but more to the Android system but I am looking if any one have an idea on how to fix this problem.

The library is working perfectly up to Android 6.0 (haven't tested with 6.1 and 7.0 yet).

The problem is happening in the function "sysfs_get_device_list" in the file "linux_usbfs.c". I am getting an errno code of 13 which is permission denied. It seems that Android has changed the permission of accessing the directory /sys/bus/usb/devices. USB Permission has been asked and granted through the USB Manager beforehand (needed for previous Android versions as well) and I tried to request access for READ_EXTERNAL_STORAGE too but this has no effect on it.

So I wonder if any one can come with a workaround for this?

mcuee, 2016-11-25:
Please refer to this one as well. https://github.com/libusb/libusb/issues/188

mcuee, 2016-12-29:
You may want to try Pull Request #242.

kennethtang4, 2017-01-05:
#242 can solve the problem. The naming of the function is confusing though. The function libusb_wrap_fd is escentially opening the usb by file descriptor. Also, it will not add the usb device the to device list in libusb, so mind that some libraries depend on libusb may need to be modified to make it working.

mcuee, 2020-01-06:
I will close this for now. But the comment is good. I will add this link to the #242 .

closed 2020-01-06 by mcuee