libusb/libusb #235: Where is `INSTALL` file?

kuanyen, 2016-11-29:
```
Installation Instructions for Windows
*************************************

If you are compiling for MinGW or cygwin, please refer to the INSTALL file.
```
But I don't find any INSTALL file.

mcuee, 2016-12-02:
INSTALL file is generated when running autogen.sh or bootstrap.sh when running under Linux, Mac OS X or Cygwin or MSYS/MSYS2.

GPery, 2018-09-10:
Considering the fact it's referenced in INSTALL_WIN.txt which exists off the bat, it might be worth mentioning there.

E3V3A, 2019-01-18:
Yes please! Please add a separate and more clear INSTALL file.
For example. I would like to just compile the *examples* without having to re-compile the entire library.  
How can I do this using MinGW on x64??

LudovicRousseau, 2019-01-18:
The `.tar.bz2` archives in https://github.com/libusb/libusb/releases contains the `INSTALL` file you are looking for.
I don't know if we must keep this issue open.

commit 178e433b (2020-08-18): Windows: Update INSTALL_WIN.txt

Add info regarding where the 'INSTALL' file comes from.

Remove link for Windows SDKs as this is no longer relevant.

Update to account for Visual Studio 2019.

Closes #235
closed 2020-08-18 by dickens