libusb/libusb #237: Slow listing of usb devices

pali, 2016-12-07:
Sent to ML, but after 8 months without any answer: https://sourceforge.net/p/libusb/mailman/message/34985373/

```
Hi!

I'm in step of migrating application from libusb 0.1 to libusb 1.0 and 
see one big problem in libusb 1.0 Linux implementation.

Listing of usb devices is too slow. When usb device appear on USB bus 
just for 0.3 - 0.7 seconds, then libusb 1.0 does not see it. libusb 0.1 
does not have any problem and perfectly detect it.

It means that I cannot migrate application to libusb 1.0 because it 
stops working, it does not detect usb device in slow period of second.

Is there any way how to solve this problem to speed up listing of usb 
devices on Linux? Currently Linux distributions are going to deprecate 
and remove libusb 0.1, so I will not able to use working libusb 0.1 
library.

I suspect that slow listing is because of asynchronous nature of Linux 
libusb 1.0 backend...
```

LudovicRousseau, 2016-12-07:
Hello,
Have you tried to use the hotplug mechanism?
http://libusb.sourceforge.net/api-1.0/group__hotplug.html

pali, 2016-12-07:
Yes, and that libusb-1.0 hotplug mechanism is unusable too. As wrote I think the whole libusb is slow due to asynchronous code. So hotplug (as async too) cannot help here.

Here is testing code:
```
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <signal.h>
#include <unistd.h>

#include <libusb-1.0/libusb.h>

static volatile sig_atomic_t signal_quit;

static void signal_handler(int signum) {
  signal_quit = 1;
  (void)signum;
}

static int hotplug_callback(struct libusb_context *ctx, struct libusb_device *dev,
                     libusb_hotplug_event event, void *user_data) {
  libusb_device_handle *handle = NULL;
  struct libusb_device_descriptor desc;
  char buf[1024];
  int rc;

  (void)libusb_get_device_descriptor(dev, &desc);

  if (event == LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT) {
    printf("Unplugged: idVendor=0x%04x idProduct=0x%04x\n", desc.idVendor, desc.idProduct);
    return 0;
  } else if (event != LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED) {
    printf("Unhandled event %d\n", event);
    return 0;
  }

  memset(buf, 0, sizeof(buf));

  rc = libusb_open(dev, &handle);
  if (rc == LIBUSB_SUCCESS) {
    (void)libusb_get_string_descriptor_ascii(handle, desc.iProduct, (unsigned char*)buf, sizeof(buf));
    libusb_close(handle);
  }

  printf("Plugged: idVendor=0x%04x idProduct=0x%04x iProduct=%s\n", desc.idVendor, desc.idProduct, buf);
  return 0;
}

int main (void) {
  libusb_hotplug_callback_handle handle;
  int rc;

  libusb_init(NULL);

  rc = libusb_hotplug_register_callback(NULL, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED | LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT,
                                        0, LIBUSB_HOTPLUG_MATCH_ANY, LIBUSB_HOTPLUG_MATCH_ANY,
                                        LIBUSB_HOTPLUG_MATCH_ANY, hotplug_callback, NULL,
                                        &handle);
  if (rc != LIBUSB_SUCCESS) {
    printf("Error creating a hotplug callback\n");
    libusb_exit(NULL);
    return EXIT_FAILURE;
  }

  signal(SIGINT, signal_handler);

  while (!signal_quit) {
    libusb_handle_events_completed(NULL, NULL);
    usleep(10000);
  }

  libusb_hotplug_deregister_callback(NULL, handle);
  libusb_exit(NULL);

  return 0;
}
```

And here is combined output from dmesg and above testing code:
```
[ 2732.336276] usb 3-1: new high-speed USB device number 26 using xhci_hcd
[ 2732.352813] usb 3-1: unable to get BOS descriptor
[ 2732.353563] usb 3-1: New USB device found, idVendor=0421, idProduct=0106
[ 2732.353566] usb 3-1: New USB device strings: Mfr=34, Product=38, SerialNumber=0
[ 2732.353568] usb 3-1: Product: Nokia USB ROM
[ 2732.353569] usb 3-1: Manufacturer: Nokia
[ 2732.654600] usb 3-1: USB disconnect, device number 26
[ 2733.625071] usb 3-1: new high-speed USB device number 27 using xhci_hcd
[ 2733.728148] usb 3-1: New USB device found, idVendor=0421, idProduct=0105
[ 2733.728151] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[ 2733.728153] usb 3-1: Product: Nokia N900 (Update mode)
[ 2733.728154] usb 3-1: Manufacturer: Nokia
[ 2733.728155] usb 3-1: SerialNumber: XXXXXXXXXXXXXXXXXX
[ 2734.288626] NET: Registered protocol family 35
[ 2734.303286] usbcore: registered new interface driver cdc_phonet
Plugged: idVendor=0x0421 idProduct=0x0105 iProduct=Nokia N900 (Update mode)
[ 2735.353984] usb 3-1: USB disconnect, device number 27
Unplugged: idVendor=0x0421 idProduct=0x0105
```

As you can see there is just 0.3s window (from 2732.336276 to 2732.654600) to detect and immediatelly open usb device "Nokia USB ROM". And this cannot be done by hotplug or other asynchronous waiting. Busy waiting needs to be used for such action.

Above libusb-1.0 hotplug test code detected just usb device "Nokia N900 (Update mode)" and it was later then 1s, which is slow as hell and absolutelly unusable.

Note that libusb 0.1 does not have this problem and can dectect & open usb device "Nokia USB ROM" in that 0.3s window.

mcuee, 2016-12-07:
I thiink you need to provide more detail what you want to do. And it is good to provide the debug log (set LIBUSB_DEBUG=4 and then run your libusb program) to see where is the delay. And you can even provide the debug log of your libusb-0.1 program as well (set USB_DEBUG=255 and run your libusb-0.1 program).

mcuee, 2016-12-07:
BTW, what is your Linux version? As far as I know, majority of the Linux distro use libusb-compat-0.1 which is based on libusb-1.0.

mcuee, 2016-12-07:
BTW, I think this is better discussed in the mailing list. You need to subscribe to the mailing list in order to post, please go to this URL.
https://lists.sourceforge.net/lists/listinfo/libusb-devel

When your previously posted, it did not make to the mailing list. Even though the other user forwarded it to the mailing list, it does not have the details. Once you provide more details, I think there will be more meaningful discussions in the mailing list.

pali, 2016-12-07:
Here is combined log from dmesg and above testing hotplug code. It was started with env LIBUSB_DEBUG=4 and against libusb-1.0.20 compiled from tarball:
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000031] [00004170] libusb: debug [libusb_init] created default context
[ 0.000074] [00004170] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.000093] [00004170] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000110] [00004170] libusb: debug [op_init] bulk continuation flag supported
[ 0.000114] [00004170] libusb: debug [op_init] zero length packet flag supported
[ 0.000123] [00004170] libusb: debug [op_init] sysfs can relate devices
[ 0.000126] [00004170] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000219] [00004171] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.000442] [00004170] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.000452] [00004170] libusb: debug [linux_get_device_address] scan usb3
[ 0.000487] [00004170] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.000493] [00004170] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.000496] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.000585] [00004170] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.000591] [00004170] libusb: debug [linux_get_device_address] scan usb4
[ 0.000621] [00004170] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.000626] [00004170] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.000630] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.000707] [00004170] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.000713] [00004170] libusb: debug [linux_get_device_address] scan usb1
[ 0.000743] [00004170] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.000747] [00004170] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.000751] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.000825] [00004170] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.000831] [00004170] libusb: debug [linux_get_device_address] scan 1-1
[ 0.000861] [00004170] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.000865] [00004170] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.000869] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.000890] [00004170] libusb: debug [linux_get_parent_info] Dev 0x11bc3a0 (1-1) has parent 0x11bc2e0 (usb1) port 1
[ 0.000937] [00004170] libusb: debug [linux_get_device_address] getting address for device: 1-1.5 detached: 0
[ 0.000943] [00004170] libusb: debug [linux_get_device_address] scan 1-1.5
[ 0.000973] [00004170] libusb: debug [linux_get_device_address] bus=1 dev=3
[ 0.000978] [00004170] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[ 0.000981] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[ 0.001001] [00004170] libusb: debug [linux_get_parent_info] Dev 0x11bd710 (1-1.5) has parent 0x11bc3a0 (1-1) port 5
[ 0.001103] [00004170] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.001109] [00004170] libusb: debug [linux_get_device_address] scan usb2
[ 0.001138] [00004170] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.001143] [00004170] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.001147] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.001239] [00004170] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.001250] [00004170] libusb: debug [linux_get_device_address] scan 2-1
[ 0.001284] [00004170] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.001293] [00004170] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.001300] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.001330] [00004170] libusb: debug [linux_get_parent_info] Dev 0x11be5c0 (2-1) has parent 0x11bd820 (usb2) port 1
[ 0.001390] [00004170] libusb: debug [linux_get_device_address] getting address for device: 2-1.5 detached: 0
[ 0.001400] [00004170] libusb: debug [linux_get_device_address] scan 2-1.5
[ 0.001433] [00004170] libusb: debug [linux_get_device_address] bus=2 dev=3
[ 0.001442] [00004170] libusb: debug [linux_enumerate_device] busnum 2 devaddr 3 session_id 515
[ 0.001449] [00004170] libusb: debug [linux_enumerate_device] allocating new device for 2/3 (session 515)
[ 0.001480] [00004170] libusb: debug [linux_get_parent_info] Dev 0x11be680 (2-1.5) has parent 0x11be5c0 (2-1) port 5
[ 0.001600] [00004170] libusb: debug [linux_get_device_address] getting address for device: 2-1.8 detached: 0
[ 0.001611] [00004170] libusb: debug [linux_get_device_address] scan 2-1.8
[ 0.001637] [00004170] libusb: debug [linux_get_device_address] bus=2 dev=4
[ 0.001643] [00004170] libusb: debug [linux_enumerate_device] busnum 2 devaddr 4 session_id 516
[... cut ...]
[13.935167] [00004170] libusb: debug [libusb_unref_device] destroy device 3.1
[13.935169] [00004170] libusb: debug [usbi_remove_pollfd] remove fd 6
[13.935181] [00004170] libusb: debug [usbi_remove_pollfd] remove fd 8
[13.935196] [00004171] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

As you can see libusb-1.0 detected some device after "Nokia USB ROM" was connected, but it is slow and just received detached=1 event... So it is unusable...

And about libusb 0.1, I'm using real libusb 0.1, not libusb-compat around libusb-1.0.

pali, 2016-12-07:
Here is libusb 0.1 code which use busy waiting:

```
#include <stdio.h>
#include <string.h>
#include <signal.h>

#include <usb.h>

static volatile sig_atomic_t signal_quit;

static void signal_handler(int signum) {
  signal_quit = 1;
  (void)signum;
}

static int usb_device_is_valid(struct usb_device * dev) {
  static char found[1 << 16];

  if (dev->descriptor.idVendor != 0x0421)
    return 0;

  if (found[dev->descriptor.idProduct])
    return 0;

  found[dev->descriptor.idProduct] = 1;
  return 1;
}

static struct usb_device * usb_search_device(struct usb_device * dev, int level) {
  int i;
  struct usb_device * dev2;

  if (!dev)
    return NULL;

  if (usb_device_is_valid(dev))
    return dev;

  for (i = 0; i < dev->num_children; i++) {
    dev2 = usb_search_device(dev->children[i], level + 1);
    if (dev2)
      return dev2;
  }

  return NULL;
}

int main() {
  usb_dev_handle * udev;
  struct usb_bus * bus;
  struct usb_device *dev;
  struct usb_device *dev2;
  char buf[1024];

  usb_init();
  usb_find_busses();

  signal(SIGINT, signal_handler);

  while (!signal_quit) {
    usb_find_devices();

    dev = NULL;

    for (bus = usb_get_busses(); bus; bus = bus->next) {
      if (bus->root_dev)
        dev = usb_search_device(bus->root_dev, 0);
      else {
        for (dev2 = bus->devices; dev2; dev2 = dev2->next) {
          dev = usb_search_device(dev2, 0);
          if (dev)
            break;
        }
      }
      if (dev)
        break;
    }

    if (!dev) {
      usleep(0xc350);
      continue;
    }

    memset(buf, 0, sizeof(buf));

    udev = usb_open(dev);
    if (udev) {
      usb_get_string_simple(udev, dev->descriptor.iProduct, buf, sizeof(buf)-1);
      usb_close(udev);
    }

    printf("Plugged: idVendor=0x%04x idProduct=0x%04x iProduct=%s\n", dev->descriptor.idVendor, dev->descriptor.idProduct, buf);
  }

  return 0;
}
```

It detect "Nokia USB ROM" usb device without any problem. Here is combined output from dmesg and this libusb 0.1 code:

```
[12539.007580] usb 3-2: new high-speed USB device number 10 using xhci_hcd
[12539.024076] usb 3-2: unable to get BOS descriptor
[12539.024694] usb 3-2: New USB device found, idVendor=0421, idProduct=0106
[12539.024698] usb 3-2: New USB device strings: Mfr=34, Product=38, SerialNumber=0
[12539.024700] usb 3-2: Product: Nokia USB ROM
[12539.024702] usb 3-2: Manufacturer: Nokia
Plugged: idVendor=0x0421 idProduct=0x0106 iProduct=Nokia USB ROM
[12539.325608] usb 3-2: USB disconnect, device number 10
[12540.304322] usb 3-2: new high-speed USB device number 11 using xhci_hcd
[12540.407548] usb 3-2: New USB device found, idVendor=0421, idProduct=0105
[12540.407560] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[12540.407562] usb 3-2: Product: Nokia N900 (Update mode)
[12540.407563] usb 3-2: Manufacturer: Nokia
[12540.407564] usb 3-2: SerialNumber: XXXXXXXXXXXXXXXXXX
Plugged: idVendor=0x0421 idProduct=0x0105 iProduct=Nokia N900 (Update mode)
[12542.020674] usb 3-2: USB disconnect, device number 11
```

As you can see libusb 0.1 (real!, not compat wrapper) is working fine and can detect usb devices in time. libusb-1.0 either hotplug code or similar busy waiting loop is slow and so is unusable...

mcuee, 2016-12-11:
Can you post the debug log for libusb 0.1 as well? Thanks. You can set USB_DEBUG=255 for that purpose.

mcuee, 2016-12-11:
Setting this as Linux for now.

pali, 2016-12-11:
I do not know why you need that log as libusb 0.1 is working and such busy waiting log is long, but here is (again combined with dmesg):

```
usb_set_debug: Setting debugging level to 255 (on)
usb_os_init: Found USB VFS at /dev/bus/usb
usb_os_find_busses: Found 004
usb_os_find_busses: Found 003
usb_os_find_busses: Found 002
usb_os_find_busses: Found 001
usb_os_find_devices: Found 001 on 004
skipping descriptor 0x30
skipped 1 class/vendor specific endpoint descriptors
usb_os_find_devices: Found 001 on 003
usb_os_find_devices: Found 004 on 002
skipped 1 class/vendor specific interface descriptors
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 003 on 002
usb_os_find_devices: Found 002 on 002
usb_os_find_devices: Found 001 on 002
error obtaining child information: Inappropriate ioctl for device
error obtaining child information: Inappropriate ioctl for device
usb_os_find_devices: Found 003 on 001
skipping descriptor 0xFF
skipping descriptor 0xB
skipped 2 class/vendor specific endpoint descriptors
skipped 5 class/vendor specific interface descriptors
skipping descriptor 0x25
skipped 1 class/vendor specific endpoint descriptors
skipped 15 class/vendor specific interface descriptors
usb_os_find_devices: Found 002 on 001
usb_os_find_devices: Found 001 on 001
error obtaining child information: Inappropriate ioctl for device
usb_os_find_devices: Found 001 on 004
skipping descriptor 0x30
skipped 1 class/vendor specific endpoint descriptors
usb_os_find_devices: Found 001 on 003
usb_os_find_devices: Found 004 on 002
skipped 1 class/vendor specific interface descriptors
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 003 on 002
usb_os_find_devices: Found 002 on 002
usb_os_find_devices: Found 001 on 002
error obtaining child information: Inappropriate ioctl for device
error obtaining child information: Inappropriate ioctl for device
usb_os_find_devices: Found 003 on 001
skipping descriptor 0xFF
skipping descriptor 0xB
skipped 2 class/vendor specific endpoint descriptors
skipped 5 class/vendor specific interface descriptors
skipping descriptor 0x25
skipped 1 class/vendor specific endpoint descriptors
skipped 15 class/vendor specific interface descriptors
usb_os_find_devices: Found 002 on 001
usb_os_find_devices: Found 001 on 001
error obtaining child information: Inappropriate ioctl for device
usb_os_find_devices: Found 001 on 004
skipping descriptor 0x30
skipped 1 class/vendor specific endpoint descriptors
usb_os_find_devices: Found 001 on 003
usb_os_find_devices: Found 004 on 002
skipped 1 class/vendor specific interface descriptors
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 003 on 002
usb_os_find_devices: Found 002 on 002
usb_os_find_devices: Found 001 on 002
error obtaining child information: Inappropriate ioctl for device
error obtaining child information: Inappropriate ioctl for device
usb_os_find_devices: Found 003 on 001
skipping descriptor 0xFF
skipping descriptor 0xB
[... 2268 lines cut ...]
error obtaining child information: Inappropriate ioctl for device
usb_os_find_devices: Found 001 on 004
skipping descriptor 0x30
skipped 1 class/vendor specific endpoint descriptors
usb_os_find_devices: Found 001 on 003
^Cusb_os_find_devices: Found 004 on 002
skipped 1 class/vendor specific interface descriptors
skipped 1 class/vendor specific interface descriptors
usb_os_find_devices: Found 003 on 002
usb_os_find_devices: Found 002 on 002
usb_os_find_devices: Found 001 on 002
error obtaining child information: Inappropriate ioctl for device
error obtaining child information: Inappropriate ioctl for device
usb_os_find_devices: Found 003 on 001
skipping descriptor 0xFF
skipping descriptor 0xB
skipped 2 class/vendor specific endpoint descriptors
skipped 5 class/vendor specific interface descriptors
skipping descriptor 0x25
skipped 1 class/vendor specific endpoint descriptors
skipped 15 class/vendor specific interface descriptors
usb_os_find_devices: Found 002 on 001
usb_os_find_devices: Found 001 on 001
error obtaining child information: Inappropriate ioctl for device
```

pali, 2016-12-12:
@mcuee Thats all? Or do you need anything else?

mcuee, 2016-12-13:
Yes that is all. I am thinking that your specific use case may be which libusb under Linux may not be able to support. I could be wrong. Maybe it is better you use the mailing list to tell a bit more details what is your use case and then probably there is a way out.

pali, 2016-12-13:
Use case (as already wrote): start communication with device at correct time. If computer does not negotiate device protocol in 0.3s time window, then device disconnect from usb.

I already sent email to ML at least two times and other people forwarded it to ML again.

mcuee, 2016-12-14:
You need to subscribe to the mailing list in order to post. Anyway, your
description for the use case is not that clear, what do you want to achieve
in the end? If you provide more details and ask in the mailing list, maybe
there are ways  to sort out (could be even non-libusb  solutions).

pali, 2016-12-14:
What I need to achieve? Communicate with that device ("Nokia USB ROM ") which is on usb bus just for 0.3s.

mcuee, 2016-12-14:
Communicate with the device for what?

mcuee, 2016-12-14:
> Communicate with the device for what?
Eg: Software update?

pali, 2016-12-14:
Short: Query various parameters, change boot settings or load own boot code.
In details: See my project 0xFFFF: https://github.com/pali/0xFFFF

But reason for device communication is irrelevant in this bug report. libusb 1.0 does not even detect usb devices!

mcuee, 2016-12-14:
Having all these details may help. Anyway, I can sense your frustration. I'll shut up for now to see if others can help you better.

msva, 2017-08-02:
any progress on this?
ping devs ( @LudovicRousseau @dickens @hjelmn @pbatard )

I just read this discussion and the thread on maillist and...
I'd like to say that I think that forcing people to subscribe to maillist to even ***take a look an issue*** is a bad way to go (the ML discussion contains no discussion on the issue, but only "reporter should subscribe the ML").

Can't say for @pali, but I already have too big incoming traffic from many other maillists to subscribe to one more just to track one single issue.

Well, yes, I understand that it is "this is opensource, baby, so patches or gtfo" opinion exists, but libusb is too important system package on GNU/Linux these days to just ignore this issue.

This exact issue brakes use-case of flashing very mush devices (not only nokia, but also some mediatek ones, and even more). And standing to "non-libusb-way" is bad option too, imho.

So, can you, please, re-implement the blocking method like it was in libusb0 for this very cases?

marmistrz, 2017-08-17:
I'm the Arch Linux maintainer of a package affected by this bug and developed by @pali. The fact that the package is affected by this bug makes the whole packaging difficult and error-prone because libusb0 is not shipped by the distribution anymore.

I have to depend on a Debian's binary copied around because it's currently the only way of linking with libusb0. Could you please look into this issue that has been open for such a long time?

dickens, 2017-08-25:
@pali are you using the udev or netlink option for hotplug? If using udev, what happens if you switch to netlink?

pali, 2017-08-25:
I'm using code from comment https://github.com/libusb/libusb/issues/237#issuecomment-265406150 As you can see I have not specified udev or netlink in libusb_hotplug_register_callback() call.

daveyoung, 2017-08-28:
Pali, as I said I got a working test result with Fedora libusb, but after checking the dmesg I get below, so apparently my time slot window is longer than yours, it is about 2.83 seconds, I do not know why the difference happens, my kernel is 4.13.0-rc6+:
[ 6594.074552] usb 2-1: Product: Nokia USB ROM
[ 6594.074554] usb 2-1: Manufacturer: Nokia
[ 6594.376517] usb 2-1: USB disconnect, device number 23
[ 6595.229111] usb 2-1: new high-speed USB device number 24 using xhci_hcd
[ 6595.483747] usb 2-1: New USB device found, idVendor=0421, idProduct=0105
[ 6595.483757] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[ 6595.483762] usb 2-1: Product: Nokia N900 (Update mode)
[ 6595.483766] usb 2-1: Manufacturer: Nokia
[ 6595.483770] usb 2-1: SerialNumber: 4D554D333335333037
[ 6596.906483] usb 2-1: USB disconnect, device number 24

pali, 2017-08-28:
Time slot is not longer, look at output again:
```
[ 6594.074554] usb 2-1: Manufacturer: Nokia
[ 6594.376517] usb 2-1: USB disconnect, device number 23
```
It is just 0.3s before device leave and disconnects. In log is missing line `new high-speed USB device number 23` which would tell exact time when device was discovered by kernel.

dickens, 2017-08-28:
@pali this is a compile-time option, not something you specify at runtime. Looking at the log debug log again I see udev messages. Try this again after compiling libusb with the --disable-udev option when running the configure script.

daveyoung, 2017-08-29:
Here is the output of my test result with pali's test code:
[usbtest]#  ./a.out 
Plugged: idVendor=0x0421 idProduct=0x01c7 iProduct=
^C[root@dhcp-128-65 usbtest]#  ./a.out 
Plugged: idVendor=0x0421 idProduct=0x0106 iProduct=
Unplugged: idVendor=0x0421 idProduct=0x0106
Plugged: idVendor=0x0421 idProduct=0x0105 iProduct=
Unplugged: idVendor=0x0421 idProduct=0x0105

OS: Fedora rawhide, mainline kernel 4.13.0-rc6+
libusb-0.1.5-10.fc27.x86_64
(based on libusb-compat)

marmistrz, 2017-09-08:
@pali, I've just recalled that I at one point had libusb-compat before I replaced it with libusb0 and successfully flashed the device. I guess it was a Nokia N950, though, not N900.

Should I try with an N900 + libusb-compat to try to reproduce it on my side?

rosset, 2019-07-06:
hi guys, any progress on this?

hjelmn, 2019-07-07:
This fell off of the radar. Nothing has been done on this as far as I know. Is this still a problem?

pali, 2019-07-18:
@hjelmn see third comment where is reproducer/test code. And in 8th comment is code for libusb 0.1 for comparison. So it can be used to detect if there is still a problem.

dickens, 2020-01-17:
@pali we’re you ever able to try building and running with netlink instead of udev?  libusb 0.1 finds devices through devtmpfs by scanning /dev/bus/usb.  These device nodes are likely to show up quite soon after the device is enumerated.  libusb 1.0 listens for uevents from the kernel through a netlink socket.  There are quite a few uevents that the kernel sends out while setting up a new device, only one of which tells libusb that a new device has arrived.  This also happens in a dedicated thread, so there may be scheduling latencies at play.

pali, 2020-01-17:
@dickens Last time when I tested that 1.0 code I used pre-compiled binaries which were available in linux distributions (IIRC Debian). I do not remember more details, it was 2 or 3 year ago? And I had no motivation to try it again as everybody ignored this issue and there was fully working solution = use working 0.1 library. As stated, reproducer/test code is in this ticket issue.

mcuee, 2020-01-17:
@rosset Probably you can try the suggestion from Chris to see if the issue gets resolved by building libusb with netlink and not udev.

mcuee, 2020-12-10:
Wiki FAQ page updated.
https://github.com/libusb/libusb/wiki/FAQ#Known_quirksdifferences_between_libusbcompat01_and_legacy_libusb01

pali, 2021-01-05:
I tested again reproducer from the https://github.com/libusb/libusb/issues/237#issuecomment-265406150 and issue is still there after 5 years. I used up-to-date Debian stable system which comes with libusb-1.0-0 package in version 2:1.0.22-2. As described device with ids idVendor=0421, idProduct=0106 is not detected by that code. And code from comment https://github.com/libusb/libusb/issues/237#issuecomment-265436838 can detect it without problem.

mcuee, 2021-01-06:
@pali Yes the stock libusb-1.0-0 package in Debain stable will be built against libudev so you will still see the problem. You can refer to the comments by Chris.

Have you tried to build libusb using netlink (disable libudev in the build) and the see if that helps.

mcuee, 2021-01-06:
On the other hand, I can see that Debian still ships the old libusb-0.1 and not libusb-compat-0.1.
https://packages.debian.org/search?keywords=libusb

pali, 2021-01-06:
I do not have a control what configuration distributions ships, I can only build my SW against provided packages. And if dependent SW (=libusb) provides buggy configuration then it is buggy and unusable as anybody can (and distributions like Debian are doing it) compile it with this buggy configuration.

I know that Debian is still shipping libusb-0.1, it was not removed because at that time 5 years ago nobody reacted to that bug report and Debian maintainers decided to still ship libusb-0.1 library for which was not non-buggy alternative.

mcuee, 2021-01-06:
@pali I understand the frustrations. 

On the other hand, keep calling libusb buggy configuration does not really help anyone. We can argue it is the limitation of libudev which libusb depends on. And we are unlikely to remove the libudev dependancies any time. 

And Chris already mentioned the potential work-around to test with netlink and it seems nobody wants to try it to see if that can fix the issue. If that can fix the issue, maybe Debian can ship libusb-compat-0.1 with some special configurations. They can probably even create a new libusb-1.0 package (change some name) to accomodate application like yours.

pali, 2021-01-06:
Well, I do not have time to do more tests and try what works or what not. I have provided testing applications in this issue which anybody can test and check if it works or not. No special HW is needed just to check that any USB device appears in tested application after e.g. 0.05s or 0.1s after inserting it into USB port. If upstream developers/maintainers do not want to handle this issue seriously (they have everything needed for testing, debugging, reproducing and fixing this bug) then the best thing is to stick with working, but older library libusb 0.1. If somebody is proposing alternative to currently working solution with libusb 0.1, then it is up to him to handle issues and fix them. And not to application developers. And if upstream do not want to communicate and handle this issue for more than 5 years, I have no reason to continue trying handle it in any way. I have really done everything what I was able; providing reproducer code which any of maintainers/libusbdeveloper can run and check.

mcuee, 2021-01-06:
I can totally understand your frustration. On the other hand, I guess in this case it will be status quo unless someone jumps in to propose better solution for you or for libusb.

pali, 2021-01-06:
@mcuee: As you can see a ball is now (for 5 years) on libusb developers/maintainers. It is clear bug in the libusb 1.0 library. If developers/maintainers do not want to do anything for 5 years, then I really do not have motivation to do anything else with libusb 1.0

I already have working solution: libusb 0.1. I was there for a long time and until developers/maintainers fix libusb 1.0 it would be still there...

mcuee, 2021-01-06:
It is okay. Since you already have your workaround, I will consider case closed on both side. 

On the libusb side, I will just say it is a limitation of libudev/libusb. The workaround is to ask the user to build libusb using netlink. Or you can switch to other library if the workaround is deemed not acceptable.

Anyway, I will still leave the ticket open and maybe others have better ideas.

mcuee, 2021-01-06:
As for disto package maintainers, one possible work-around is to create two new packages, libusb-compat-0.1_netlink and libusb-1.0_netlink, link the application with these two libraries, if you do not want to keep shipping the old libusb-0.1.

rosset, 2021-01-06:
> Well, I do not have time to do more tests and try what works or what not. I have provided testing applications in this issue which anybody can test and check if it works or not. No special HW is [...]

@pali  can you please provide the details how to currently reproduce the issue, so I can help to test / debug.

@mcuee  I not be able to fix the issue, but I'm glad to help debug / test if someone provide the next steps, or maybe some patches.

I changed my hardware during the past couple years so I'm not sure if I can reproduce the issue atm.

mcuee, 2021-01-06:
Wiki pages updated:
https://github.com/libusb/libusb-compat-0.1/wiki
https://github.com/libusb/libusb/wiki/FAQ

mcuee, 2021-01-06:
@rosset Thanks. No need any patch. Just build libusb with "./configure --disable-udev" and then try again.
1) Step one: download libusb-1.0.24 source code, build with "./configure --disable-udev", install it.
2) to play safe, rebuild libusb-compat-0.1 again and install it (may not be necessary)
3) rebuild your application, or the test code. Compare the results with before the rebuild.

hjelmn, 2021-01-06:
Seems really odd to me that Nokia thought 0.3 seconds was an acceptable timeout for dropping out of update mode. I can understand the frustration that udev does not pass the device off to libusb in a timely fashion. This is a perfect use case for using netlink.

hjelmn, 2021-01-06:
We certainly could fix this by making netlink runtime selectable. It was compile time because udev is the process that sets the permissions for the device. udev is used by all major Linux distros.

pali, 2021-01-06:
@rosset: testing code is in the post https://github.com/libusb/libusb/issues/237#issuecomment-265406150 run it together with dmesg (e.g. by `dmesg -w & ./test`) and verify that any usb device which you plug into computer is detected by testing code in 0.01s after dmesg prints `New USB device found` line. libusb 0.1 does not have any problem to do this detection.

mcuee, 2021-01-06:
@hjelmn That is a good solution -- by making netlink runtime selectable.

pali, 2021-01-06:
> Seems really odd to me that Nokia thought 0.3 seconds was an acceptable timeout for dropping out of update mode.

First: It is not Nokia, but TI and it is bootrom. Moreover it applies to all OMAP3 devices (maybe also OMAP4) and it is standard to not slowdown boot process by waiting longer time period. I do not see nothing odd with 0.3s timeout.
Second: 10+ years ago libusb-0.1 was able to detect device in 0.01s, so timeout 0.3 is 30x times more. And that somebody later decided to implement a new usb library which is radically slower is not really TI or Nokia problem.

hjelmn, 2021-01-07:
> First: It is not Nokia, but TI and it is bootrom. Moreover it applies to all OMAP3 devices (maybe also OMAP4) and it is standard to not slowdown boot process by waiting longer time period. I do not [...]

It is not about slower. It is about correctness. Most USB devices CAN NOT be used before udev finishes processing them. We need to know when the device is ready for use. I am surprised it is taking this long in your case considering I have never seen it take that long with any of my devices. Most devices are detected almost instantaneously.

I doubt this has anything to do with the asynchronous nature of libusb. Considering this:

```
[ 5.173378] [00004171] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 5.173390] [00004171] libusb: debug [linux_get_device_address] scan 3-2
[ 5.175602] [00004171] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 1
[ 5.175615] [00004171] libusb: debug [udev_hotplug_event] udev hotplug event. action: remove.
```

libusb finds out about the device plug then unplug in ~ 23ms. That is much smaller than the 300ms the device is in the mode. That leaves the rest of the time in udev. libusb has a dedicated thread listending to udev events so they are processed as soon as the thread is woken up (much less than 1ms).

pali, 2021-01-07:
Ok, so issue is somewhere else? Do you need more logs for debugging then what I provided years ago in https://github.com/libusb/libusb/issues/237#issuecomment-265429119 ?

pali, 2021-01-08:
Sorry, currently I do not have a time for other tests. But as I provided all required information, including how to reproduce it, you can do it too.

tormodvolden, 2021-01-12:
As hjelmn hinted, this is a good way to see what udevd is doing:
```
sudo udevadm control --log-priority=debug
journalctl -f -o short-precise
```
In particular udevd is processing its rules compiled from {/lib,/etc}/udev/rules which might involve running a number of external processes (look for RUN and PROGRAM) in addition to many built-in commands. Since the introduction of "bind" events there can also be many rules that unnecessary are run twice. However I don't know if that affects the delay in notification to libusb.

Is there a trick to synchronize the relative timestamps from libusb with the absolute ones in the system log?

mcuee, 2021-01-17:
Here is what I get with a 3 year old Acer Swift 3, by plugging a USB to Serial Converter on an external USB 3 hub. Tried a few unplug and plug.

```
[ 2403.776674] usb 1-2.3: USB disconnect, device number 15
[ 2403.776674] usb 1-2.3: USB disconnect, device number 15
[ 2403.776674] usb 1-2.3: USB disconnect, device number 15
[ 2403.777357] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 2403.777357] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 2403.777393] ftdi_sio 1-2.3:1.0: device disconnected
[ 2403.777357] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 2403.777393] ftdi_sio 1-2.3:1.0: device disconnected
[ 2403.777393] ftdi_sio 1-2.3:1.0: device disconnected
[ 5.546662] [00008d2a] libusb: debug [linux_get_device_address] getting address for device: 1-2.3 detached: 0
[ 5.546680] [00008d2a] libusb: debug [linux_get_device_address] scan 1-2.3
[ 5.548496] [00008d2a] libusb: debug [linux_get_device_address] getting address for device: 1-2.3 detached: 1
[ 5.548516] [00008d2a] libusb: debug [udev_hotplug_event] udev hotplug event. action: remove.
[ 5.548558] [00008d29] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 5.548573] [00008d29] libusb: debug [handle_event_trigger] event triggered
[ 5.548582] [00008d29] libusb: debug [handle_event_trigger] hotplug message received
[ 5.548591] [00008d29] libusb: debug [libusb_get_device_descriptor]  
Unplugged: idVendor=0x0403 idProduct=0x6001
[ 5.548612] [00008d29] libusb: debug [libusb_unref_device] destroy device 1.15
[ 5.558728] [00008d29] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 5.558747] [00008d29] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
[ 2424.991352] usb 1-2.3: new full-speed USB device number 16 using xhci_hcd
[ 2424.991352] usb 1-2.3: new full-speed USB device number 16 using xhci_hcd
[ 2424.991352] usb 1-2.3: new full-speed USB device number 16 using xhci_hcd
[ 2425.098495] usb 1-2.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[ 2425.098495] usb 1-2.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[ 2425.098495] usb 1-2.3: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[ 2425.098501] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2425.098501] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2425.098504] usb 1-2.3: Product: FT232R USB UART
[ 2425.098504] usb 1-2.3: Product: FT232R USB UART
[ 2425.098507] usb 1-2.3: Manufacturer: FTDI
[ 2425.098501] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2425.098507] usb 1-2.3: Manufacturer: FTDI
[ 2425.098510] usb 1-2.3: SerialNumber: A8007Ub5
[ 2425.098504] usb 1-2.3: Product: FT232R USB UART
[ 2425.098510] usb 1-2.3: SerialNumber: A8007Ub5
[ 2425.098507] usb 1-2.3: Manufacturer: FTDI
[ 2425.098510] usb 1-2.3: SerialNumber: A8007Ub5
[ 2425.107653] ftdi_sio 1-2.3:1.0: FTDI USB Serial Device converter detected
[ 2425.107653] ftdi_sio 1-2.3:1.0: FTDI USB Serial Device converter detected
[ 2425.107653] ftdi_sio 1-2.3:1.0: FTDI USB Serial Device converter detected
[ 2425.107743] usb 1-2.3: Detected FT232RL
[ 2425.107743] usb 1-2.3: Detected FT232RL
[ 2425.107743] usb 1-2.3: Detected FT232RL
[ 2425.108803] usb 1-2.3: FTDI USB Serial Device converter now attached to ttyUSB1
[ 2425.108803] usb 1-2.3: FTDI USB Serial Device converter now attached to ttyUSB1
[ 2425.108803] usb 1-2.3: FTDI USB Serial Device converter now attached to ttyUSB1
[26.875124] [00008d2a] libusb: debug [linux_get_device_address] getting address for device: 1-2.3 detached: 0
[26.875142] [00008d2a] libusb: debug [linux_get_device_address] scan 1-2.3
[26.875202] [00008d2a] libusb: debug [linux_get_device_address] bus=1 dev=16
[26.875209] [00008d2a] libusb: debug [udev_hotplug_event] udev hotplug event. action: add.
[26.875213] [00008d2a] libusb: debug [linux_enumerate_device] busnum 1 devaddr 16 session_id 272
[26.875218] [00008d2a] libusb: debug [linux_enumerate_device] allocating new device for 1/16 (session 272)
[26.875260] [00008d2a] libusb: debug [linux_get_parent_info] dev 0x7fda280023d0 (1-2.3) has parent 0x55fdf3dab6a0 (1-2) port 3
[26.875292] [00008d29] libusb: debug [usbi_wait_for_events] poll() returned 1
[26.875304] [00008d29] libusb: debug [handle_event_trigger] event triggered
[26.875308] [00008d29] libusb: debug [handle_event_trigger] hotplug message received
[26.875312] [00008d29] libusb: debug [libusb_get_device_descriptor]  
[26.875317] [00008d29] libusb: debug [libusb_open] open 1.16
[26.875337] [00008d29] libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/016, errno=13
[26.875352] [00008d29] libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
[26.875357] [00008d29] libusb: debug [libusb_open] open 1.16 returns -3
Plugged: idVendor=0x0403 idProduct=0x6001 iProduct=
[26.885434] [00008d29] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[26.885450] [00008d29] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
[26.902637] [00008d2a] libusb: debug [linux_get_device_address] getting address for device: 1-2.3 detached: 0
[... 50 lines cut ...]
[ 2437.396694] usb 1-2.3: FTDI USB Serial Device converter now attached to ttyUSB1
[... cut ...]
[39.182326] [00008d29] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[39.182371] [00008d29] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
[39.219294] [00008d2a] libusb: debug [linux_get_device_address] getting address for device: 1-2.3 detached: 0
[39.219333] [00008d2a] libusb: debug [linux_get_device_address] scan 1-2.3
[39.219396] [00008d2a] libusb: debug [linux_get_device_address] bus=1 dev=17
[39.219399] [00008d2a] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.

```

mcuee, 2021-01-17:
This is what udev is doing with one unplug and one plug event.

```
mcuee@ubuntu2004:~/build/libusb/issue237$ sudo udevadm control --log-priority=debug
[sudo] password for mcuee: 

mcuee@ubuntu2004:~/build/libusb/issue237$ sudo journalctl -f -o short-precise
-- Logs begin at Tue 2020-12-22 16:54:16 +08. --
Jan 17 13:48:04.899727 ubuntu2004 mtp-probe[36187]: checking bus 1, device 17: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3"
Jan 17 13:48:04.899970 ubuntu2004 mtp-probe[36187]: bus: 1, device: 17 was not an MTP device
Jan 17 13:48:07.392549 ubuntu2004 ModemManager[991]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.3': not supported by any plugin
Jan 17 13:51:59.058559 ubuntu2004 sudo[36271]: pam_unix(sudo:auth): Couldn't open /etc/securetty: No such file or directory
Jan 17 13:52:03.333023 ubuntu2004 sudo[36271]: pam_unix(sudo:auth): Couldn't open /etc/securetty: No such file or directory
Jan 17 13:52:03.352286 ubuntu2004 sudo[36271]:    mcuee : TTY=pts/2 ; PWD=/home/mcuee/build/libusb/issue237 ; USER=root ; COMMAND=/usr/bin/udevadm control --log-priority=debug
Jan 17 13:52:03.353668 ubuntu2004 sudo[36271]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 17 13:52:03.362295 ubuntu2004 sudo[36271]: pam_unix(sudo:session): session closed for user root
Jan 17 13:52:40.584092 ubuntu2004 sudo[36295]:    mcuee : TTY=pts/2 ; PWD=/home/mcuee/build/libusb/issue237 ; USER=root ; COMMAND=/usr/bin/journalctl -f -o short-precise
Jan 17 13:52:40.584685 ubuntu2004 sudo[36295]: pam_unix(sudo:session): session opened for user root by (uid=0)
[ 2722.241383] usb 1-2.3: USB disconnect, device number 17
[ 2722.241383] usb 1-2.3: USB disconnect, device number 17
[ 2722.241383] usb 1-2.3: USB disconnect, device number 17
[ 2722.242104] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 2722.242104] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 2722.242161] ftdi_sio 1-2.3:1.0: device disconnected
[ 2722.242104] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 2722.242161] ftdi_sio 1-2.3:1.0: device disconnected
[ 2722.242161] ftdi_sio 1-2.3:1.0: device disconnected
Jan 17 13:52:49.686211 ubuntu2004 systemd-udevd[352]: ttyUSB1: Device (SEQNUM=8545, ACTION=remove) is queued
Jan 17 13:52:49.686310 ubuntu2004 systemd-udevd[352]: Validate module index
Jan 17 13:52:49.686375 ubuntu2004 systemd-udevd[352]: Check if link configuration needs reloading.
Jan 17 13:52:49.688121 ubuntu2004 kernel: usb 1-2.3: USB disconnect, device number 17
Jan 17 13:52:49.688452 ubuntu2004 kernel: ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
Jan 17 13:52:49.688624 ubuntu2004 kernel: ftdi_sio 1-2.3:1.0: device disconnected
Jan 17 13:52:49.686866 ubuntu2004 systemd-udevd[352]: Successfully forked off 'n/a' as PID 36298.
Jan 17 13:52:49.686999 ubuntu2004 systemd-udevd[352]: ttyUSB1: Worker [36298] is forked for processing SEQNUM=8545.
Jan 17 13:52:49.687298 ubuntu2004 systemd-udevd[352]: gpiochip508: Device (SEQNUM=8546, ACTION=remove) is queued
Jan 17 13:52:49.687625 ubuntu2004 systemd-udevd[36298]: ttyUSB1: Processing device (SEQNUM=8545, ACTION=remove)
Jan 17 13:52:49.687711 ubuntu2004 systemd-udevd[352]: Successfully forked off 'n/a' as PID 36299.
Jan 17 13:52:49.687870 ubuntu2004 systemd-udevd[352]: gpiochip508: Worker [36299] is forked for processing SEQNUM=8546.
Jan 17 13:52:49.688320 ubuntu2004 systemd-udevd[352]: gpiochip0: Device (SEQNUM=8547, ACTION=remove) is queued
Jan 17 13:52:49.688682 ubuntu2004 systemd-udevd[36298]: value '[dmi/id]sys_vendor' is 'Acer'
Jan 17 13:52:49.688707 ubuntu2004 systemd-udevd[352]: Successfully forked off 'n/a' as PID 36300.
Jan 17 13:52:49.688792 ubuntu2004 systemd-udevd[36299]: gpiochip508: Processing device (SEQNUM=8546, ACTION=remove)
Jan 17 13:52:49.688844 ubuntu2004 systemd-udevd[352]: gpiochip0: Worker [36300] is forked for processing SEQNUM=8547.
Jan 17 13:52:49.688946 ubuntu2004 systemd-udevd[36298]: value '[dmi/id]sys_vendor' is 'Acer'
Jan 17 13:52:49.689094 ubuntu2004 systemd-udevd[352]: ttyUSB1: Device (SEQNUM=8548, ACTION=unbind) is queued
Jan 17 13:52:49.689281 ubuntu2004 systemd-udevd[352]: ttyUSB1: Device (SEQNUM=8549, ACTION=remove) is queued
Jan 17 13:52:49.689430 ubuntu2004 systemd-udevd[36299]: value '[dmi/id]sys_vendor' is 'Acer'
Jan 17 13:52:49.689512 ubuntu2004 systemd-udevd[352]: 1-2.3:1.0: Device (SEQNUM=8550, ACTION=unbind) is queued
Jan 17 13:52:49.689708 ubuntu2004 systemd-udevd[352]: 1-2.3:1.0: Device (SEQNUM=8551, ACTION=remove) is queued
Jan 17 13:52:49.689714 ubuntu2004 systemd-udevd[36300]: gpiochip0: Processing device (SEQNUM=8547, ACTION=remove)
Jan 17 13:52:49.689844 ubuntu2004 systemd-udevd[36299]: value '[dmi/id]sys_vendor' is 'Acer'
Jan 17 13:52:49.689913 ubuntu2004 systemd-udevd[352]: 1-2.3: Device (SEQNUM=8552, ACTION=unbind) is queued
Jan 17 13:52:49.690105 ubuntu2004 systemd-udevd[352]: 1-2.3: Device (SEQNUM=8553, ACTION=remove) is queued
Jan 17 13:52:49.690213 ubuntu2004 systemd-udevd[36300]: value '[dmi/id]sys_vendor' is 'Acer'
Jan 17 13:52:49.690506 ubuntu2004 systemd-udevd[36300]: value '[dmi/id]sys_vendor' is 'Acer'
Jan 17 13:52:49.691713 ubuntu2004 systemd-udevd[36298]: ttyUSB1: No reference left, removing '/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A8007Ub5-if00-port0'
[... cut ...]
Jan 17 13:52:56.045685 ubuntu2004 systemd-udevd[36300]: Unloaded link configuration context.
Jan 17 13:52:56.045696 ubuntu2004 systemd-udevd[36299]: Unloaded link configuration context.
Jan 17 13:52:56.045758 ubuntu2004 systemd-udevd[36298]: Unloaded link configuration context.
Jan 17 13:52:56.050920 ubuntu2004 systemd-udevd[352]: Worker [36298] exited
Jan 17 13:52:56.051127 ubuntu2004 systemd-udevd[352]: Worker [36299] exited
Jan 17 13:52:56.051151 ubuntu2004 systemd-udevd[352]: Worker [36300] exited

```

mcuee, 2021-01-18:
This is an interesting feature if it is possible.
>Is there a trick to synchronize the relative timestamps from libusb with the absolute ones in the system log?

[55 more replies]