libusb/libusb #239: libusb_transfer buffer type is `unsigned char *`, convert to

kuldeepdhaka, 2016-12-11:
```
struct libusb_transfer {
	/** Handle of the device that this transfer will be submitted to */
	libusb_device_handle *dev_handle;
	... <snip> ....
	
	/** Data buffer */
	unsigned char *buffer;

	/** Number of isochronous packets. Only used for I/O with isochronous
	 * endpoints. */
     int num_iso_packets;
}
```

why is `unsigned char *buffer;` being used, why not just `void *buffer`.
its always a headache to cast a buffer to `unsigned char *` or compiler issue a warning.
also `unsigned char *` add nothing special.
Also, moving to `void *` will have no compatibility issue.

imaami, 2021-02-15:
This issue is just incorrect. `unsigned char *` is indeed very special on a fundamental level. The C language standard explicitly says a thing or two about memory addressing and pointer types. Please look into it.

mcuee, 2021-04-06:
So we should just close this, right? Any objections?

ljluestc, 2023-06-23:
changing the data type of buffer from unsigned char * to void * should not introduce compatibility issues