libusb/libusb #240: Find device type

Waishon, 2016-12-12:
Hi,

whats the proper way to check if a connected usb device is a keyboard or a mouse?

It should be possible to check where it is mounted, but is it also makeable with libusb?

diabolo38, 2016-12-12:
usb keyboard/mice are HID class  device inspecting interface/device  descriptor class/subclass/prootcol field tell  you what they are . But deeper HID descriptor analyses may be required to find out if it is "a regular keyboard"  and/or mousse  or just a key in an headset etc ..

i'm scared your requirement is out of lib scope.

mcuee, 2016-12-13:
I think this is a question and it is better to ask in libusb mailing list. You may need to provide more details of what you want to achieve and what is the OS of concern.

mcuee, 2017-01-04:
Close this issue here. If the OP wants to pursue further, please ask in the mailing list.

closed 2017-01-04 by mcuee