libusb/libusb #241: Ideas on Intermittent Pipe error [Errno 32]

c1728p9, 2016-12-12:
I'm getting an intermittent pipe error when using libusb 1.0.20 on Ubuntu 16.04. I also tried the latest code from master - b14d0a49e8014c0043673160ace2910e5742a1d2 and had this same problem. The code I'm using communicates over interrupt endpoints and as a stress test does this repeatedly to multiple devices. The test I'm using accesses libusb through the python library pyusb.

I also capture this event on a USB analyzer, and the interrupt endpoints in use are not stalled (the API indicates that libusb_interrupt_transfer returns LIBUSB_ERROR_PIPE if the endpoint is halted). It appears that the transaction is getting canceled, since in the USB logs show an IN transaction is started (log has SSPLIT IN) but is never finished (no CSPLIT OUT).

Any ideas on how I could narrow this problem down? I set LIBUSB_DEBUG=2 but I don't see anything when the error occurs. To see if pyusb was the problem I ran the same script on Windows 10 using libusb 0.1 (I couldn't get libusb 1.0 to work) and didn't have any problems.  I know python masks off some signals, is it possible this would have an effect on libusb? Also any recommendations on how I could debug this?

mcuee, 2016-12-13:
Again this is a question better asked in the mailing list where you may get better support.

Anyway, you can try to set LIBUSB_DEBUG=4 to get more verbose debug message. As for pyusb related issues you can ask in the pyusb mailing list. Basically you need to have the supported driver (say WinUSB) and you need to put libusb-1.0.dll in the place Python can find it.

c1728p9, 2016-12-13:
Rewrote the stress test in C and was able to capture a log with LIBUSB_DEBUG=4. I send this info to the mailing list, so I'm closing this issue.

c1728p9, 2016-12-14:
@mcuee I forgot to mention it, but I opened a pyusb issue as well here: https://github.com/walac/pyusb/issues/162

mcuee, 2016-12-14:
Thanks. I do not see anything in the mailing list. You need to subscribe to the mailing list in order to post.

mcuee, 2016-12-14:
You can also post the C debug log here.

mcuee, 2016-12-14:
https://github.com/libusb/libusb/wiki/Windows#known-restrictions
"With WinUSB, when LIBUSB_RECIPIENT_INTERFACE is used for the transfer, the WinUSB DLL forces the low byte of wIndex to the interface number, regardless of what you set it to. "

I am not so sure if you hit the limitation of WinUSB or not. Probably you can switch to libusbK driver to see if that helps.

mcuee, 2016-12-14:
So re-open this issue since we need to see if this is a real bug or not.

c1728p9, 2016-12-14:
Looks like my initial email is on hold - Your email "Is being held until the list moderator can review it for approval.". I subscribed to the mailing list but I haven't gotten a confirmation email yet. Below is the log: 
[log_shortened.txt](https://github.com/libusb/libusb/files/652771/log_shortened.txt)

The pipe error captured in the log is here:
```
[2247.308788] [00004f30] libusb: debug [reap_for_handle] urb type=1 status=-32 transferred=0
[2247.308789] [00004f30] libusb: debug [handle_bulk_completion] handling completion status -32 of bulk urb 1/1
[2247.308790] [00004f30] libusb: debug [handle_bulk_completion] detected endpoint stall
[2247.308791] [00004f30] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 10000ms
[2247.308793] [00004f30] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f41f80009d8 has callback 0x7f4206da4520
[2247.308794] [00004f30] libusb: debug [sync_transfer_cb] actual_length=0
```

mcuee, 2016-12-15:
Changed the label -- the pyusb ticket is for Windows but this issue report is a bit different since it is for Ubuntu Linux.

mcuee, 2016-12-31:
This is also being discussed in the mailing list -- Alan Stern is suggesting the OP to capture the usbmon trace to further debug the issue. This may not be a libusb issue, could be a kernel issue as well.

mcuee, 2017-01-24:
Any updates on this? If not, may close this issue within another month.

c1728p9, 2017-01-25:
Hi @mcuee, no major updates on this yet. I no longer have the hub I tested this on, but I should get another soon. When the same devices (full speed) are used directly with the host computer's hub (not multi-TT), USB is far more unstable. If I use a high speed device I don't see any problems, even running the same test non-stop for several days.

c1728p9, 2017-02-04:
@mcuee got another good hub and started testing again. I was able to reproduce it and sent a usbmon log to the mailing list.

mcuee, 2017-02-07:
As discussed in the mailing list, Alan Stern thinks this is not related to libusb but rather the kernel. So I will close this ticket here. Will reopen if it turns out to be related to libusb later.

closed 2017-02-07 by mcuee

c1728p9, 2017-02-07:
Sounds good. Thanks for the help investigating this.