libusb/libusb #244: Unexpected cancelled transfers

ecejeff, 2016-12-20:
Running on Windows 7 x64, with a 32-bit DLL. Using libusb 1.0.21 nano version 11156.

I've got transfers running on the following endpoints:

Endpoint 0x83, 13 transfers
Endpoint 0x86, 11 transfers
Endpoint 0x87, 1 transfer

They all run happily, returning good data and being resubmitted for as long as I care to run them.

When I cancel all transfers on EP 0x86 (in rapid succession), I will usually (though not always) experience cancelled transfers (i.e. transfer->status reads LIBUSB_TRANSFER_CANCELLED) on the other endpoints, despite the fact that libusb_cancel_transfer is not called on these transfers. 

Sometimes it's all other transfers. Sometimes only a subset. Sometimes it's none at all.

My current workaround is to resubmit any transfers that weren't actually canceled by me. This isn't perfect though, as any partially collected data goes missing.

mcuee, 2016-12-30:
This is a bit strange. Can you post the USB Descriptors of your device? And it is good that you could post the code snippets and the libusb debug message as well (set LIBUSB_DEBUG=4).

mcuee, 2016-12-31:
BTW, is this a Windows specific problem? Just wondering if you can test under Linux or Mac OS X.

ecejeff, 2017-01-13:
Device Descriptor (using xusb.exe):

```
c:\Libraries\libusb>C:\Libraries\libusb\Win32\Debug\examples\xusb.exe abcd:001f
Using libusb v1.0.21.11156

Opening device ABCD:001F...
libusb: info [windows_get_device_list] The following device has no driver: '\\.\
USB#VID_1AB1&PID_0E11#DP8C163352109'
libusb: info [windows_get_device_list] libusb will not be able to access it.

Reading device descriptor:
            length: 18
      device class: 255
               S/N: 3
           VID:PID: ABCD:001F
         bcdDevice: 0102
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 7
   Class.SubClass.Protocol: 00.00.00
       endpoint[0].address: 81
           max packet size: 0020
          polling interval: 08
       endpoint[1].address: 02
           max packet size: 0020
          polling interval: 08
       endpoint[2].address: 83
           max packet size: 0020
          polling interval: 01
       endpoint[3].address: 84
           max packet size: 0020
          polling interval: 04
       endpoint[4].address: 05
           max packet size: 0020
          polling interval: 04
       endpoint[5].address: 86
           max packet size: 0020
          polling interval: 01
       endpoint[6].address: 87
           max packet size: 0008
          polling interval: 01

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Suprock Technologies"
   String (0x02): "Wireless Module Receiver"
   String (0x03): "WMR1128"
   String (0xEE): "MSFT100 "

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):
   Failed: Pipe error
Releasing interface 0...
Closing device...
```

Log, with additional messages on libusb_cancel_transfer and when the spurious cancellations happen:
[... 764 lines cut ...]
[ 9.178000] [0000259c] libusb: debug [usbi_handle_transfer_completion] transfer 00908B6C has callback 049E2760
[ 9.179000] [0000259c] libusb: debug [libusb_submit_transfer] transfer 00908B6C
[ 9.179000] [0000259c] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 83 with interface 0
[ 9.179000] [0000259c] libusb: debug [winusbx_submit_bulk_transfer] reading 128 bytes
[ 9.179000] [0000259c] libusb: debug [usbi_add_pollfd] add fd 8 events 1
spurious cancel on transfer 009082D4
[ 9.179000] [0000259c] libusb: debug [libusb_submit_transfer] transfer 009082D4
[ 9.179000] [0000259c] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 83 with interface 0
[ 9.179000] [0000259c] libusb: debug [winusbx_submit_bulk_transfer] reading 128 bytes
[ 9.179000] [0000259c] libusb: debug [usbi_add_pollfd] add fd 10 events 1
spurious cancel on transfer 0093A06C
[ 9.179000] [0000259c] libusb: debug [libusb_submit_transfer] transfer 0093A06C
[ 9.179000] [0000259c] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 83 with interface 0
[ 9.179000] [0000259c] libusb: debug [winusbx_submit_bulk_transfer] reading 128 bytes
[ 9.179000] [0000259c] libusb: debug [usbi_add_pollfd] add fd 14 events 1
[ 9.179000] [0000259c] libusb: debug [libusb_get_next_timeout] next timeout in 0.659401s
[ 9.179000] [000025d0] libusb: debug [libusb_submit_transfer] transfer 0093AA44
[ 9.179000] [0000259c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 9.179000] [000025d0] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 02 with interface 0
[ 9.179000] [0000259c] libusb: debug [handle_events] poll fds modified, reallocating
[ 9.179000] [000025d0] libusb: debug [winusbx_submit_bulk_transfer] writing 2 bytes
[ 9.179000] [0000259c] libusb: debug [handle_events] poll() 15 fds with timeout in 100ms
[ 9.179000] [000025d0] libusb: debug [usbi_add_pollfd] add fd 15 events 4
[ 9.179000] [0000259c] libusb: debug [handle_events] poll() returned 1
```

ecejeff, 2017-02-16:
I noticed something on the mailing list recently, in an unrelated discussion.

From Peter Stuge on the libusb-devel mailing list (**bold mine**):
> Successfully submitted transfers are stored by libusb in a list, if
you cancel them before they are processed by the host controller then
they are simply ignored (**remember to cancel them in reverse order to
avoid race conditions**).  The callback is called also for cancelled
transfers.

I am cancelling callbacks in the same order that I originally submitted them (though as libusb sometimes returns callbacks out of order as per bug #126, there is no guarantee that the order for the last resubmit is the same as the round of submits). Could the cancellation order be related to the problem I am seeing? If so, is this behavior documented anywhere?

UniversalSerialBummer, 2018-03-09:
I am also experiencing unexpected cancels. In my case they occur when i queue transfers on a second pipe while sending and reusing transfers (as fast as possible, USB3 heavy bulk data load.) on the first.

I have traced the winusb driver and it reports no errors, just a cancel that i did not ask for.
Is this a race in libusb perhaps ?

dickens, 2020-08-18:
Now that transfers on Windows are individually cancelled, I don't think this should still be an issue.  Can anyone on this thread confirm?

ecejeff, 2020-08-18:
Seems to have gone away. 

I never saw this bug on Windows 10, but updating to the latest git did fix the issue on my Windows 7 machine.

mcuee, 2021-01-23:
Close this for now.

closed 2021-01-23 by mcuee