libusb/libusb #247: add usbvideo.sys support?

mattapperson, 2016-12-29:
I am trying to control cameras (Logitech c930e's) that are UVC compatible. Everything works fine on OS X but on windows, installing WinUSB just results in nothing working.
It seems to me, that adding support to LibUSB for usbvideo.sys driver based devices would make this library a whole lot more "cross platform" as this is based on my limited understanding the sort of default driver used in windows for UVC webcams.

Am I barking up the wrong tree? I am just trying to add support to my app to control a set of ptz presets while viewing and streaming a webcam...

mattapperson, 2016-12-29:
BTW, I know this is sort of a crappy issue to file... but I think I am on the right track, just not sure... any feedback would be great, even if it is just to tell me I'm an idiot lol

mcuee, 2016-12-30:
I think you should look at the new usbdk Windows backend inside libusb 1.0.21 release. It will not use WinUSB and the original camera function will still work.

mcuee, 2016-12-30:
usbvideo.sys is not a generic driver (WinUSB and libusbk are generic USB drivers) and it may well not provide the things libusb required. There are quite some compromises adding the generic HID device support already inside the Windows backend.

mcuee, 2017-01-04:
I will close this ticket as I do not think it is feasible to add a usbvideo.sys backend. usbdk backend could be a potential solution any way.

closed 2017-01-04 by mcuee