libusb/libusb #250: usbdk_get_device_list randomly fails (which causes libusb to fail too)

leoetlino, 2017-01-03:
(This was originally reported in #232, but I'm not sure it's actually related.)

usbdk_get_device_list randomly fails to get a device list, which obviously causes libusb to fail as well. It seems to be unable to open the usbdk driver (0x5 is access denied AFAIK):

>Driver file operation error. Failed to open device symlink \.\UsbDk (Error code = 0x5)"

The weird thing is that this happens randomly and without any visible patterns. It could work perfectly without a single failure, or fail then succeed if you try again immediately, only to fail again randomly some time later, *or* it could just keep failing every single time.

This is a really big issue for what we plan to use the usbdk backend for in [Dolphin](https://github.com/dolphin-emu/dolphin/pull/4408), because some games require the very first device list (populated with libusb_get_device_list) to contain all devices the player wants to use, and ignore any later ones. So the first call to get a device list must *succeed*, or the game will be unplayable.

[Dolphin devs](https://www.reddit.com/r/emulation/comments/5k8ho4/looking_for_feedback_potentially_changing_usb/) are willing to switch from the WinUSB to the usbdk backend to get complete USB support on Windows, but this issue is really a blocker, along with #232 (seemingly random crashes with transfers; control or isochronous).

mcuee, 2017-01-04:
I notice that during testing of usbdk (under Windows 10) there are system sounds which I am not sure but may indicate the original driver being detached and reattached -- this may be the reason you see this if you require fast device list. Is it okay for Dolphin to retry the device list a few times?

mcuee, 2017-01-04:
I myself have not encounter this issue yet. I only see the error message when usbdk is disabled which is the expected behavior.

c:\Program Files\UsbDk Runtime Library>UsbDkController.exe -u
Uninstalling UsbDk driver
UsbDk driver uninstall succeeded

c:\Program Files\UsbDk Runtime Library>UsbDkController.exe -n
Enumerate USB devices
Driver file operation error. Failed to open device symlink \\.\UsbDk
(The system cannot find the file specified.  Error code = 2)
Enumeration failed

leoetlino, 2017-01-04:
Hmm, that could be a workaround. Keep calling libusb_get_device_list until it works. Alternatively… does libusb's usbdk backend support hotplugging? If it does, we won't have to poll and check for new devices manually.

Error 2 is "not found"; getting it when trying to open the usbdk driver seems to make sense if usbdk is not installed. However, the one we get is error 5, which is "access denied", and I don't understand why that randomly happens…

mcuee, 2017-01-04:
I see. Unfortunately hotplug is not supported yet for libusb WinUSB/libusbK/HID backend. I think it is the same for usbdk backend.

leoetlino, 2017-01-04:
We now have a workaround for usbdk_get_device_list, but the other issue is still a very major one we can't seem to fix or avoid: libusb is crashing Dolphin when submitting a transfer (#232). More specifically, an isochronous transfer for a camera game which seems to use large transfer packets; it works totally fine on Linux but instantly crashes libusb when using usbdk.

@dmitryfleytman do you have any idea as to how to fix that issue?

dmitryfleytman, 2017-01-05:
@leoetlino We are looking into this, but it will greatly help us if you provide more info:

- check where exactly it crashes, as I asked in https://github.com/libusb/libusb/issues/232#issuecomment-267264967
- provide exact scenario where this crash occurs, i.e. how do you run dolphin, which game do you run,  etc.

Thanks,
Dmitry

dmitryfleytman, 2017-01-05:
@leoetlino, error 5, "access denied" on device enumeration happens because UsbDk is already used by another thread/process.

leoetlino, 2017-01-05:
Was the trace useful? https://github.com/libusb/libusb/files/672199/UsbDkTrace.zip

The crash reported in #232 can be reproduced very simply, by using the [usbdk build of Dolphin](http://dl.dolphin-emu.org/prs/pr-4465-dolphin-latest-dbg-x64.7z) (here's the [source PR](https://dolp.in/pr4465) if needed), plugging in the device and launching the game. Dolphin will then shortly crash with usbdk. (It doesn't crash with WinUSB/Linux.)

You do need to have a specific game (Your Shape) + specific accessory (the game's bundled USB camera) to reproduce the crash related to large isochronous transfers (can't remember the size of each packet, but they were larger than usual in the camera games)

However, testers have also reported crashes when submitting control transfers/regular isochronous transfers for microphones. If you have a Logitech microphone (046d:0a03) and a Wii game which uses the microphone, you can use it to try isochronous transfers. If not, WiiMC (linked in the OP) and a USB drive should be enough to test control/bulk transfers.

---

Thanks for the hint for error 5! I was going to say that Dolphin only uses UsbDk from one thread, but you made me realise that we were actually using libusb concurrently from several threads (for controller passthrough features). We are using different libusb contexts; is that just not a supported use case?

dmitryfleytman, 2017-01-08:
> 
> Was the trace useful? https://github.com/libusb/libusb/files/672199/UsbDkTrace.zip <https://github.com/libusb/libusb/files/672199/UsbDkTrace.zip>
Partially, it shows that the issue in on user mode level, everything is ok from UsbDk driver’s point of view.

> The crash reported in #232 <https://github.com/libusb/libusb/issues/232> can be reproduced very simply, by using the usbdk build of Dolphin [...]

Ok, we will try to reproduce the problem and then get back to you.

leoetlino, 2017-01-09:
Thanks for looking into it! A small question though: is it allowed to use several libusb contexts and enumerate devices from several (pretty much independant) threads? That currently fails with -5 (access denied) with UsbDk, but it appeared to work fine with WinUSB (and Linux)

dmitryfleytman, 2017-01-15:
> 
> Thanks for looking into it! A small question though: is it allowed to use several libusb contexts and enumerate devices from several (pretty much independant) threads? That currently fails with -5 (access denied) with UsbDk, but it appeared to work fine with WinUSB (and Linux)

Currently UsbDk control device may be opened exclusively only, therefore you’re getting error while trying to open it again from different thread.
I would suggest sharing the same libusb context between your application threads.

leoetlino, 2017-01-15:
I see, thanks for the advice! This should work good enough for our purposes.

Since this is not a libusb issue, but a limitation in UsbDk, I'm closing this.

mcuee, 2017-01-24:
@leoetlino Are you okay to close this ticket or you want to leave it open?

leoetlino, 2017-01-24:
There are still some failures even when using a single libusb context, mostly when enumerating devices from two separate threads, which is why I reopened the issue, but it's not very important as we've kept the workaround. Feel free to close this issue if you want.

mcuee, 2017-01-24:
I see. Will leave this issue open for now.

goldendeng, 2018-02-06:
@leoetlino 
I have the same problem, but I use the same libusb context in different threads, usdk still has the same mistake. How do you handle it?

leoetlino, 2018-02-06:
Just ended up calling it several times until it works.

leoetlino, 2019-05-21:
Closing this again because it appears the thread safety issue with device enumeration is not related to the usbdk code, but rather to libusb (possibly only the Windows specific parts?) itself.

avsdev-cw, 2020-03-04:
Not entirely sure this thread should of been closed. I've been hitting my head against issues with libuvc + libusb + usbdk for a while and this does seem to be an issue with libusb/usbdk of compatibility between the 2.

The following exception led me to this thread, and the behavior is exactly the same, and funnily enough, I'm using multiple threads. Unfortunately, passing the USB context around is a bit of an issue for me since the additional threads are actually DLL plugin imports.

```
Exception at 0x7ff976aaa839, code: 0xe06d7363: C++ exception, flags=0x1 (execution cannot be continued) (first chance) in UsbDkHelper
Driver file operation error. Failed to open device symlink \\.\UsbDk (Access is denied.  Error code = 5)
```

I'm willing to open another thread on this if preferred, but a deeper look/better solution than just thread bashing usbdk til it works should really be sorted.

avsdev-cw, 2020-03-04:
Further to my previous comment, I pulled in 2 of the plugins and modified them to accept a libusb_context * and a mutex guard. It seems that even sharing a single libusb_context across the threads does not fix the issue.

Additionally, if you get one of the devices to start streaming, the calls to open the other start to get blocked in the calls to fetch a list of devices/open a device. (regardless of individual contexts or shared context).

Worth mentioning, as has been before, that this is explicitly a windows-only issue (the same code on linux works a treat - the only difference is a macro-guarded libusb_set_option() call to use usbdk)

leoetlino, 2020-03-04:
Alright, I'll reopen this issue then. FWIW, we ended up dropping support for usbdk and using several contexts with WinUSB after we identified and fixed bugs that prevented the WinUSB backend from working.

avsdev-cw, 2020-03-04:
Thanks. Did you end up sticking with libusb or writing your own interface to WinUSB using that API?

leoetlino, 2020-03-04:
We are still using libusb as it lets us avoid having to write platform specific code. libusb works well enough on Linux and Windows as far as Dolphin is concerned.

mcuee, 2023-02-17:
@leoetlino and @avsdev-cw 

Just wonder if you can test libusb-1.0.26 release and libusb git HEAD to see if the issues are still there. Thanks.

avsdev-cw, 2023-02-17:
Unfortunately the project we were using libusb in has been dropped and we no longer have the test environment running, sorry!

mcuee, 2023-02-17:
> Unfortunately the project we were using libusb in has been dropped and we no longer have the test environment running, sorry!

No problem. Thanks for the reply.

leoetlino, 2023-02-17:
@mcuee Dolphin has been using 1.0.26 since April 2022 but I think the majority of users are not using the usbdk backend. Sadly we haven't been able to recommend it for a while because of general instability issues and even reports of blue screens iirc.

As I am a Linux user and our main tester has recently switched (from Windows) to Linux as well, I think it would be complicated for us to test this... sorry. I'll assume that the problem has been fixed and close this issue.

closed 2023-02-17 by leoetlino

mcuee, 2023-02-17:
@leoetlino 

Thanks a lot for the comments.