libusb/libusb #251: libusb doesn't find virtual USB devices

nickich, 2017-01-05:
I'm using Eltima Network Gate to access remote devices, but libusb doesn't find them.
When I run listdevs, the virtual devices came up with Linux VID/PID. I was looking at 'windows_winusb.c' and it seems that it thinks the devices are hubs...

mcuee, 2017-01-06:
I think this is the correct behavior since it acts a virtual USB hub for the network device. Why do you think it is a bug? What is the issue you are facing?

nickich, 2017-01-06:
It's actually a virtual USB device. The libusb thinks it is a hub and since it can't get its VID/PID, libusb assigns Linux ones. 
Same virtual device is recognised and enumetated properly by other sw.

mcuee, 2017-01-09:
No I do not think this is true. It is not a virtual USB device. Rather those remote USB device connect to it is a virtual USB device. Itself is a virtual HUB. If it is a USB device, virtual or real, it will have a USB VID/PID.

Ref: http://www.eltima.com/products/usb-over-ethernet/

nickich, 2017-01-11:
The remote device connects as a virtual device on a local machine. This virtual device has PID/VID, but the libusb doesn't recognize it as such.

mcuee, 2017-01-12:
Please refer to Pull Request 257. https://github.com/libusb/libusb/pull/257

closed 2018-01-05 by dickens