libusb/libusb #252: Add native event handling support

mcuee, 2017-01-07:
Ref: https://sourceforge.net/p/libusb/mailman/message/34083803/

From Chris.

This patch series introduces an abstraction layer into the libusb core that
allows different platforms to use native constructs to achieve the required
functionality. Two examples of these constructs are event notification
mechanisms and programmable timers that provide notification of expiration.

The highlights of this patch series are as follows:

  * Completely eliminate all POSIX emulation for the Windows/WinCE backends
  * Add support for eventfd on Linux platforms
  * Enable the use of libusb_get_pollfds() and friends on Windows/WinCE
  * Does not break ABI for any currently existing applications [1]

dickens, 2017-05-04:
I wanted to open a quick discussion about this patch series. As I have been reworking it, I am growing less fond of having libusb_get_pollfds() work on Windows/WinCE. I have two issues with this:

1) The name--both 'poll' and 'fds' are POSIX/UNIX notions.
2) The libusb_pollfd structure (and also its name)--on 64-bit Windows builds, the size of this structure would suddenly change. As I noted in the original patch series, this shouldn't really cause issues as no Windows user could have used this API in the past.

I would like to propose adding a new set of functions using the naming convention of 'event source', as that is what these things truly are. This would give us:

libusb_event_sources_handle_timeouts()
libusb_get_event_sources()
libusb_free_event_sources()
libusb_set_event_source_notifiers()

There will also be a new libusb_event_source structure to go along with these functions.

We may also want to consider marking the previous functions as deprecated.

I'm at a fork in the road with reworking the patch series, so maintainers please chime in with your opinions.

hjelmn, 2017-05-04:
I think we probably should increase the minor version if we deprecate existing functions. Is that the plan?

dickens, 2017-06-06:
By deprecate I mean mark with an attribute, same as libusb_get_port_path(). But I agree, it may be time to go with 1.1.0 given the breadth of these changes.

Maintainers, do you lean one way or the other in terms of this API addition? I'd like to get this in soon and kickstart the next release.

commit d67eb5be (2020-08-13): core: Introduce platform events abstraction

The way in which system handles or resources are represented differs
greatly between Unix-like operating systems and Windows. Ever since
Windows support was added to libusb, Windows been emulating principles
of Unix-like operating systems such as file descriptors and poll().

This commit introduces an abstraction layer that completely removes the
need to perform any emulation. Fundamentally there are three things that
each platform provides to libusb:

  1) A signallable event
  2) A timer (not required, but useful)
  3) A means to wait for event so [... cut]
closed 2020-08-13 by dickens