libusb/libusb #253: Feature request: .NET wrapper library

andrasfuchs, 2017-01-10:
The DotNetLibUsb project last release (v2.2.8) was on 2010-10-29, so it does not reflect the more recent changes in libusb. 
Also when I looked into its source code, it looks like it uses a few platform dependent DllImports, like kernel32.dll, setupapi.dll, advapi32.dll besides the libusb-1.0.dll and winusb.dll user-mode libraries.
If I understand it correctly, libusb is an abstraction layer and it is capable to use the winusb, libusb-win32 and libusbK kernel-mode drivers, so if it had a wrapper, it should be enough to wrap libusb without referencing the drivers and other platform dependent DLLs.
According to StackOverflow it is still DotNetLibUsb which is most commonly recommended as an library to access USB devices, so this, frequently updated libusb remains invisible to the .NET developers.
If it is needed, I would be glad to help to implement the wrapper, but before I begin, please let me know if I missed something and there is an easy way to work with libusb in the .NET environment.
If you are satisfied with the quality of the wrapper, would you consider to accept the pull request and include it in the future releases of libusb?

mcuee, 2017-01-10:
Just take note libusbdotnet is not part of libusb project. If you are keen,
you may want to talk to the author Travis on the improvement. I doubt he
has time for this project any more. So you may want to fork the project or
take over the project if you are serious about the project.

mcuee, 2017-01-11:
BTW, you may want to read the document before jumping into the
project. http://libusbdotnet.sourceforge.net/V2/Index.html

mcuee, 2017-01-11:
Please refer to ticket 224 as well.
https://github.com/libusb/libusb/issues/224

andrasfuchs, 2017-01-12:
Ok, I'll wait for the reply from Travis then (I've contacted him on SourceForge). 

Hmm, it is stated in #224 that the wrapper would not be part of the libusb project in the end. That's a shame, because it would reach much more people that way.

The .NET Core implementation is a very good idea, I'd support that.

I'll get back to you as soon as I could discuss this with Travis.

mcuee, 2017-01-12:
Thanks. In general, the language wrappers are not included in the libusb project. 
BTW, you may want to contact Travis directly through email ( libusbdotnet AT gmail.com). To tell you the truth, I myself did not like the way Sourceforge provided to contract the author.

andrasfuchs, 2017-01-12:
Thank you for his email address, I forwarded my message there.

Do you see a chance to at least include the compiled wrapper DLL in the windows binaries package (even if it's in a different repo)?

mcuee, 2017-01-12:
I do not think so -- at least not in the Sourceforge download. You know, in reality it is not fair to just bundle the dotnet wrapper, there are quite some language wrappers out there, for example pyusb is very popular. It is out of the scope for libusb project.

mcuee, 2017-01-12:
BTW, libusbdotnet is quite old, the last release in on 29-Oct-2010. It kind of works but it probably needs major efforts to revamp -- just a warning.

andrasfuchs, 2017-01-12:
Yes, I don't think dotnetlibusb should be included, mainly because it's so old. 
As I mentioned in my first comment here, I think it would be more appropriate to create a new wrapper from scratch, which is lightweight, but reflects all the recent changes in libusb (and probably based on .NET Core). 
The end result would be just one or two DLLs, nothing else. The C# source would be stored in a separate repo.

Anyway, if Travis replies, I'll probably talk with @mobidev111 and @TAGC who were in the #224 thread.

TAGC, 2017-01-12:
Incidentally, I just happened to open up a [new question on StackOverflow](http://stackoverflow.com/questions/41610802/cross-platform-usb-communication-using-asp-net-core) related to this concern a short while ago.

andrasfuchs, 2017-01-13:
@TAGC Wow, that's a pro question :) I only use a self-hosted .NET Framework app on Windows to communicate with my microcontroller, so I can't help you with that unfortunately.

What I can tell so far that you can find the libusb-1.0.dll's interop parts in LibUsbDotNet's source in the '\Src\LibWinUsb\MonoLibUsb\MonoLibUsbApi.cs' file. I would compare that with the public methods of the most recent libusb. The header file with the methods is in this repo, [here](https://github.com/libusb/libusb/blob/master/libusb/libusb.h).
I can image that the framework refuses to load the DLL if the signatures are different (and they probably are, because MonoLibUsbAps.cs was last refreshed in 2010).

mcuee, 2017-01-13:
So looks like there is an effort here.
https://www.nuget.org/packages/CoreCompat.LibUsbDotNet/2.2.8-r101
It is stated that 

> "This is a prerelease version of LibUsbDotNet for .NET Core".

andrasfuchs, 2017-01-13:
Thanks @mcuee for the link!

I'm not sure if the user (@quamotion) is only the publisher or he also maintains the package, but I invited him to join our discussion here anyway.

TAGC, 2017-01-13:
@mcuee That's the library I've tried using for cross-platform USB communication since months ago, and it doesn't seem to work on Linux.

mcuee, 2017-01-14:
I mentioned the pre-release port as well in one of my mailing list post as well. But there seems no interests in the mailing list.
Ref: http://marc.info/?l=libusb-devel&m=148031035826936&w=2

andrasfuchs, 2017-01-15:
I can confirm that neither the original nor the core version of LibUsbDontNet works :/

I've also tested to the two other options on Windows: libusbK and WinUSBNet.

* libusbK was also created by Travis. It was last updated on 2014-04-27, it supports only its own kernel-mode driver and Windows.
* WinUSBNet is well structured and more .NET-conform in naming. It was created by @madwizard-thomas and it's still active. It supports WinUSB only on Windows.

If I don't get a reply from Travis, I will spend a day next week investigating the possibility of creating a new wrapper from scratch which would be a one-to-one lightweight wrapper for the current libusb, so it would have all its features and its platform-independence.
My guess is that as long as it is kept simple it would work with both .NET Framework and .NET Core on Windows and Linux.

mcuee, 2017-01-15:
I am not so sure what you mean when you say libusbdotnet does not work. It should work. It is a bit outdated but it should work under Windows/Linux/Mac OS X. It does not support .Net Core but it should work under Linux and Mac OS X under Mono. I just tried libusbdotnet under Mac OS X and the console based example program (benchmarkcon works fine). The GUI based examples were more meant for Windows.

mcuee, 2017-01-15:
For simple C# wrappers, you may be able to refer to libusbK's example.
  https://github.com/mcuee/usb-travis/blob/master/libusbK/bindings/libusbK.cs

As for .NET Core, I think it is a subset of .Net, so you need to be care of using the features not supported by .Net Core.

mcuee, 2017-01-15:
`MyMacMini:LibUsbDotNet_Bin xiaofanc$ mono BenchmarkCon.exe vid=0x04d8 pid=0xfa2e
WARNING: The runtime version supported by this application is unavailable.
Using default runtime: v4.0.30319
LibUsbDotNet USB Benchmark v2.2.8.104
Copyright (c) 2010 Travis Robinson. <libusbdotnet@gmail.com>
website: http://sourceforge.net/projects/libusbdotnet
Benchmark device 04D8:FA2E opened..
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
Bulk Write (Ep01h) max packet size: 32

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..

Avg. Bytes/s: 294315.98 Transfers: 66 Bytes/s: 294315.98
Avg. Bytes/s: 294875.18 Transfers: 138 Bytes/s: 295389.65
Avg. Bytes/s: 292416.83 Transfers: 209 Bytes/s: 287754.01
Avg. Bytes/s: 291199.67 Transfers: 279 Bytes/s: 287625.16
Avg. Bytes/s: 290577.47 Transfers: 350 Bytes/s: 288158.03
Avg. Bytes/s: 290374.11 Transfers: 421 Bytes/s: 289375.78
Avg. Bytes/s: 291064.03 Transfers: 493 Bytes/s: 295164.66
Avg. Bytes/s: 291642.45 Transfers: 565 Bytes/s: 295665.65
Avg. Bytes/s: 291337.40 Transfers: 636 Bytes/s: 288932.45
Avg. Bytes/s: 291418.75 Transfers: 708 Bytes/s: 292139.31
qAvg. Bytes/s: 290857.79 Transfers: 778 Bytes/s: 285303.18
stopped Ep81h thread.
stopped Ep01h thread.
Loop Test Information
	Vid / Pid       : 04D8h / FA2Eh
	Interface #     : 00h
	Priority        : Normal
	Buffer Size     : 4096
	Buffer Count    : 1
	Display Refresh : 1000 (ms)
	Transfer Timeout: 5000 (ms)
	Retry Count     : 0
	Verify Data     : Off

Bulk Read (Ep81h) max packet size: 32
	Total Bytes     : 3190784
	Total Transfers : 779
	Avg. Bytes/sec  : 290872.20
	Elapsed Time    : 10.97 seconds

Bulk Write (Ep01h) max packet size: 32
	Total Bytes     : 3190784
	Total Transfers : 779
	Avg. Bytes/sec  : 290874.83
	Elapsed Time    : 10.97 seconds

Press any key to exit..

TAGC, 2017-01-15:
My experience:

 - `libusbdotnet` (.NET Framework-based) works for Windows
 - `corecompat.libusbdotnet` (.NET Core-based) works for Windows
 - `corecompat.libusbdotnet` (.NET Core-based) does not work for Linux

mcuee, 2017-01-15:
Actually I am not so sure where the CoreCompat version comes from. Probably it is just a quick hack (based on the name and version number) of the existing libusbdotnet code and try to get it work under Linux or Mac OS X. It probably needs quite some work. I am not sure how much efforts it will need.

On the other hand, libusbdotnet is probably quite old and it is probably a good idea to create something new.

andrasfuchs, 2017-01-15:
Hm... interesting, thanks for the info!

I've tried LibUsbDotNet on Windows, and I got NullReferenceException every time I tried to open a device.
I'll double check it with the provided examples next week and I will be back to you with the results.

On the other hand LibUsbDotNet's let us access only 54 public API methods instead of the 89 available in the library, so it would still need an update I think. [Here is a comparison](https://docs.google.com/spreadsheets/d/1sNY7vqtrPs3n6b5w8joHh-KPzfnkQRmX5EacvzP9mic/edit?usp=sharing).
I'm still not familiar with the inner workings of LibUsbDotNet, but it looks like it might switch between WinUSB and libusb depending on the current platform.

mcuee, 2017-01-15:
Yes libusbdotnet uses libusb-1.0 under Mac OS X and Linux using Mono. And under Windows it uses either libusb-1.0 or WinUSB. And it actually support hotplug under Windows even though libusb-1.0 does not support hotplug. It also supports its own simple implementation of hotplug under Linux and Mac OS X before libusb-1.0 supports hotplug under Linux and Mac OS X.

In any case, libusbdotnet is old, that is for sure. And I checked with Travis before, he also thinks that the codes are not the best.

mcuee, 2017-01-15:
@TAGC Hi David -- does Docker support libusb? I somehow remember it does not work with USB access.

TAGC, 2017-01-15:
@mcuee I'm not sure, but I can definitely install libusb on a Docker container and get it to see my USB device via `lsusb`.

andrasfuchs, 2017-01-16:
I looked into the source codes of libusb and LibUsbDotNet, so I put together a project plan for the new wrapper library. Let's call it simply LibUSBNet.

- [ ] Stage 1 - basic API access
    - [ ] Cover all 89 API calls in the LibUSB.API namespace. All method names should reflect the original names in libusb-1.0.dll (in the same order, for easier future comparisons). Enums and Structs should follow the normal .NET convention since they will be used outside this namespace.
    - [ ] Translate the tests from libusb from C to C#.
    - [ ] The library's version should reflect libusb's version number + it's own build number, like 1.0.13.42.
- [ ] Stage 2 - quality control
    - [ ] Replicate at least some of the test from libusb using C#
    - [ ] Generate documentation from smart comments (of methods, enums, structs)
- [ ] Stage 3 - public release
    - [ ] Set up AppVeyor to make automatic builds
    - [ ] Create a NuGet package (I'm not sure if we would need two separate packages to target the .NET Framework and .NET Core, I have to check that.) The NuGet package should include the most recent compiled libusb-1.0.dll, add the references to the project, etc.
    - [ ] Set up automatic testing with Travis CI or Jetkins CI
    - [ ] Link to our project from libusb readme/wiki
- [ ] Stage 4 - OO classes
    - [ ] Build the more C#-friendly OO USB classes. They call the API methods without using anything fancy (meaning no .NET Framework-specific library or interop calls other then libusb-1.0.dll). At this point Travis Robinson's in-depth knowledge would be gold, since he understands how would this be the most logical.
    - [ ] All the classes would be in the LibUSB namespace, only with one static root class and instance classes for devices, interfaces, etc. All the objects here should strictly follow the .NET naming conventions and documentation principals. 
    - [ ] New example projects to demonstrate the managed-only OOP approach.
    - [ ] Cover all API functions, so theoretically the LibUSB.API namespace could be set internal, without losing any functionality for the library user who can access only the public OO classes.
- [ ] Stage 5 - maintenance
    - [ ] Reply to and fix issues
    - [ ] Implement feature requests
    - [ ] Update the package if new version of libusb comes out

Please let me know if I missed something.

To be honest, I must admit that this is a much bigger project this way then I expected at the beginning. I think the .NET community would much needed a recently updated, high quality library to access libusb. My estimation is that it would take at least 250-300 developer hours to make this work, so I would need all the help I can get, if we decide to give a try.

Let me know if you can also invest some time into it, or know someone who would. According to my schedule I think I can do 10-15 hours per week for the next few months.

If you think it's not worth it, then I will start to use WinUSBNet, that looks pretty solid on Windows.

mcuee, 2017-01-21:
Looks like you have a great plan. I think Stage 1 to 3 will already be a good contribution to the community. 

It may be a good idea to post to the libusb mailing list to see if there are more people interested, especially for Stage 4, I am not so sure if Travis has time for this but there may be some experts who can contribute the idea.

I myself do not know anything about dotnet, but I can certainly help on testing. I am not really a developer in the libusb project and I do not know much details of the internals of libusb, rather I help on the technical support and testing side.

TAGC, 2017-01-21:
@andrasfuchs @mcuee You guys might like to know that some guy figured out a workaround ([link](http://stackoverflow.com/a/41765048/1636276)). Tested communication between my ASP.NET Core web application and a microcontroller over USB and it works fine both when the app is running natively under Windows, and within a docker container using Docker Toolbox. :)

mcuee, 2017-01-21:
As for the name, LibUSBNet may not be a good name, there is already a project (probably dead) called libusbnet: https://github.com/vavrusa/libusbnet , probably "libusb.NET" or even "libusbdotnet NG" or things like that.

mcuee, 2017-01-21:
@TAGC Glad to know that. So can we say "CoreCompat.LibUsbDotNet" already works under Windows and Linux with the workaround?  I assume your Docker container using Docker Toolbox is using Linux, right?

mcuee, 2017-01-21:
@TAGC BTW, now I see you are using Docker Toolbox for the USB access. There is a FAQ that latest Docker container can not pass through USB unless using the older Docker Toolbox.

https://docs.docker.com/docker-for-mac/faqs/#/can-i-pass-through-a-usb-device-to-a-container

mcuee, 2017-01-21:
@andrasfuchs On the other hand, even if "CoreCompat.LibUsbDotNet" works, I think what you proposed is still a good idea, take note libusbdotnet development has stopped long ago.

TAGC, 2017-01-21:
@mcuee Yep. CoreCombat.LibUsbDotNet + a docker file like this:

```
FROM microsoft/aspnetcore:1.1.0
ENTRYPOINT ["dotnet", "bin/Debug/netcoreapp1.1/publish/<<Application>>.dll"]
...
RUN apt-get update
RUN apt-get install -y libusb-1.0-0-dev usbutils
RUN ln -s /lib/x86_64-linux-gnu/libusb-1.0.so.0 /usr/share/dotnet/shared/Microsoft.NETCore.App/1.1.0/libusb-‌​1.0.dll
```

Docker Toolbox for Windows essentially runs a Linux VM using Virtualbox, and docker containers are launched within that VM.

mcuee, 2017-01-21:
@andrasfuchs 

BTW, if you are looking at libusbK, indeed it was also created by Travis. Travis is also the current maintainer of libusb-win32. For both libusbK and libusb-win32, I am also the project admin mainly for technical support and testing. 

libusbK (the library) supports device using WinUSB, libusbk.sys (libusbK kernel driver) and libusb0.sys (libusb-win32 kernel driver). It has a simple C# wrapper as well.

In fact, libusb Windows backend uses libusbK.dll when it is presents to access device using WinUSB, libusbK.sys (working well) and libusb0.sys (not working well). If libusbK.dll is not present, then libusb Windows backend will only supports WinUSB but not libusbk.sys and libusb0.sys.

mcuee, 2017-01-21:
@TAGC BTW, I do not want to spoil the fun with "CoreCompat.LibUsbDotNet" but take note I do not see the source. If it is based on libusbdotnet (which I strongly believe), then libusbdotnet's LGPL v3 license needs to be observed, which means the modifies source need to be made available.

TAGC, 2017-01-21:
@mcuee This doesn't modify the source code, though. It's literally just creating a symbolic link on the filesystem so it finds `libusb`.

mcuee, 2017-01-22:
@TAGC This has nothing to do with libusb or the workaround -- rather it is where you can get the source code of "CoreCompat.LibUsbDotNet". I assume it is a modification of Travis' libusbdotnet which is licensed under LGPL V3. The owner in Nuget site is "quamotion" and it seems the owner does not provide any license or source code along with the package, that is violating LGPL V3 license.

TAGC, 2017-01-22:
@mcuee Ah, I see what you mean. For what it's worth, I'm pretty sure I tried contacting the owners of that package months ago for help with this issue and got no response.

mcuee, 2017-01-22:
I see. Probably you want to try again. From what I see, his other packages does provide license info and authors info, including libusb-win32 package.

Ref: 
https://www.nuget.org/packages/libusb-win32-driver/
https://www.nuget.org/packages/libusbmuxd/

I'll try to contact the company behind the packages directly using the info here.
   http://quamotion.mobi/Home/About

qmfrederik, 2017-01-22:
I'm the maintainer of the CoreCompat.LibUsbDotNet package. The source code for this package is available on GitHub: https://github.com/corecompat/libusbdotnet. 

Like most packages in the CoreCompat repository, it's a fork which adds support for .NET Core. The idea is to merge the changes back upstream.

Thanks for pointing out we didn't provide license details and a link to the GitHub site on NuGet! I've updated the package to include that information.

Regarding LibUsbDotNet: it works on Linux as well (as pointed out above).
For it to work, LibUsbDotNet needs to find the native (`libusb`) library. 
Symlinking `libusb-1.0.so.0` to `libusb-1.0.dll` is one way to do it. What .NET Core expects is that you `[DllImport("usb-1.0")]` and that there would be a `usb-1.0.dll` on Windows (where `LoadLibrary` can find it); a `libusb-1.0.so` (with version information in suffix if required) on Linux (in a location where `dlopen` can find it) and a `libusb-1.0.dylib` on macOS.

.NET Core (well, NuGet) also has a deployment mechanism which allows you to ship the native library (.so/.dll) file with your NuGet package; that makes life even easier.
We do the same thing for our C# wrapper for libimobiledevice; here is some example code:
- P/Invoke declarations: [UsbmuxdNativeMethods.cs](https://github.com/libimobiledevice-win32/imobiledevice-net/blob/master/iMobileDevice-net/Usbmuxd/UsbmuxdNativeMethods.cs)
- Package which contains the Linux native libraries: [project.json](https://github.com/libimobiledevice-win32/imobiledevice-net/blob/master/native/runtime.linux-x64.iMobileDevice-net/project.json)
- Some additional code which attemps to make sure the libraries are loaded correctly: [NativeLibraries.cs](https://github.com/libimobiledevice-win32/imobiledevice-net/blob/master/iMobileDevice-net/NativeLibraries.cs)

I don't currently plan to do much work on the CoreCompat.LibUsbDotNet package but I don't see why it wouldn't be a good starting point.

If you're planning to continue to work on LibUsbDotNet, feel free to fork our repo or send us pull requests. 
I think the main part which is missing is to clean up the `DllImport` statements and provide packages for the native libraries on all platforms you want to support.
You should also be able to remove all Mono-specific code, as Mono now also is capable of running libraries compiled for .NET Core.

Let me know if you have questions.

mcuee, 2017-01-22:
@qmfrederik Thanks a lot for the update!

andrasfuchs, 2017-01-23:
@mcuee Thanks for the info, and the work you're doing on these projects!  

> And it actually support hotplug under Windows even though libusb-1.0 does not support hotplug under Windows.  

Is it still the case with the newer versions? In the libusb-1.0.dll there 3 public methods to support hotplug, They are just not in the libusbdotnet wrapper class (yet). Making them available would solve this issue, or libusb needs some changes too?  
Also, according to [this document](https://msdn.microsoft.com/en-us/library/windows/hardware/ff540196(v=vs.85).aspx), the newer WinUSB supports isochronous transfers. Do you know if libusb supports this with WinUSB?

@qmfrederik Thanks, it sounds good. I'll take a look at the repo you mentioned this week, and see if I can contribute there, instead of starting a new project.

mcuee, 2017-01-24:
@andrasfuchs I do not know much about the internal working mechanism of libusbdotnet. However I know libusbdotnet supports hotplug under Windows. It also tries to support hotplug under Mac OS X and Linux but I just tested under Mac OS X and the support does not seem to work under Mac OS X.

As for libusb, it does officially support hotplug under Linux and Mac OS X. But there is no official support for hotplug under Windows as of now.

libusb Windows does not support isochronous transfer with the Windows WinUSB backend now. However, the Windows usbdk backend supports isochronous transfer.

All in all, you can say that libusb Windows backend is lacking behind Mac OS X and Linux now.

mcuee, 2017-01-24:
@qmfrederik BTW, I checked with Travis, he is telling me for libusbdotnet, you can choose either LGPL v3 or GPL v3. I tend to think LGPL v3 will be easier to use for a library. So maybe you can change the license in Nuget site to LGPL v3.

andrasfuchs, 2017-01-30:
@qmfrederik Thank you for merging my pull request to your repo. It is just a small update to libusbdotnet, but I will continue to work on it, if I can see that it reaches many people. 
Do you know if/when will @Stevie-O merge your pull requests to his repo? Are you in contact with him?

@mcuee Thanks for the info, that's good to know.

mcuee, 2017-01-31:
@andrasfuchs FYI, Travis has made @qmfrederik the admin of libusbdotnet project.

andrasfuchs, 2017-01-31:
Great stuff, I'm glad! That opens up some opportunities, I'll discuss the details with @qmfrederik.

qmfrederik, 2017-01-31:
@andrasfuchs @mcuee Yes, the upstream LibUsbDotNet project has been updated to clarify it's on LGPL (+ GPL) instead of GPL only. I've created an issue in the CoreCompat repo to update the license information there as well.

I don't think @Stevie-O 's LibUsbDotNet fork is currently maintained (let me know if otherwise), so I kept CoreCompat for now.

I'm also open to creating a dedicated LibUsbDotNet org on GitHub if that makes sense.

mcuee, 2017-02-01:
@qmfrederik Glad to see that libusbdotnet gets new life. As for a separate repo, I think now you are in charge so you decide. :-)

mcuee, 2017-02-01:
Label this as "wontfix" since it is out of the scope for libusb project. On the other hand, I will keep this ticket open for now since it creates good positive results.

mcuee, 2017-02-07:
Follow-up discussion here:
https://github.com/LibUsbDotNet/LibUsbDotNet/issues/5

mcuee, 2017-03-01:
Close this now since the new LibUsbDotNet github site is up.

closed 2017-03-01 by mcuee