libusb/libusb #254: LibUSB MacOS libusb_bulk_transfer fails with timeout error code

DevAljosa, 2017-01-10:
I am working on a MacOS project where I need to communicate with device over USB. For windows I use libusb-win32 or WinUSB driver and both work great. I have MacOS El Capitan (10.11) installed in virtual machine (Vmware). I succesfully installed libusb drivers (1.0.21) and built libusb-1.0.21.dylib with xcode and I can load it in my project. When I run my project from Delphi and debug it in Mac I can connect to device, sending data to device seems to work but reading data from device always fails. I am using synchronous call libusb_bulk_transfer. Everytime I try to read I get timout error (-7). I installed Device monitor studio to sniff USB traffic and I can see that device responds with some data (after timeount) and ABORT PIPE URB. But function call returns error and buffer is empty. As you can see in attached image there is 2sec of timeout I set in a call to prevent hang up. I tried almost everything without success. Same device with WinUSB or LibUSB-win32 driver installed in windows works fine. Any idea what could be wrong?

[image]

[image]

mcuee, 2017-01-10:
Get an Apple official support MAC OS X environment and see if you are still
facing such an issue. It could be a VMware issue or your unofficial MAC OS
X installation issue.

DevAljosa, 2017-01-10:
I had thoughts there could be problem with non-official Mac OS in Vmware but I tried it anyways. Thanks for your suggestion I will get a "real" mac machine to try it again.

crazenator, 2017-04-25:
Hello fellows,

I've similar issue but on beaglebone black, running jessie-linux; I'm using asynchronous API. I can connect mobile device, switch it to Google accessory and start communicating; somewhere in the middle, a reception would time-out and no retries would help unless I close libusb device handle and reopen it. If I see the packet capture logs, I can see IN-NAK for the message. Can anyone help in fixing it

PratikGajera, 2017-09-21:
Hi,
I have same problem in beaglebone black and installed ubuntu OS in it. When I read bulk data from bulk-in endpoint using synchronous libusb bulk API then only read 64 bytes data from it and using asynchronous libusb bulk API then only read 1536 bytes data from it with libusb handle events control API. And without use libusb handle events control API then successfully read expected data(200000 bytes) from it, but getting late call back response due to I coundn't send another command to device. And look at my small demo code.

i) Asynchronous

int do_exit = 0;

void my_callback(struct libusb_transfer *transfer)
{

    switch(transfer->status)
    {
        case LIBUSB_TRANSFER_ERROR: break;
        case LIBUSB_TRANSFER_NO_DEVICE: break;
        case LIBUSB_TRANSFER_CANCELLED:  break;
        case LIBUSB_TRANSFER_COMPLETED:
								save_file(transfer->buffer,transfer->actual_length);
                                 break;
        case LIBUSB_TRANSFER_TIMED_OUT:
									
									break;
        case LIBUSB_TRANSFER_STALL: 
									break;
        case LIBUSB_TRANSFER_OVERFLOW:
									break;
        default: 
    }

        do_exit = 1;
}

main()
{
 // libusb initialization done

 transfer_in  = libusb_alloc_transfer(0);
                              
 libusb_fill_bulk_transfer( transfer_in, dh, 0x82,rcvBuff, 200000,my_callback, NULL, 200);
 
 ret = libusb_submit_transfer(transfer_in);
 
 while(!do_exit)
 {
    libusb_handle_events(NULL);
 }

}

ii) Synchronous

main()
{

 ret = libusb_bulk_transfer(dh, 0x82,rcvBuff,200000,&transferred, 0);
 
}

PratikGajera, 2017-09-21:
I have same problem in beaglebone black and installed ubuntu OS in it. When I read bulk data from bulk-in endpoint using synchronous libusb bulk API then only read 64 bytes data from it and using asynchronous libusb bulk API then only read 1536 bytes data from it with libusb handle events control API. And without use libusb handle events control API then successfully read expected data(200000 bytes) from it, but getting late call back response due to I coundn't send another command to device. And look at my small demo code.

i) Asynchronous

int do_exit = 0;

void my_callback(struct libusb_transfer *transfer)
{

    switch(transfer->status)
    {
        case LIBUSB_TRANSFER_ERROR: break;
        case LIBUSB_TRANSFER_NO_DEVICE: break;
        case LIBUSB_TRANSFER_CANCELLED:  break;
        case LIBUSB_TRANSFER_COMPLETED:
								save_file(transfer->buffer,transfer->actual_length);
                                 break;
        case LIBUSB_TRANSFER_TIMED_OUT:
									
									break;
        case LIBUSB_TRANSFER_STALL: 
									break;
        case LIBUSB_TRANSFER_OVERFLOW:
									break;
        default: 
    }

        do_exit = 1;
}

main()
{
 // libusb initialization done

 transfer_in  = libusb_alloc_transfer(0);
                              
 libusb_fill_bulk_transfer( transfer_in, dh, 0x82,rcvBuff, 200000,my_callback, NULL, 200);
 
 ret = libusb_submit_transfer(transfer_in);
 
 while(!do_exit)
 {
    libusb_handle_events(NULL);
 }

}

ii) Synchronous

main()
{

 ret = libusb_bulk_transfer(dh, 0x82,rcvBuff,200000,&transferred, 0);
 
}

arrosev, 2019-02-22:
I have the same problem, I use submit_bulk_transfer to read video stream from a uvc camera, but the ReadPipeAsyncTO always tell me time out

fcapozzi, 2019-03-16:
iMac 10.14.3 mojave.  Using LimeSDRmini in SDRAngel the SoapySDR driver usr libusb version 1.0.22 
I can start and read device but if i stop reception  it is not possible to rearm it.  When stopping many  darwin_abort_transfer warning pops up. If i try to reconnect device the readStream() method throws a TIMEOUT error. 
Are some libusb fix available for osx environment ?
Tnx, Fabio.

mcuee, 2020-07-29:
Please provide debug logs which may help to sort out this issue.

mcuee, 2020-12-07:
Close this for now as there are no updates for a long time. Please reopen if new info comes in.

closed 2020-12-07 by mcuee