libusb/libusb #255: Windows: libusb_claim_interface() return LIBUSB_ERROR_NOT_SUPPORTED

YanjinTian, 2017-01-11:
Device ：USB 2.0 Camera
This function returns LIBUSB_ERROR_NOT_SUPPORTED When i use.
``` libusb: debug [unsupported_claim_interface] unsupported API call for 'claim_interface' (unrecognized device driver) ```

mcuee, 2017-01-12:
You will have to install a supported driver like WinUSB or usbdk.

YanjinTian, 2017-01-12:
I used the most recent version of Zadig to install the supported driver(WinUSB (v6.1.7600.16385) / libusbK (v3.0.7.0)),but it is still returning LIBUSB_ERROR_NOT_SUPPORTED

mcuee, 2017-01-12:
This is quite unlikely. Please post the code snippets to demonstrate the problem and post the full debug log as well.

mcuee, 2017-01-12:
BTW, why do you want to use libusb with a Camera? In general, libusb is not the right tools to use with a camera.

mcuee, 2017-01-12:
You may want to read this guide as well in order to provide more relevant info.
https://github.com/libusb/libusb/wiki/Troubleshooting

mcuee, 2017-02-01:
Will close this in another week if no more info are provided.

dickens, 2017-03-01:
Closing due to lack of response.

closed 2017-03-01 by dickens

xiejiezhang, 2020-08-05:
Is this issuse solved? @mcuee

mcuee, 2020-08-05:
Do not reply to old closed issue like this one. Open a new issue if you have any.

mcuee, 2020-08-05:
Read the WiKi, FAQ, etc, before asking questions. Do provide details of device, the OS, the code, the debug log, etc, when you raise an issue.

ravipratap8, 2021-06-21:
DevTools listening on ws://127.0.0.1:53985/devtools/browser/f085b501-5c4e-43b7-9717-11dc0d9267ac
[17904:15248:0621/082815.349:ERROR:device_event_log_impl.cc(214)] [08:28:15.349] USB: usb_service_win.cc:389 Could not read device interface GUIDs: The system cannot find the file specified. (0x2)
[17904:15248:0621/082815.375:ERROR:device_event_log_impl.cc(214)] [08:28:15.375] USB: usb_device_handle_win.cc:1058 Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)
[17904:15248:0621/082815.377:ERROR:device_event_log_impl.cc(214)] [08:28:15.377] USB: usb_device_handle_win.cc:1058 Failed to read descriptor from node connection: A device attached to the system is not functioning. (0x1F)

mcuee, 2021-06-21:
Please refer to new ticket. Do not comment further here.
https://github.com/libusb/libusb/issues/938