libusb/libusb #259: WebUSB support on Windows requires custom driver

larsgk, 2017-01-16:
It seems strange that a custom WinUSB driver is needed for libusb in order to work.  Shouldn't it be possible 'just' to work on directly supporting e.g. the latest WinUsb driver shipping with windows (that loads automatically with the right MS OS 1.0 descriptors in place):

https://bugs.chromium.org/p/chromium/issues/detail?id=637404

mcuee, 2017-01-17:
I do not understand you, libusb does not need a custom WinUSB driver. It supports those WCID USB device with the proper MS OS descriptors in place. Please post more information.

mcuee, 2017-01-17:
Since you are using a composite device with CDC + custom, I think you are using IAD. I am not so sure if you hit the current libusb Windows backend's limitation or not.
https://github.com/libusb/libusb/issues/85

mcuee, 2017-01-17:
https://github.com/WICG/webusb/issues/75
Pete is helping there.

mcuee, 2017-01-24:
From what I read, this issue does not seem to relate to libusb Windows, rather the FW is not working very well under Windows.

mcuee, 2017-02-01:
I am closing this as the conclusion is that libusb is not the issue (other than the IAD issue covered in ticket #85.

closed 2017-02-01 by mcuee