libusb/libusb #260: Android hotplug permission issue

jiangrui02, 2017-01-17:
it seems that libusb can not monitor hotplug event.
The code:
      linux_netlink_socket = socket(PF_NETLINK, socktype, NETLINK_KOBJECT_UEVENT);
create socket failed with errno 13, it's a permission issue.

Is  there any suggestion or solution? thanks.

mcuee, 2017-01-17:
I think if you are using netlink, then it means udev is not available or not usable. In that case, hotplug is not supported.

mcuee, 2017-01-17:
In any case, there are multiple issues with Android as of now. I am not so sure which libusb version and whether it is the stock libusb you are using. Please refer to issue #188 and #242 as well.

mcuee, 2017-02-01:
Closing this one. Please reopen if you think #188 and #242 do not cover your issue.

closed 2017-02-01 by mcuee