libusb/libusb #262: usbdk (or libusb?): Large isochronous transfers do not work

leoetlino, 2017-01-27:
It looks like large isochronous transfers are broken on Windows with usbdk (while smaller ones do work fine). 

```
00172692	100.01865387	[11780] [290.514810] [00002de0] libusb: debug [usbi_handle_transfer_completion] transfer 000000002DD9CAD8 has callback 0000000000C56530	
00172693	100.01867676	[11780] [290.514810] [00002de0] libusb: debug [sync_transfer_cb] actual_length=37	
00172694	100.01868439	[11780] [290.514810] [00002de0] libusb: debug [windows_handle_events] checking fd 3 with revents = 0001	
00172695	100.01869965	[11780] [290.514810] [00002de0] libusb: debug [usbi_remove_pollfd] remove fd 3	
00172696	100.01871490	[11780] [290.514810] [00002de0] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 234, size 0	
00172697	100.01875305	[11780] [290.514810] [00002de0] libusb: error [windows_transfer_callback] detected I/O error 234: [234] More data is available.	
00172698	100.01877594	[11780] [290.514810] [00002de0] libusb: debug [usbi_handle_transfer_completion] transfer 000000002CF336E8 has callback 0000000000A90680	
00172699	100.01879883	[11780] 29:25:154 IOS\USB\LibusbDevice.cpp:350 E[IOS_USB]: [057e:030a 1] Isochronous transfer (endpoint 0x81) failed: LIBUSB_TRANSFER_ERROR	
```

As shown in the log, the transfer seems to complete fine, but something is not handling ERROR_MORE_DATA properly, which causes libusb to consider the whole transfer as failed.

A usbdk dev clarified that this could happen when using ReadFile, but I'm not familiar at all with Windows APIs (or libusb internals)…

FYI, the game submits 64 isochronous packets in one transfer, an individual packet taking up to 3060 bytes (IIRC), and with a total size of 64768 bytes. Could there be a packet size limit on Windows which causes this to fail, or require proper handling of ERROR_MORE_DATA?

diabolo38, 2017-01-27:
Various window versoin  do have diffeent size limit see https://msdn.microsoft.com/en-us/library/windows/hardware/ff538112(v=vs.85).aspx it could be 64K but  lesss or  more it depend on the max packet size

leoetlino, 2017-01-28:
I see, I'll try messing with the total transfer size and see if that helps.

mcuee, 2018-04-20:
You can try also the latest 1.0.22 version which supports isoc transfer with libusbk driver.

mcuee, 2018-04-20:
Close this for now. Please reopen when you try 1.0.22 version and provide the info.

closed 2018-04-20 by mcuee