libusb/libusb #263: Unable to build on windows with mingw fails on make

rcmpayne, 2017-01-27:
Trying to build on windows with mingw/msys and make is failng with the below.

```
$ make
make  all-recursive
make[1]: Entering directory `/c/github/libusb'
Making all in libusb
make[2]: Entering directory `/c/github/libusb/libusb'
  CC     libusb_1_0_la-core.lo
In file included from libusbi.h:248:0,
                 from core.c:44:
os/threads_windows.h:40:8: error: redefinition of 'struct timespec'
 struct timespec {
        ^
In file included from c:\mingw\include\sys\time.h:3:0,
                 from core.c:34:
c:\mingw\include\time.h:102:8: note: originally defined here
 struct timespec
        ^
make[2]: *** [libusb_1_0_la-core.lo] Error 1
make[2]: Leaving directory `/c/github/libusb/libusb'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/c/github/libusb'
make: *** [all] Error 2
```

mcuee, 2017-01-27:
This seems to be a MinGW.org problem. I have asked in the MInGW mailing
list but nobody replied. It happened before as well.
-- 
Xiaofan

diabolo38, 2017-01-28:
same kind of issue seam to appear on window build depending sdk installed , I suggest you use mingw64 build instead of .org, you'll likley  face issue later on link with .org missing some win api  (it was the case on 1.5 year back). I had no issue building 1.0.21 with mingw64  (32 & 64 bits version)  couple week ago.

rcmpayne, 2017-01-30:
what version and build do you recommend for windows?

https://mingw-w64.org/doku.php/download

diabolo38, 2017-01-30:
I download  files from sourceforge  mingw build that elaev more optin on what you et and how/where  to install it
for exemple  win32 bit  i use https://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/4.9.1/threads-posix/sjlj/ 
I tend to use quite old version 4.9.1 or 4.8.5  right now 
posix thread, seh exception model for 64 bit and sjlj for 32 bit

mcuee, 2017-01-31:
@rcmpayne You can also try MSYS2 at http://msys2.github.io/

rcmpayne, 2017-01-31:
Switching to msys2 fixed all my errors except ideviceinstaller which i will post a bug in that section.

closed 2017-01-31 by rcmpayne