libusb/libusb #266: libusb_exit reference counting

cfneese, 2017-02-02:
In core.c, shouldn't line 2188 use a postfix decrement instead of a prefix decrement?

In my tests on Windows, default_context_refcnt is 1 after two calls to `libusb_init(NULL)`, and on the first call to `libusb_exit(NULL)` the default context is destroyed with line 2188 as coded.

LudovicRousseau, 2017-02-02:
I think you should add some libusb logs.

``default_context_refcnt`` value should be 2 after 2 calls to ``libusb_init(NULL)`` and you should have 2 logs lines:
``"created default context"`` from https://github.com/libusb/libusb/blob/master/libusb/core.c#L2103
and
``"reusing default context"`` from https://github.com/libusb/libusb/blob/master/libusb/core.c#L2077

dickens, 2017-02-02:
There is no issue with the reference counting that I can identify. I suspect if there was it would have been caught long ago. Closing unless it can be proven otherwise (please provide logs if so).

closed 2017-02-02 by dickens