libusb/libusb #268: hotplug events stop working in parent after fork(), when child calls libusb_exit()

attie, 2017-02-10:
Hi All,

I've come across a problem, and was wondering if anyone here could help.

Fundamentally, the issue I've come up against, is that by calling libusb_exit() in the child, the parent no longer sees any hotplug events.

I've described it in more detail, with an example application, in this SO question:
http://stackoverflow.com/questions/42145020/libusb-1-0-hotplug-events-stop-working-in-parent-after-fork-when-child-calls

Attie

dickens, 2017-02-10:
We will have to update the documentation to indicate that this isn't supported. The problem is that when you fork, you have two copies on the same context referring to the same file descriptors. When you call libusb_exit() in the child, it will write message(s) to the control pipe in order to deregister all the hotplug callbacks. This pipe is of course shared with the parent, so assuming that nothing is calling libusb_handle_events() in the child at the time that libusb_exit() is called, the parent will be the one to see the message and will end up deregistering the hotplug callback there.

On a related note, I've wanted to add the CLOEXEC flag to libusb file descriptors for some time. I assume this would completely solve what you're trying to achieve?

hjelmn, 2017-02-10:
@dickens I don't know how we missed adding this to the documentation. It came up way back in 2002 with libusb 0.1. Threads are recommended over fork and they get what they get when they use fork.

I am not opposed to adding CLOEXEC but that only helps us on Linux. We still have undefined behavior on OS X.

attie, 2017-02-10:
Thanks for the discussion guys, I suspected that something like that was happening. but couldn't prove it through.

I've been thinking it over, and I'll probably `fork()` early (before calling `libusb_init()`) to avoid the problem... I'll have to use a pipe between the two processes to provide the communication (what application to start for what USB device), but that's not really a problem.

`O_CLOEXEC` _might_ help, depending on how much of the shutdown is negotiated with the kernel... if the call to `close()` is going to kill off the other process' functionality too, then it's possibly a worse place to be in?

dickens, 2017-03-01:
Closed via aa73b2ec

closed 2017-03-01 by dickens