libusb/libusb #272: problem including libusb in Qt

javadhabibi, 2017-02-21:
hi,i want to use libusb in Qt 5.7 on win7. but i am confused what should be add to Qt? there is many files in libusb-1.0.21 folder: [here](https://freefr.dl.sourceforge.net/project/libusb/libusb-1.0/libusb-1.0.21/libusb-1.0.21.tar.bz2)
which of them should be used?
thanks.

diabolo38, 2017-02-21:
the package on SF only contain source theirs no prebuilt binary for mingw or msvc as the wiki say here https://github.com/libusb/libusb/wiki/Windows#binary-snapshots :( 
package with binary must exist somewhere i have use one long time ago but  can' find where anymore (but old libusb-win32). 

mingw/gcc  built from source is not so easy but opening and built of msvc project shall produce a usable dll.

mcuee, 2017-02-22:
Hmm, the wiki site updated. libusb binary is released in Sourceforge and now also on github along with the source code release.

MinGW/GCC build is now easy with infrastructure like MSYS2.

mcuee, 2017-02-22:
Will close this issue. If you have more problem, ask in QT side.

closed 2017-02-22 by mcuee

diabolo38, 2017-02-22:
@mcuee all sourcefoge files  package are source only they don not include any binary ! so the wiki is sttill worn on this point.
but it do i confirm the package on github via https://github.com/libusb/libusb/releases contain binary

mcuee, 2017-02-22:
No Sourceforge file release folder has both the source code archive and
Windows binary release. Please check. As for Wiki, you may have to refresh
your page. I just updated it.
-- 
Xiaofan

mcuee, 2017-02-22:
For example, libusb 1.0.21 Windows binary us here with file name
libusb-1.0.21.7z.

https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.21/
-- 
Xiaofan

diabolo38, 2017-02-22:
rigth the 7z do have binary but it is the only file  all others , zip .gz do not 
that  is misleading has their is no infromation on SF about what package contain what :-( 
Now that pretty much everything is on github maybe release on source-force  could be droped ?

javadhabibi, 2017-02-22:
thanks,problem solved by compiling the libusb(dll) solution in visual studio.
you are right.there is no binary file.