libusb/libusb #273: libusb_open() returns -12

javadhabibi, 2017-02-22:
hi, why libusb_open() returns -12. i first get the all devices vid/pid and then open(MY FLASH VID PID) . but it returns -12.
thanks

diabolo38, 2017-02-23:
Did you install any back end  driver ? This is mandatory unless your dev is hid or contain specifiq desc that assoiciate to winusb. Read FAQ use of zadig make install fast and easy .

javadhabibi, 2017-02-24:
that's Right. it is solved with using zadig . but another problem occur that i have described in: [](https://github.com/pbatard/libwdi/issues/78)

closed 2017-03-01 by mcuee

lwei2, 2020-09-04:
My friend, I also met the problem of libusb_open-12. How did you solve it?

mcuee, 2020-09-07:
@lwei2 Please use libusb mailing list for support. Do not use the github issue tracker.