libusb/libusb #289: Unexpected error from libusb_submit_transfer.

romeojulietthotel, 2017-04-03:
This is where I get an error:

https://github.com/myriadrf/LimeSuite/blob/master/src/ConnectionSTREAM/ConnectionSTREAM.cpp#L570

This problem was encountered in 1.0.21 but not seen in 1.0.20.

Also see:
https://github.com/myriadrf/LimeSuite/issues/114

dickens, 2017-05-12:
The only reason libusb_submit_transfer() would return LIBUSB_ERROR_BUSY is if the transfer being submitted has already been submitted and had not yet completed. I notice that the code which is submitting the transfer does not check the 'done' field before submitting, so perhaps this is the place to start looking.

mcuee, 2017-10-07:
Close this issue for now. Will reopen if got more info.

closed 2017-10-07 by mcuee