libusb/libusb #291: embeded system usage

ahmadeqbal, 2017-04-19:
hi every one,
i'm using libusb in intel edison board 
lsusb command shows the usb device but when i try to run the code I get this error:
```
root@admin:/root# ./out
Error finding USB device
```
here is lsusb command output:
```
root@admin:/root# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0483:3748 STMicroelectronics ST-LINK/V2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
i'm using [this example](https://github.com/libusb/libusb/blob/master/examples/listdevs.c);
Do you have any idea?
Thanks,

mcuee, 2017-04-26:
Can you print out the debug log by setting env variable LIBUSB_DEBUG=4 and then run your program.

mcuee, 2017-10-19:
Close this for now as no feedbacks from OP for a long time.

closed 2017-10-19 by mcuee