libusb/libusb #294: Asychronous API Receive Timeout - Won't Fix unless USB Device is Closed and Re-Opened

crazenator, 2017-04-25:
Hello fellows,

I've similar issue but on beaglebone black, running jessie-linux; I'm using asynchronous API. I can connect mobile device, switch it to Google accessory and start communicating; somewhere in the middle, a reception would time-out and no retries would help unless I close libusb device handle and reopen it. If I see the packet capture logs, I can see IN-NAK for the message. Can anyone help in fixing it

crazenator, 2017-04-26:
I've noticed that if the transfer size is multiple of 512 bytes, the read attempt after writing the data fails. FYI, I'm using bulk endpoints for communication.

dickens, 2017-04-26:
Sounds like your device may be expecting a ZLP to indicate the end of the
transfer. What happens if you set the LIBUSB_TRANSFER_ADD_ZERO_PACKET in
the transfer flags before you submit?

crazenator, 2017-04-26:
I tried the flag but it didn't work

mcuee, 2017-04-26:
It could be related to problem with the device FW (eg: data toggle related). I remember similar things last time. Please try to capture usbmon logs and then post the questions to the libusb-devel mailing list and I believe Alan Stern will be able to tell the problems. Thanks.

diabolo38, 2017-09-18:
If you get 1536 bytes then i'll expect transfer to stop ie you get all your data as you get a short packet. You need to issue a new request to get more data.
What gadget are you unning on the device?

PratikGajera, 2017-09-19:
I have used iris scanner. Its application is already in windows but I want to develop for linux platform in beaglebone balck. But a problem occurs in reading large amount bulk data from asynchrouns bulk-in endpoint. When data is read without libusb_control_events() then all expected data is getting from call back function, but call back function is responding late due to which I coundn't send another command  to device. And if I used libusb_control_events() then I am getting time out error.

crazenator, 2017-09-19:
Just for the record, if I use LIBUSB_TRANSFER_ADD_ZERO_PACKET flag to send ZLP, it is more stable. From USB specifications, I can confirm that it states that if the length of the packet is multiple of 512 bytes, ZLP is required to avoid this issue.
I knew this 512 bytes boundary for control packet but didn't know about data packets. I hope this helps, thanks to @dickens

mcuee, 2017-10-10:
So I assume I can close the issue. If not, please reopen.

closed 2017-10-10 by mcuee