libusb/libusb #296: README file required

andimik, 2017-05-04:
I had to copy README.md to README in order to get `autogen.sh` running!

Otherwise I get this error:

> $ ./autogen.sh 
> autoreconf: Entering directory `.'
> autoreconf: configure.ac: not using Gettext
> autoreconf: running: aclocal --force -I m4
> autoreconf: configure.ac: tracing
> autoreconf: running: libtoolize --copy --force
> libtoolize: putting auxiliary files in `.'.
> libtoolize: copying file `./ltmain.sh'
> libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
> libtoolize: copying file `m4/libtool.m4'
> libtoolize: copying file `m4/ltoptions.m4'
> libtoolize: copying file `m4/ltsugar.m4'
> libtoolize: copying file `m4/ltversion.m4'
> libtoolize: copying file `m4/lt~obsolete.m4'
> autoreconf: running: /usr/bin/autoconf --force
> autoreconf: running: /usr/bin/autoheader --force
> autoreconf: running: automake --add-missing --copy --force-missing
> Makefile.am: error: required file './README' not found
> autoreconf: automake failed with exit status: 1
>

dickens, 2017-05-04:
README should be symlinked to README.md. What version of autotools are you running? And what platform are you building on?

dickens, 2017-06-06:
Giving this one more week for more info and then closing.

andimik, 2017-06-07:
I found the error 

Please don't link that as it does not work in the zip file. A user has to clone the project, which makes no sense if the PC is offline and gets everything from downloaded zip source files.

LudovicRousseau, 2017-10-18:
1. Where did you get the `.zip` file?
1. Why don't you fetch the `.tar.bz2` file since you use a Unix system?

andimik, 2017-11-06:
Because you always get a zip here on github.com 

See button Clone or download

LudovicRousseau, 2017-11-06:
I see. github does **not** distribute a correct ``.zip`` file.

Either clone the repository, or use a complete ``.tar.bz2`` archive available from the libusb web site http://libusb.info/

closed 2017-11-06 by LudovicRousseau