libusb/libusb #297: libusb cannot recover after device unplug on Windows

ecejeff, 2017-05-05:
I've got a device with 6 concurrent bulk transfers to the same endpoint. Sometimes when unplugging the device, libusb gets into a state where it cannot recover. On a fast computer (like my development PC), this happens maybe 10% of the time. On a slow computer it's closer to 100%. Tested on various systems and it definitely follows this trend.

Tested on Windows 7 64-bit and Windows XP 32-bit.

Seems to be an issue where the file descriptors and transfers get out of sync. Seems like a race condition to me.

Here's the beginning of the relevant log portion, where we can see a couple of successful transfers full of data. Everything is going great...
```
...
[43.002000] [00001afc] libusb: debug [libusb_get_next_timeout] next timeout in 0.506613s
[43.002000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[43.002000] [00001afc] libusb: debug [handle_events] poll fds modified, reallocating
[43.002000] [00001afc] libusb: debug [handle_events] poll() 7 fds with timeout in 100ms
[43.100000] [00001afc] libusb: debug [handle_events] poll() returned 1
[43.100000] [00001afc] libusb: debug [windows_handle_events] checking fd 1 with revents = 0001
[43.100000] [00001afc] libusb: debug [usbi_remove_pollfd] remove fd 1
[43.100000] [00001afc] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 22720
[43.100000] [00001afc] libusb: debug [usbi_handle_transfer_completion] transfer 00783D14 has callback 0F6F2EF0
[43.100000] [00001afc] libusb: debug [libusb_submit_transfer] transfer 00783D14
[43.100000] [00001afc] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 83 with interface 0
[43.100000] [00001afc] libusb: debug [winusbx_submit_bulk_transfer] reading 22720 bytes
[43.100000] [00001afc] libusb: debug [usbi_add_pollfd] add fd 1 events 1
[43.101000] [00001afc] libusb: debug [libusb_get_next_timeout] next timeout in 0.505906s
[43.101000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[43.101000] [00001afc] libusb: debug [handle_events] poll fds modified, reallocating
[43.101000] [00001afc] libusb: debug [handle_events] poll() 7 fds with timeout in 100ms
[43.199000] [00001afc] libusb: debug [handle_events] poll() returned 1
[43.199000] [00001afc] libusb: debug [windows_handle_events] checking fd 2 with revents = 0001
[43.199000] [00001afc] libusb: debug [usbi_remove_pollfd] remove fd 2
[43.199000] [00001afc] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 0, size 22720
[43.199000] [00001afc] libusb: debug [usbi_handle_transfer_completion] transfer 00783C8C has callback 0F6F2EF0
[43.199000] [00001afc] libusb: debug [libusb_submit_transfer] transfer 00783C8C
[43.199000] [00001afc] libusb: debug [winusbx_submit_bulk_transfer] matched endpoint 83 with interface 0
[43.199000] [00001afc] libusb: debug [winusbx_submit_bulk_transfer] reading 22720 bytes
[43.199000] [00001afc] libusb: debug [usbi_add_pollfd] add fd 2 events 1
```

At this point libusb realizes the device is gone, and calls 4 of the 6 callbacks with a stall error code. The remaining 2 transfers time out, and libusb tries to call AbortPipe, which fails (because the device is gone?). 
```
[43.199000] [00001afc] libusb: debug [libusb_get_next_timeout] next timeout in 0.505905s
[43.199000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[43.199000] [00001afc] libusb: debug [handle_events] poll fds modified, reallocating
[43.199000] [00001afc] libusb: debug [handle_events] poll() 7 fds with timeout in 100ms
[43.231000] [00001afc] libusb: debug [handle_events] poll() returned 1
[43.232000] [00001afc] libusb: debug [windows_handle_events] checking fd 3 with revents = 0001
[43.232000] [00001afc] libusb: debug [usbi_remove_pollfd] remove fd 3
[43.232000] [00001afc] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 31, size 7104
[43.232000] [00001afc] libusb: debug [windows_transfer_callback] detected endpoint stall
[43.232000] [00001afc] libusb: debug [usbi_handle_transfer_completion] transfer 00783F34 has callback 0F6F2EF0
[43.232000] [00001afc] libusb: debug [libusb_free_transfer] transfer 00783F34
[43.232000] [00001afc] libusb: debug [libusb_get_next_timeout] next timeout in 0.571308s
[43.233000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[43.233000] [00001afc] libusb: debug [handle_events] poll fds modified, reallocating
[43.233000] [00001afc] libusb: debug [handle_events] poll() 6 fds with timeout in 100ms
[43.235000] [00001afc] libusb: debug [handle_events] poll() returned 1
[43.235000] [00001afc] libusb: debug [windows_handle_events] checking fd 4 with revents = 0001
[43.235000] [00001afc] libusb: debug [usbi_remove_pollfd] remove fd 4
[43.235000] [00001afc] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 31, size 0
[43.235000] [00001afc] libusb: debug [windows_transfer_callback] detected endpoint stall
[43.235000] [00001afc] libusb: debug [usbi_handle_transfer_completion] transfer 00783EAC has callback 0F6F2EF0
[43.235000] [00001afc] libusb: debug [libusb_free_transfer] transfer 00783EAC
[43.236000] [00001afc] libusb: debug [libusb_get_next_timeout] next timeout in 0.667623s
[43.236000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[43.236000] [00001afc] libusb: debug [handle_events] poll fds modified, reallocating
[43.236000] [00001afc] libusb: debug [handle_events] poll() 5 fds with timeout in 100ms
[43.236000] [00001afc] libusb: debug [handle_events] poll() returned 1
[43.236000] [00001afc] libusb: debug [windows_handle_events] checking fd 5 with revents = 0001
[43.236000] [00001afc] libusb: debug [usbi_remove_pollfd] remove fd 5
[43.236000] [00001afc] libusb: debug [windows_transfer_callback] handling I/O completion with errcode 31, size 0
[43.236000] [00001afc] libusb: debug [windows_transfer_callback] detected endpoint stall
[... 327 lines cut ...]
[50.950000] [00001afc] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[50.950000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[50.950000] [00001afc] libusb: debug [handle_events] poll() 3 fds with timeout in 100ms
[50.950000] [00001afc] libusb: debug [handle_events] poll() returned 2
[50.950000] [00001afc] libusb: debug [windows_handle_events] checking fd 1 with revents = 0001
[50.950000] [00001afc] libusb: error [windows_handle_events] could not find a matching transfer for fd 1
[50.950000] [00001afc] libusb: error [handle_events] backend handle_events failed with error -5
[50.950000] [00001afc] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[50.950000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[50.950000] [00001afc] libusb: debug [handle_events] poll() 3 fds with timeout in 100ms
[50.950000] [00001afc] libusb: debug [handle_events] poll() returned 2
[50.950000] [00001afc] libusb: debug [windows_handle_events] checking fd 1 with revents = 0001
[50.950000] [00001afc] libusb: error [windows_handle_events] could not find a matching transfer for fd 1
[50.950000] [00001afc] libusb: error [handle_events] backend handle_events failed with error -5
[50.950000] [00001afc] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[50.950000] [00001afc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[50.950000] [00001afc] libusb: debug [handle_events] poll() 3 fds with timeout in 100ms
[50.950000] [00001afc] libusb: debug [handle_events] poll() returned 2
[50.950000] [00001afc] libusb: debug [windows_handle_events] checking fd 1 with revents = 0001
[50.950000] [00001afc] libusb: error [windows_handle_events] could not find a matching transfer for fd 1
[50.950000] [00001afc] libusb: error [handle_events] backend handle_events failed with error -5
...
```

This pattern repeats forever and the program never collects more data.

ecejeff, 2017-05-05:
By the way, I don't get timeout callbacks for the 2 transfers that "time out". I only get error callbacks on the 4 that "stalled".

brendanbates89, 2017-05-05:
From a cursory glance at the source code (and I am not well versed on libusb): it appears to me that the two transfers must never get removed from the flying transfers list, which means they refer to stale file descriptors.  In the `windows_handle_events` method, all flying transfers are looped through and matched to file descriptors.  I wonder if after the `AbortPipe` operation fails, the transfers should somehow find their way to the `usbi_handle_transfer_cancellation` function so that they are removed from the flying transfers list.  I could be completely wrong on this however.

mcuee, 2017-09-10:
In any case, libusb Windows does not support hotplug as of now.

mcuee, 2017-10-24:
Close this for now. Will reopen this if things still happen after the hotplug feature is implemented.

brendanbates89, 2017-10-24:
I have no idea if this is related to hotplug or not, but I figure I'll comment our "fix" anyways: the [patch is here](https://gist.github.com/brendanbates89/2a9d748607bf02cc819a1a11f4c8917c).  Again, I have no idea if this is the proper fix or not, but to sum it up: During `AbortPipe`, something bad would happen to trigger a "no device" error.  This would cause the `usbi_handle_transfer_cancellation` to never get called, so the transfer is actually left hanging in that state.  Another transfer would then attempt to use the transfer fd, which was never cleaned up.  So, when the "no device" error from `AbortPipe` happens, a list of these bad transfers is maintained and then the `usbi_handle_transfer_cancellation` is called on each bad transfer.  This has solved most of our problems, though I am not sure if it's really the best way to fix this.

ecejeff, 2017-10-24:
I don't see how hotplug (callbacks on new devices and devices leaving) is related to this bug in any way.

dickens, 2020-08-18:
With all the recent changes to how Windows handles transfers and events, I'd like to know if this is still an issue.  Can anyone on this thread confirm?

ecejeff, 2020-08-18:
Using the latest git I now get a different error when unplugging one of two devices now:

libusb: error [windows_iocp_thread] GetQueuedCompletionStatus failed: [31] A device attached to the system is not functioning.
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2
libusb: warning [handle_timeout] async cancel failed -5 errno=2

And the other device becomes hung as well.

ecejeff, 2020-08-18:
Here's a more detailed log (not the same run, but the same effect).

[libusb_debug_log.txt](https://github.com/libusb/libusb/files/5091243/libusb_debug_log.txt)

dickens, 2020-08-18:
Thanks for the quick feedback!  I just pushed a commit (37e8b1334e59485e1d4735f5f67b31eba37ae5c9) to address an issue pointed out by your log.  Can you try again?

ecejeff, 2020-08-18:
It fixed the unplugging issue, but I've uncovered another in the process of testing the fix. It happens during tear down.

```
C:\Libraries\Asphodel_Win64_b727e2c1d314c36642ae0b3571db0676e5aa66a5>asphodel_streaming.exe
Found 2 devices!
Enabling 2 streams from UVB687
Enabling 3 streams from WMRP2147
Press any key to stop data collection...
Disabling 2 streams from UVB687
Disabling 3 streams from WMRP2147
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: warning [do_close] A cancellation for an in-flight transfer hasn't completed but closing the device handle
```
I then get a segfault (the windows equivalent rather). Here's the exception:

```
Unhandled exception at 0x000007FEDD5A0988 (libusb-1.0.dll) in asphodel_streaming.exe: 0xC0000005: Access violation reading location 0x0000000000000040. occurred
```

Here' the backtrace:
```
libusb-1.0.dll!windows_handle_transfer_completion(usbi_transfer * itransfer) Line 763
	at C:\Libraries\libusb\libusb\os\windows_common.c(763)
libusb-1.0.dll!handle_event_trigger(libusb_context * ctx) Line 2083
	at C:\Libraries\libusb\libusb\io.c(2083)
libusb-1.0.dll!handle_events(libusb_context * ctx, timeval * tv) Line 2192
	at C:\Libraries\libusb\libusb\io.c(2192)
libusb-1.0.dll!libusb_handle_events_timeout_completed(libusb_context * ctx, timeval * tv, int * completed) Line 2293
	at C:\Libraries\libusb\libusb\io.c(2293)
Asphodel64.dll!usb_poll_device(AsphodelDevice_t * device, int milliseconds, int * completed) Line 2432
	at c:\codebuild\tmp\output\src570898187\src\bitbucket.org\suprocktech\asphodel\src\asphodel_usb.c(2432)
```

I've seen similar warnings on older versions, but never a segfault. I understand this may be an artifact of my code, but is there an easy way for the library to guard against this?

dickens, 2020-08-18:
Closing a device while there are outstanding transfers is 100% an application error.  Handling this gracefully has been discussed in #540, #610 and #703.

mcuee, 2021-06-05:
I will close this and please refer to #703.

mcuee, 2023-02-06:
Relevant info:
1) https://github.com/libusb/libusb/issues/703
2) https://github.com/libusb/libusb/issues/182
3) https://github.com/libusb/libusb/issues/297
4) https://github.com/libusb/libusb/issues/540
5) https://github.com/libusb/libusb/issues/610

ijprest-haptx, 2024-10-14:
We've experienced something similar on device disconnects.  In particular, we get the `error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command` message, which corresponds to Win32's `ERROR_BAD_COMMAND`.

We're currently experimenting with this small patch, which just translates `ERROR_BAD_COMMAND` (and a couple other errors, for good measure) into `LIBUSB_ERROR_NO_DEVICE` inside the `*do_bulk_transfer` functions:  https://github.com/libusb/libusb/commit/9e0349e9cb1908c42d62a78358b601d656f2de56.  Initial tests show that it solves the problem in our case, and there is precedent elsewhere in libusb for converting these specific Win32 errors into `LIBUSB_ERROR_NO_DEVICE`.

We're happy to upstream this change, if you want it.  Let me know...

mcuee, 2025-09-09:
@ijprest-haptx 

Now that Windows hotplug PR has been merged. Just wondering if you can test libusb master to see if you still have the issues or not. Thanks.