libusb/libusb #299: Access to device in while in hotplug_callback()

hinxx, 2017-05-11:
I've registered a callback with libusb_hotplug_register_callback() and I'm trying to call libusb_get_string_descriptor_ascii() when new device appears. It fails with LIBUSB_ERROR_BUSY while in separate thread that is doing libusb_handle_events_timeout_completed().

What is the preferred practice in doing I/O when acting on LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED | LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT events?

Should I deffer I/O and do it in another thread since I'm providing a library that wraps around libusb?

dickens, 2017-05-12:
Yes, device I/O must be deferred to another thread or at least outside of the context of event handling. The reason that the synchronous APIs cannot be used from within a hotplug callback is because they need to do event handling themselves, and this cannot be safely done recursively.

I'll close this as soon as the online documentation is updated to reflect that the synchronous APIs return LIBUSB_ERROR_BUSY from event handling context. The documentation hasn't been updated since the latest release.

hinxx, 2017-05-15:
Yep, that solved it. Thanks!

hjelmn, 2017-08-19:
@dickens Has the documentation been updated?

dickens, 2017-08-19:
@hjelmn not yet. It would push out documentation relating to APIs that aren't available in any release yet, so I'm waiting until the release.

dickens, 2018-03-25:
Documentation for release has been updated.

closed 2018-03-25 by dickens

adamsunp, 2022-10-26:
> Yep, that solved it. Thanks!

Hi, I have the same problem with you, how to use deffrent context ?   I only register a hotplug_callbak with event "LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED ",   when I add a device, it will return LIBUSB_ERROR_BUSY.

adamsunp, 2022-10-26:
My code is similar to this： https://github.com/libusb/libusb/issues/601